realsenseai/realsense-ros #1663: error control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11 while trying slam

demoncoderad, 2021-02-03:
Hey!
So I am trying to do a slam thingy. I am doing this for a showcase currently (i'll use the D435 and T265 for more after my exams) where I just have to show that it can walk around and it gives in text the directions for going from x point in map to a point y. That I somehow am able to do with help of https://neilyoung.serveblog.net/ But I wish to have Object avoidance, and when I saw this I thought this might be quick easy way to get it done. But im facing some problems, after installing this and running` roslaunch realsense2_camera rs_camera.launch` i get a warning ` 03/02 20:30:26,300 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11` and it doesn't go further. I would appreciate some help. I'm using Ubuntu 18.04 and installed with the same steps just changing ros-kinetic to ros-melodic.
here's the full output
`
... logging to /home/adityarc/.ros/log/833fd1a6-6630-11eb-b16a-dc85dea045af/roslaunch-pop-os-6296.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://pop-os:34391/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 78 lines cut ...]
[ INFO] [1612364424.533627699]: setupStreams...
[ INFO] [1612364424.556888751]: insert Depth to Stereo Module
[ INFO] [1612364424.556989185]: insert Color to RGB Camera
 03/02 20:30:24,610 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
[ INFO] [1612364424.660969509]: SELECTED BASE:Depth, 0
[ INFO] [1612364424.681647468]: RealSense Node Is Up!
 03/02 20:30:24,746 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:24,797 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:24,949 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:25,026 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:25,077 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:25,232 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:25,282 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:25,435 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:25,485 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:26,250 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 03/02 20:30:26,300 WARNING [139819288090368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
^C[camera/realsense2_camera_manager-2] killing on exit
[camera/realsense2_camera-3] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

RealSenseSupport, 2021-02-05:
@demoncoderad Did you get chance to try if any issue to run RealSenseViewer?  And also noticed that the USB type is USB2. If it's not intended to be USB2, please check USB cable or USB port to make it USB3 connection.

demoncoderad, 2021-02-07:
Hi, it works good in realsense viewer. I have a USB 3 and USB 2 port so I plugged in the T265 (afaik) in USB 3 and D435 on Usb 2

RealSenseSupport, 2021-02-10:
@demoncoderad Is the issue existing if you only connect T265 or D435 only? Thanks for your feedback!

doronhi, 2021-02-10:
Hi,
realsense2_camera, from package ros-melodic-realsense2-camera, uses a librealsense2 version built with RSUSB backend.
realsense-viewer is not installed via ros-melodic-realsense2-camera. Therefore I assume you also installed librealsense2-utils which brings along another version of librealsense2 build with v4l backend.

You did not specify what machine you have but still, since the viewer works well for you, I would recommend removing the ros-melodic-realsense2-camera installation and build realsense2_camera from source. It will link with the librealsense2 version already installed on your machine (v4l backend), the same one realsense-viewer uses.

demoncoderad, 2021-02-12:
Thanks, i'll look into it @doronhi 
and @RealSenseSupport it basically hanged there when I run with either of the two, or both

RealSenseSupport, 2021-02-19:
@demoncoderad With the suggestions from doronhi, any update from your side? Thanks!

RealSenseSupport, 2021-03-16:
@demoncoderad Any update from your side? Looking forward to your reply. Thanks!

@bigbellmercy As your issue happened on different camera and platforms, could you please create another issue to track it separately so the clue will be more clear? Thanks!

bigbellmercy, 2021-03-16:
> @demoncoderad Any update from your side? Looking forward to your reply. Thanks!
> 
> @bigbellmercy As your issue happened on different camera and platforms, could you please create another issue to track it separately so the clue will be more clear? Thanks!

Yes, a new issue was created as #1758. Thanks.

RealSenseSupport, 2021-03-22:
@bigbellmercy Thanks for your quick action! We'll look into that.
@demoncoderad Any update from your side? Please note that the issue will be closed if we don't hear from you for another week. Thanks!

RealSenseSupport, 2021-03-29:
@demoncoderad Sorry that we didn't get response from you for weeks, and this issue will be closed at this point. Please create another ticket if you still have questions or issues. Thanks!

closed 2021-03-29 by RealSenseSupport

sasilva1998, 2022-03-30:
was this solved, I am having the same issue, I am getting the same message:

`(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11`

at first I get some pointcloud from the D435 but then it stops, I am using a Jetson Nano, I could make it work about a year ago, along with the T265 but now I am stuck with this.

ingramator, 2022-04-13:
This bug still persists

YXGuan, 2022-04-20:
Same issue here

B7O7B, 2022-04-25:
@YXGuan @ingramator @sasilva1998, I encountered the same problem and after a bunch of hours spent online, I read realsense engineers are still working on that issue, and in some cases, that error message can be ignored. Also, what I did was reset the camera upon launch. `roslaunch realsense2_camera rs_camera.launch initial_reset:=true` . In addition, you can try updating your kernel, Incase you haven't done that in a while.

SyllogismRXS, 2022-06-23:
I am running ROS2 Galactic on Ubuntu 20.04 and I was experiencing this same error message. The issue was that the realsense2 camera node was trying to publish the raw point clouds over the WiFi network to a desktop computer that was running rviz2 (in ROS2, two computers running ROS2 nodes on the same local network will automatically detect publishers/subscribers because of DDS. You can change this behavior by changing the [ROS_DOMAIN_ID](https://docs.ros.org/en/foxy/Concepts/About-Domain-ID.html#the-ros-domain-id)). I imagine if you are running ROS1, you might experience the same error message if you are trying to publish the raw point cloud over a WiFi network (but you would have had to manually change the ROS_MASTER_URI in ROS1). 

This network issue was strange to track down for me, so I imagine that might be why this issue still exists. It worked fine when I used the realsense-viewer program, but failed when running the ROS2 node. I second guessed the USB 3 cable, the camera hardware, but in the end the only difference I could really point to was that ROS2 was performing network operations and the realsense-viewer wasn't.  

tldr: make sure you aren't publishing the camera's raw point clouds over a WiFi network.

kokun66, 2025-04-11:
is this issue solved? I am having same error messages :<