realsenseai/realsense-ros #1664: ordered_pc generates bogus messages

mferenduros, 2021-02-03:
PR #1627 [modifies the point-cloud message width & height](https://github.com/IntelRealSense/realsense-ros/commit/c5403a28af7a82a30e970f78d4e2c1a3649686b0#diff-a54ad2538ccd7b6cce4c6c6d8f57af062bded3d802b88e7b938faefb6f02f4caR2252) but does not update the row_step to match. pcl::fromPCLPointCloud2() uses row_step to advance through the image, and hence segfaults when fed a message generated with ordered_pc message.

MartyG-RealSense, 2021-02-03:
Hi @mferenduros  Thanks very much for your report.  I will tag @doronhi the RealSense ROS developer into this discussion to highlight it to them.

seanroelofs, 2021-02-04:
I have been running into segfaults when working with the d435 point cloud in the last few days. This is a real issue.

seanroelofs, 2021-02-04:
@mferenduros did you find a quick patch for this?

joelbudu, 2021-02-04:
I am also facing this same issue with segmentation faults pcl::fromPCLPointCloud2() with the ordered_pc

doronhi, 2021-02-04:
As rviz2 experiences no issues, can you detail a method for reproduction? example node maybe?

joelbudu, 2021-02-04:
@doronhi 
This is an example callback subscribing to a PointCloud2 topic. The pcl::fromROSMsg(*msg, *cloud) will cause a Segmentation fault.

```
void GraspDetectorGPD::cloud_callback(const sensor_msgs::msg::PointCloud2::SharedPtr msg)
{
      PointCloudPointNormal::Ptr cloud(new PointCloudPointNormal);
      pcl::fromROSMsg(*msg, *cloud);
}
```

Extracted this from the a large codebase - [https://github.com/intel/ros2_grasp_library/blob/master/grasp_ros2/src/grasp_detector_gpd.cpp#L178](url) in this  the [ros2_grasp_library](https://github.com/intel/ros2_grasp_library)
cc: @sharronliu

mferenduros, 2021-02-04:
> As rviz2 experiences no issues, can you detail a method for reproduction? example node maybe?

I will make a quick node+launchfile to demonstrate the crash, but a simple `rostopic echo -n 1 /camera/depth/color/points | head` should show that row_step is not equal to width*point_step as you would expect, but rather is equal to data.size().

rviz indeed appears to ignore the row_step and just walks through the entire cloud [in a single loop](https://github.com/ros-visualization/rviz/blob/melodic-devel/src/rviz/default_plugin/point_cloud2_display.cpp#L119) assuming that rows are tightly packed.

@seanroelofs My workaround was to rewrite [pcl::fromROSMsg](https://github.com/ros-perception/perception_pcl/blob/melodic-devel/pcl_conversions/include/pcl_conversions/pcl_conversions.h#L560) like so:
`  template<typename T>
  void fromROSMsgFixed(const sensor_msgs::PointCloud2 &cloud, pcl::PointCloud<T> &pcl_cloud)
  {
    pcl::PCLPointCloud2 pcl_pc2;
    pcl_conversions::toPCL(cloud, pcl_pc2);
    pcl_pc2.row_step = pcl_pc2.point_step * pcl_pc2.width;
    pcl::fromPCLPointCloud2(pcl_pc2, pcl_cloud);
  }`

joelbudu, 2021-02-04:
I've created a PR #1668 for this  @mferenduros @doronhi

MartyG-RealSense, 2021-02-04:
Thanks very much for your PR contribution @joelbudu

mferenduros, 2021-02-05:
> As rviz2 experiences no issues, can you detail a method for reproduction? example node maybe?

Here's a node as requested, add it to a workspace with realsense2_camera and run `roslaunch pointcloud2_crash crash.launch`

seanroelofs, 2021-02-05:
I believe there is another problem related to the ordered_pc option. The colored point cloud published by the d435 does not look correct.

[image]

if you look on the bottom and right sides, you can see black points. These points are within the view of the depth sensor, but not within the view of the color camera. These points should not be published as part of the colored pointcloud because it is impossible to associate a color with them. This is important because now the grid of points in this pointcloud does not match up with the grid of points in the regular rgb image.

doronhi, 2021-02-07:
@joelbudu , @mferenduros , Thanks for your work. It really helps! Could you please check my remark on #1676.
@seanroelofs, this fix does not explain the segmentation fault in an unordered pointcloud you describe. Can you please provide a test case?

MartyG-RealSense, 2021-02-19:
Hi @mferenduros @joelbudu @seanroelofs 

Do any of you require further assistance with this case, please?  Thanks!

closed 2021-02-25 by MartyG-RealSense