realsenseai/realsense-ros #1665: D435i IMU timestamps mismatch

meghdeepj, 2021-02-04:
ROS Melodic on Nvidia Jetson Nano (Jetpack 4.4) [kernel: 4.9.140-tegra]
RealSense ROS v2.2.21
Built with LibRealSense v2.41.0
Running with LibRealSense v2.41.0
Device FW version: 05.12.07.100

**D435 IMU data's timestamp is 2x of actual**

realsense2_camera launch:

```
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: 2.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 6
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud,dispar...
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
[... 117 lines cut ...]
average delay: 0.043
	min: 0.042s max: 0.047s std dev: 0.00191s window: 6
average delay: 0.042
	min: 0.042s max: 0.047s std dev: 0.00144s window: 12
average delay: 0.042
	min: 0.042s max: 0.047s std dev: 0.00120s window: 18
average delay: 0.042
	min: 0.042s max: 0.047s std dev: 0.00105s window: 24
average delay: 0.042
	min: 0.041s max: 0.047s std dev: 0.00099s window: 30
average delay: 0.042
```
```
rostopic delay /camera/imu
subscribed to [/camera/imu]
average delay: -1612437065.055
	min: -1612437065.059s max: -1612437065.053s std dev: 0.00132s window: 378
average delay: -1612437065.059
	min: -1612437065.067s max: -1612437065.053s std dev: 0.00419s window: 778
average delay: -1612437065.061
	min: -1612437065.067s max: -1612437065.053s std dev: 0.00445s window: 1179
average delay: -1612437065.062
	min: -1612437065.067s max: -1612437065.053s std dev: 0.00400s window: 1579
average delay: -1612437065.061
```

MartyG-RealSense, 2021-02-04:
Hi @meghdeepj   Could you try setting **global_time_enabled** to False please if it is not False already?

https://github.com/IntelRealSense/realsense-ros/issues/1454#issuecomment-739059385

meghdeepj, 2021-02-05:
Thank you MartyG for your response. 
I set global_time_enabled to False, this has reduced the delay from multiple decades but there is still a delay of about 7-8 seconds in the future.

`subscribed to [/camera/imu]
average delay: -7.874
	min: -7.877s max: -7.849s std dev: 0.00323s window: 369
average delay: -7.874
	min: -7.877s max: -7.849s std dev: 0.00291s window: 768
average delay: -7.874
	min: -7.877s max: -7.849s std dev: 0.00291s window: 857

rostopic delay /camera/depth/color/points
subscribed to [/camera/depth/color/points]
average delay: 0.074
	min: 0.072s max: 0.075s std dev: 0.00105s window: 6
average delay: 0.075
	min: 0.066s max: 0.088s std dev: 0.00652s window: 12
average delay: 0.077
	min: 0.065s max: 0.094s std dev: 0.00837s window: 18
average delay: 0.075
	min: 0.061s max: 0.094s std dev: 0.00856s window: 21
`

MartyG-RealSense, 2021-02-05:
Good news to hear that setting global time to False helped.  An alternate fix for this timestamp issue that has worked for some but not for others is to add **initial_reset:=true** to the end of the roslaunch instruction to reset the camera at launch.   Does doing so produce any further improvement if you use this roslaunch command in addition to the global time = false setting that you are already using, please?

meghdeepj, 2021-02-05:
Yes that worked!
Using both **global_time_enabled=False** and **initial_reset:=true** solved the time stamp issue.

Thanks a lot

MartyG-RealSense, 2021-02-05:
Great, thanks!

meghdeepj, 2021-02-06:
The delay was reduced, but it is currently -0.001 ms (1ms in the future)
This is creating issues with the ros tf listener. 

```
ExtrapolationException: Lookup would require extrapolation into the future.  Requested time 1612446072.878581524 but the latest data is at time 1612446072.868384600, when looking up transform from frame [camera_imu_optical_frame] to frame [base_link]
```

meghdeepj, 2021-02-06:
Sorry, that was an early celebration.
The delay was reduced, but it is currently -0.001 ms (1ms in the future)
This is creating issues with the ros tf listener. 

```
average delay: -0.001
	min: -0.002s max: 0.0018s std dev: 0.00185s window: 3356
average delay: -0.001
	min: -0.002s max: 0.0018s std dev: 0.00191s window: 3756
average delay: -0.001
	min: -0.002s max: 0.0018s std dev: 0.00190s window: 4157
average delay: -0.001
	min: -0.002s max: 0.0018s std dev: 0.00190s window: 4560
average delay: -0.001

```

```
ExtrapolationException: Lookup would require extrapolation into the future.  Requested time 1612446862.443124533 but the latest data is at time 1612446862.433134079, when looking up transform from frame [camera_imu_optical_frame] to frame [base_link]

```

MartyG-RealSense, 2021-02-06:
Some advice for dealin with the **ExtrapolationException** error was provided to a D435 user in the link below.

https://answers.ros.org/question/331027/realsense-d435-lookup-would-require-extrapolation-into-the-future/

MartyG-RealSense, 2021-02-12:
Hi @meghdeepj  Do you require further assistance with this case, please?  Thanks!

meghdeepj, 2021-02-12:
Hello @MartyG-RealSense,
Sorry I forgot to respond with the issue.
Yes, I added a condition that checks if the frame exists to begin with and thereafter catching the TF exceptions. It is working.

Thanks for your help.

closed 2021-02-12 by meghdeepj

MartyG-RealSense, 2021-02-12:
No problem at all.  Thanks very much for the update - I'm pleased to hear that you developed a solution!