realsenseai/realsense-ros #1666: Capture  /camera/depth_registered/ points em python

GabrielMLB, 2021-02-04:
Hello, I'm new in ros, I'd like to take a question and receive suggestions.

I ran  
**roslaunch realsense_ros_camera rs_rgbd.launch align_depth:=true** 

I want to consume the topic:

**/camera/depth_registered/points**

My code in Python is the following:

```python
#!/usr/bin/env python
import rospy
from cv_bridge import CvBridge, CvBridgeError
from sensor_msgs.msg import Image

def callback(data):
  rospy.loginfo(rospy.get_caller_id() + 'I heard %s', data.data)

def listener():
  rospy.init_node('live_weight',anonymous=True)
  rospy.Subscriber("/camera/depth_registered/points", Image,callback=callback, queue_size=1)
  rospy.spin()

if __name__ == '__main__':
  listener()
```

I get the following error when I execute the script:

**[ERROR] [1612442204.747954507]: Client [/live_weight_19449_1612442203535] wants topic /camera/depth_registered/points to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection**

But when I execute **rostopic echo /camera/depth_registered/points** in a terminal the data arrive correctly.

MartyG-RealSense, 2021-02-04:
Hi @GabrielMLB   I hope that the Python script i the link below will be a useful reference for subscribing with rospy.Subscriber to /camera/depth_registered/points

https://answers.ros.org/question/325917/pointcloud-value-at-a-given-index/

GabrielMLB, 2021-02-04:
Thanks again @MartyG-RealSense, that solved the problem.

MartyG-RealSense, 2021-02-04:
Great news @GabrielMLB  - thanks very much for the update!

MartyG-RealSense, 2021-02-10:
Case closed due to solution provided and no further comments received.

closed 2021-02-10 by MartyG-RealSense