realsenseai/realsense-ros #1669: [ERROR] [1612488937.199917748]: An error has occurred during frame callback: map::at

darmin-hsiung, 2021-02-05:
An error has occurred during frame callback: 
[ERROR] [1612488937.199917748]: An error has occurred during frame callback: map::at

Hi there,

Specs: Running Ubuntu 18.04 LTS on an Intel NUC, Camera: L515, USB3 connection, SDK: 2.41.0.

I am trying to get a point cloud with the following launch command. roslaunch realsense2_camera demo_pointcloud.launch The point cloud and camera image show up fine and I don't visually see any problem other than the following:

I seem to consistently get the error message shown in the terminal: 
[ERROR] [1612488936.949575351]: An error has occurred during frame callback: map::at
[ERROR] [1612488937.050368488]: An error has occurred during frame callback: map::at
[ERROR] [1612488937.134904899]: An error has occurred during frame callback: map::at
...

I have tried installing the SDK with Debian Package as well as from source and then following this.

Any and all help would be greatly appreciated, thanks in advance.

doronhi, 2021-02-10:
Which realsense2_camera version, please?

girtslinde, 2021-02-13:
I'm seeing this too. I get 30 of those errors per second when I enable
`<arg name="align_depth" value="true"/>`

Versions are:
[ INFO] [1613254338.386005717]: RealSense ROS v2.2.21
[ INFO] [1613254338.386074815]: Built with LibRealSense v2.41.0
[ INFO] [1613254338.386111928]: Running with LibRealSense v2.41.0
[ INFO] [1613254339.141753266]: Device with name Intel RealSense L515 was found.
[ INFO] [1613254339.142684830]: Device USB type: 3.2
[ INFO] [1613254339.386221012]: Device Name: Intel RealSense L515
[ INFO] [1613254339.386376486]: Device FW version: 01.05.03.00

Image streams seem to be coming out fine. It just generates gigabytes of logs.

JStech, 2021-02-13:
Same issue, when run with `align_depth:=true`. This is an L515 on 20.04/Noetic, with kernel version `5.8.0-41-generic`.
```
> roslaunch realsense2_camera rs_camera.launch align_depth:=true
...
[ INFO] [1613258117.607338973]: RealSense ROS v2.2.21
[ INFO] [1613258117.607365135]: Built with LibRealSense v2.41.0
[ INFO] [1613258117.607375294]: Running with LibRealSense v2.41.0
[ INFO] [1613258118.044371259]: Device with name Intel RealSense L515 was found.
[ INFO] [1613258118.044667637]: Device USB type: 3.2
[ INFO] [1613258118.086266391]: Device Name: Intel RealSense L515
[ INFO] [1613258118.086308868]: Device FW version: 01.05.00.00
[ INFO] [1613258118.086338112]: Enable PointCloud: Off
[ INFO] [1613258118.086352564]: Align Depth: On
[ INFO] [1613258118.086362857]: Sync Mode: On
[ INFO] [1613258118.086414975]: Device Sensors: 
[ INFO] [1613258118.102626870]: L500 Depth Sensor was found.
[ INFO] [1613258118.113583227]: RGB Camera was found.
```

girtslinde, 2021-02-13:
The last debug message before the error is 
`publishAlignedDepthToOthers(...)`
so I would guess that the `map::at` exception complains about one of these in `BaseRealSenseNode::publishAlignedDepthToOthers()`:
```
        auto& info_publisher = _depth_aligned_info_publisher.at(sip);
        auto& image_publisher = _depth_aligned_image_publishers.at(sip);
```

doronhi, 2021-02-14:
Thanks for all the info.
Fixed in #1695

PR #1695 (2021-02-14): Avoid aligning confidence image
Should not align confidence image to depth as they are already aligned.
fix #1669
closed 2021-02-14 by doronhi