realsenseai/realsense-ros #1670: Is there a way to specificy to color stream type?

ulassbin, 2021-02-05:
Specifically I am interested in getting Y16 encoding because my application needs unrectified image. I have seen from 
https://community.intel.com/t5/Items-with-no-label/How-to-enable-both-Y16-unrectified-streams-from-D435-in-SDK2-0/td-p/618843
that Y16 stream is unrectified.

MartyG-RealSense, 2021-02-05:
Hi @ulassbin  In the RealSense ROS wrapper you can set stream **format** (infra, color, fisheye, depth, gyro, accel, pose) and its height, width and FPS but not the mode used for that format.

The ROS wrapper file **base_realsense_node.cpp** provides information about how the streams are defined in the wrapper.

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp#L89

ulassbin, 2021-02-05:
Changing the lines:
https://github.com/IntelRealSense/realsense-ros/blob/c5403a28af7a82a30e970f78d4e2c1a3649686b0/realsense2_camera/src/base_realsense_node.cpp#L114-L115
to
```
    _format[RS2_STREAM_COLOR] = RS2_FORMAT_Y16; // Adding this line.
    _image_format[RS2_STREAM_COLOR] = CV_16UC1;    // CVBridge type
    _encoding[RS2_STREAM_COLOR] = sensor_msgs::image_encodings::MONO16; // ROS message type
```
 and lines:
 https://github.com/IntelRealSense/realsense-ros/blob/c5403a28af7a82a30e970f78d4e2c1a3649686b0/realsense2_camera/src/base_realsense_node.cpp#L2137
 to
 ```
std::set<rs2_format> available_formats{ rs2_format::RS2_FORMAT_Y16, rs2_format::RS2_FORMAT_Y8 };
```
worked for me. When I have the time, I can make this format a parameter and send a pr.

Thanks!

closed 2021-02-05 by ulassbin

MartyG-RealSense, 2021-02-05:
Awesome to hear that you were able to develop a solution that worked for you @ulassbin - thanks very much for sharing your method and the offer of a PR contribution   :)