realsenseai/realsense-ros #1672: Frequency of depth and pointclouds got much slower when we subscribe compressedDepth

knorth55, 2021-02-05:
I found that we get low frequency of depth and pointcloud when we subscribe `compressedDepth` topic.
I think it is caused by `compressed_depth_image_transport` , but I want to share the issue.

My environment is as below:
OS: Ubuntu 18.04 + ROS Melodic
Camera: Realsense L515
librealsense2: 2.41.0-0~realsense0.3962
realsense-ros: 2.2.21

## How to reproduce

```bash
roslaunch realsense2_camera rs_rgbd.launch color_width:=960 color_height:=540 depth_width:=640 depth_height:=480
```

### when we do not subscribe `/camera/depth/image_rect_raw/compressedDepth`

both topic are almost 30 Hz.

```bash
$ rostopic hz /camera/depth/image_rect_raw
subscribed to [/camera/depth/image_rect_raw]
average rate: 30.056
	min: 0.032s max: 0.034s std dev: 0.00028s window: 29
average rate: 30.032
	min: 0.032s max: 0.034s std dev: 0.00021s window: 59
average rate: 30.030
	min: 0.027s max: 0.040s std dev: 0.00105s window: 89
```
```bash
$ rostopic hz /camera/depth_registered/points
subscribed to [/camera/depth_registered/points]
average rate: 32.844
	min: 0.012s max: 0.058s std dev: 0.01003s window: 27
average rate: 31.042
	min: 0.012s max: 0.058s std dev: 0.00884s window: 56
```

### when we subscribe `/camera/depth/image_rect_raw/compressedDepth`.
 
`/camera/depth/image_rect_raw` frequency is `1.5Hz` and we cannot get `/camera/depth_registered/points`.
```bash
$ rostopic hz /camera/depth/image_rect_raw
subscribed to [/camera/depth/image_rect_raw]
average rate: 1.642
	min: 0.609s max: 0.609s std dev: 0.00000s window: 2
average rate: 1.600
	min: 0.609s max: 0.641s std dev: 0.01571s window: 3
```

```bash
$ rostopic hz /camera/depth_registered/points
subscribed to [/camera/depth_registered/points]
no new messages
no new messages
```

cc. @sktometometo

knorth55, 2021-02-05:
I know that similar issue is reported in #369, but I didnt' know that the frequency of all the other depth related topics get lower by subscribing one topic `/camera/depth/image_rect_raw/compressedDepth`.
We can make it faster by changing `png_level` but the default is `9` (which means high quality).

RealSenseSupport, 2021-03-03:
Thanks for outlining this to the forum.

Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-03-22 by RealSenseSupport