realsenseai/realsense-ros #1673: How to send RGB stream, Depth Stream and depth data over WiFi from Jetson Nano to PC? (PS: I have already tried lot of things. Also mentioning them in comment) )

GauravKakade, 2021-02-05:
Hello, I am using a Intel D455 with a Jetson Nano. My end goal is to get a RGB and Depth stream in my PC (which is connected to the same WiFi network as of Jetson Nano). I also want to get the Distance to the objects(pixels) in the depth stream using: 
dpt_frame = pipe.wait_for_frames().get_depth_frame().as_depth_frame()
pixel_distance_in_meters = dpt_frame.get_distance(x,y)

For achieving this goal, I did following and failed:
1) Tried sending RGB and Depth frames using TCP/Sockets/Vidgear library/ Imagiz. I successfully got frames, but was not able to get depth data(to calculate distance)

2) Used ROS and published data using: "roslaunch realsense2_camera rs_camera.launch align_depth:=true" from my Jetson Nano (ROS Host) and connected my PC as ROS Master. I got frames in form of ROS messages. But it needs to be converted into cv2 format which needs cv_bridge. cv_bridge is not supported in python3 (which i am using).  

Please help me. I have already invested lot of days doing this.

MartyG-RealSense, 2021-02-05:
Hi @GauravKakade   The number of options available for a potential wi-fi based solution are fewer than a cable-based solution, but there are a few possibilities.

1.  The project in the link below used ROS in combination with RealSense and Pixhawk drone flight control software.

https://github.com/IntelRealSense/librealsense/issues/4392

2.  Intel's own open-source RealSense ethernet networking project can be adapted for wi-fi according to its documention.  the project uses Raspberry Pi boards by default but the documentation states that other compute boards can be used with minor modifications.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

3.  RealSense data can be streamed to a web browser to render the data with HTML5.  This is only wi-fi though in the sense that a computer could connect to internet-transferred data via wi-fi, rather than the RealSense data being transmitted over wi-fi directly.

https://github.com/IntelRealSense/librealsense/issues/6047

https://www.youtube.com/watch?v=eV5NIPKC_pc

MartyG-RealSense, 2021-02-11:
Hi @GauravKakade   Do you require further assistance with this case, please?   Thanks!

GauravKakade, 2021-02-11:
Hi @MartyG-RealSense Currently I am working on some other part of my project. Thank you for your suggestions. I will implement your suggestions once I am done with what I am currently doing (15 days later approx.)

MartyG-RealSense, 2021-02-11:
Thanks very much for the update @GauravKakade  - I will keep this case open for a further time period.  Good luck!

suthiwat-y, 2021-02-18:
Hi @GauravKakade I have been using a d435i with realsense2_caemra package and I did create a node which subscribed to the camera streams (both Color, Depth, and Point Cloud). I used 'ros_numpy' package to convert ROS messages into numpy arrays for examples; an RGB frame should be a numpy array of size (480, 640, 3) and I was able to use those data in my project.

Please checkout: [ros_numpy](https://github.com/eric-wieser/ros_numpy)
I hope this might help.

MartyG-RealSense, 2021-02-18:
Thanks very much @suthiwat-y for your help!

GauravKakade, 2021-02-25:
Hi @MartyG-RealSense I solved this problem by creating a ROS publisher inside a callback and included it in the program which sends RGB and Depth frames over wifi using Vigear and Imagiz libraries. Now I send the coordinates over a ROS Topic and this ROS callback takes these coordinates as an argument and publishes the distance/depth to these coordinates.

Also, Thanks for your support - @suthiwat-y and @MartyG-RealSense

MartyG-RealSense, 2021-02-25:
Greats news @GauravKakade - thanks very much for sharing the details of your method with the RealSense community!

closed 2021-02-25 by MartyG-RealSense

haquebd, 2023-03-05:
> Hi @MartyG-RealSense I solved this problem by creating a ROS publisher inside a callback and included it in the program which sends RGB and Depth frames over wifi using Vigear and Imagiz libraries. [...]

Hi Gaurav, would you share your git repo for testing?