realsenseai/realsense-ros #1678: D455: false IMU timestamp

chaotic-bruno, 2021-02-08:
ROS Melodic on Odroid XU4
RealSense ROS v2.2.21
Built with LibRealSense v2.41.0
Running with LibRealSense v2.41.0
Device FW version: 05.12.10.00

Timestamps of the IMU are corrupted.

1. Test: `/camera/motion_module/global_time_enabled: true`,  `/camera/stereo_module/global_time_enabled: true`, `initial_reset=true`
    
    ```
    $ rostopic delay /camera/imu
    average delay: -1612783706.639
            min: -1612783706.732s max: -1612783706.496s std dev: 0.04417s window: 50000

    $ rostopic delay /camera/aligned_depth_to_color/image_raw
    average delay: 0.245
            min: 0.175s max: 0.329s std dev: 0.03920s window: 32

    $ rostopic echo /camera/imu/header
    seq: 121905
    stamp: 
      secs: 3225572801
      nsecs: 931539536
    frame_id: "camera_imu_optical_frame"
    ```

2. Test: `/camera/motion_module/global_time_enabled: false`,  `/camera/stereo_module/global_time_enabled: false`, `initial_reset=true`
    
    ```
    $ rostopic delay /camera/imu
    average delay: -3.577
            min: -3.578s max: -3.572s std dev: 0.00074s window: 1189

    $ rostopic delay /camera/aligned_depth_to_color/image_raw
    average delay: 0.216
            min: 0.154s max: 0.270s std dev: 0.02799s window: 22

    $ rostopic echo /camera/imu/header
    seq: 1632
    stamp: 
      secs: 1612789275
      nsecs: 272668600
    frame_id: "camera_imu_optical_frame"
    ```

3. Test: `/camera/motion_module/global_time_enabled: true`,  `/camera/stereo_module/global_time_enabled: true`, `initial_reset=false`
    
    ```
     08/02 14:02:47,625 ERROR [2647651344] (source.cpp:130) Exception was thrown during user callback: Time is out of dual 32-bit range
     [ INFO] [1612789367.625701066]: RealSense Node Is Up!
     [ WARN] [1612789375.487373597]: Still waiting for data on topic /imu/data_raw...
    ```
   
    Message about `/imu/data_raw` ist from the IMU filter because no messages arrive...

4. Test: `/camera/motion_module/global_time_enabled: false`,  `/camera/stereo_module/global_time_enabled: false`, `initial_reset=false`
    
    ```
    $ rostopic delay /camera/imu
    average delay: -120.489
            min: -120.490s max: -120.483s std dev: 0.00074s window: 3234

    $ rostopic delay /camera/aligned_depth_to_color/image_raw
    average delay: 0.274
            min: 0.215s max: 0.343s std dev: 0.03172s window: 20

    $ rostopic echo /camera/imu/header
    seq: 23220
    stamp: 
      secs: 1612789666
      nsecs: 800081015
    frame_id: "camera_imu_optical_frame"
    ```

I need the IMU for orientation estimate with rtabmap - so I fuse the IMU Data with odometry in robot_localization. But as the IMU data is future dated, robot_localization tries to use old IMU data together with the current data of the other sensors.
Therefore the IMU is unusable in this way...

I also tested `<arg name="unite_imu_method" value="linear_interpolation"/>` and `<arg name="unite_imu_method" value="copy"/>` - this makes no difference.

MartyG-RealSense, 2021-02-08:
Hi @chaotic-bruno   I would recommend the launch settings of **Test 2** (global_time_enabled = false, initial_reset = true) based on the success of those settings in correcting IMU timestamp problems in the recent case https://github.com/IntelRealSense/realsense-ros/issues/1665

If you still have timestamps that are in the future after that, it may be worth trying the advice in the link below, such as using **ros::Time::now()** if you are making use of **waitForTransform** in your project.

https://answers.ros.org/question/331027/realsense-d435-lookup-would-require-extrapolation-into-the-future/?answer=331028#post-id-331028

chaotic-bruno, 2021-02-08:
I don't see what this has to do with the lookup of tf transforms.

I will probably do a filter before using the IMU data and just replace the header timestamp with current ROS time... not really accurate but will do the trick for me. Better than 4 seconds future dated messages from realsense-ros / librealsense...

chaotic-bruno, 2021-02-09:
Manually setting the message header to current system time works for my project.
After thinking about it, I think this is an upstream (librealsense) error - so I will close issue.
If somebody also has this problem maybe open the issue directly in [https://github.com/IntelRealSense/librealsense](https://github.com/IntelRealSense/librealsense)...

closed 2021-02-09 by chaotic-bruno