realsenseai/realsense-ros #1679: Enabling Occlusion Invalidation l515

apoorvcn47, 2021-02-08:
Hi,
I am using L515 with ROS2 Foxy driver. I believe Occlusion Invalidation is disabled in my case. I am not sure how to enable it
[image]

RealSenseSupport, 2021-02-24:
Hi @apoorvcn47 

I'm not familiar with specific setting for invalidate occlusion in ROS2.  Did you by chance figure out what was needed?

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras?_ga=2.58804653.1807786935.1613978068-1479303487.1578600965

apoorvcn47, 2021-02-24:
I did not yet

doronhi, 2021-02-25:
`ros2 param set /camera/camera l500_depth_sensor.invalidation_bypass true`

apoorvcn47, 2021-02-25:
Thanks That helped

closed 2021-02-25 by apoorvcn47