realsenseai/realsense-ros #1680: Depth registered pointcloud in ROS2

joelbudu, 2021-02-09:
Is it possible to get depth registered pointcloud in the ROS2 branches (eloquent, foxy)?
It seems the feature is not available but is present in ROS1 in [rs_rgbd.launch](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_rgbd.launch)

MartyG-RealSense, 2021-02-09:
Hi @joelbudu   My understanding is that a depth registered pointcloud is a **textured pointcloud** that has RGB aligned to depth.  

I researched your question extensively but could not find a clear example in the RealSense ROS2 wrapper for achieving a point cloud that was exactly described as depth-registered or aligned.

The documentation of the **Foxy** branch of the RealSense ROS2 wrapper suggests that applying a **pointcloud** filter will add a pointcloud topic **/camera/depth/color/points** that has texturing.

https://github.com/IntelRealSense/realsense-ros/tree/foxy#available-parameters

[image]

An example of a ROS2 launch instruction I found that includes a pointcloud filter definition is below:

**ros2 run realsense2_camera realsense2_camera --ros-args -p filters:=pointcloud**

doronhi, 2021-02-10:
Hi @joelbudu , Could you please explain what is the difference between "depth registered pointcloud" and the organized pointcloud available now in realsense2_camera (foxy)?

MartyG-RealSense, 2021-02-10:
Hi @doronhi   The old deprecated Intel ROS2 wrapper for Dashing supported a mode that it described as **Depth Aligned (i.e. Depth Registered) Point Cloud** that acted as my start point for trying to work out what 'depth registered' meant, and other sources seemed to concur that it referred to mapping RGB texturing to a depth point cloud.

https://github.com/intel/ros2_intel_realsense#visualize-depth-aligned-ie-depth-registered-point-cloud

Using the point cloud filter of the current ROS2 Foxy wrapper was the closest method that I could find to the old Dashing solution for producing a textured cloud in the current ROS2 wrapper, though you can likely suggest an alternative method.  :)

joelbudu, 2021-02-10:
@MartyG-RealSense Yes the old ROS2 wrapper published a Depth registered point cloud topic to a topic like /camera/aligned_depth_to_color/color/points.
The difference is that the Depth aligned pointcould uses the FOV of the colour sensor to create the pointcloud with a 1 to 1 mapping. 
@doronhi  The organized pointcloud available in foxy uses the FOV of the depth sensor and would only publish the points within the frame that correspond to a texture/colour point.

doronhi, 2021-02-10:
Hi @joelbudu , I think I see what you mean.
If I understand you correctly, you would like to use the aligned_depth_to_color image as a base for the ordered pointcloud.
This feature does not exist in the wrapper.

It is a little tricky to implement correctly in the current code structure since aligning the depth image to the color image happens only after the pointcloud is built and only locally in the "publishAlignedDepthToOthers" function.
As an immediate solution you could copy the section aligning the depth image [base_realsense_node.cpp, lines 1025-1044](https://github.com/IntelRealSense/realsense-ros/blob/eca337fd4dd4b1f7fd2cfd41290befbcb8daab5a/realsense2_camera/src/base_realsense_node.cpp#L1025) with `stream_type=RS2_STREAM_COLOR` and apply it before the filters are applied [line 1643](https://github.com/IntelRealSense/realsense-ros/blob/eca337fd4dd4b1f7fd2cfd41290befbcb8daab5a/realsense2_camera/src/base_realsense_node.cpp#L1643)
The reference is to the eloquent branch as I figured this is your current branch.
I didn't try it so I'm pretty sure some complications will arise.

joelbudu, 2021-02-11:
Dear @doronhi. Thanks for the pointer on this. I did indeed attempt this approach unsuccessfully yesterday, as you mentioned I was getting an error saying the alignment to the color frame couldn't be performed. I do not have a full understanding on the underlying SDK and how this library works in totality so it was quite challenging.

Please let me when you are able to do some work on this. Thanks

MartyG-RealSense, 2021-02-18:
Hi @doronhi   Could you assist @joelbudu with the question above please?  Thanks!

doronhi, 2021-02-18:
Hi @joelbudu 
The original decision to warp the color image to the depth image when creating a pointcloud and not the other way around came from the idea that it is more important to keep the depth information accurate than the color information.
I can think of 2 solutions to your problem.
1. Create the pointcloud from the warped depth image and the original color image.
2. Publish a "color-to-depth-aligned-image". You could then do all the color-image work on the warped color image and have it aligned with the pointcloud.
Neither option is currently in the work plan. The second option seems more straightforward from the wrapper's perspective.

Please let me know if it seems that it would benefit you and I'll see how and if I promote any of them.

joelbudu, 2021-02-21:
Thanks for the response @doronhi.

First of all, the feature I'm requesting is available in the ROS1 branches of this repository via [rs_rgbd.launch](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_rgbd.launch). The only issue is that it isn't available in the ROS2 versions. It seems to be using some form of nodelet using the underlying realsense library. Is it possible to have this in ROS2?

It would definitely be of great benefit if the feature is available in ROS2 for this repo, in terms of the solutions you mentioned, option 1 would be more preferable since that is consistent with the current and deprecated library. 

Thanks for your assistance.

doronhi, 2021-02-23:
Hi @joelbudu 
[rgbd_launch](https://github.com/ros-drivers/rgbd_launch) is not a RealSense package. You could apply in its repository for ROS2 support but I have no way of helping in that direction.
As for realsense2_camera publishing an ordered pointcloud aligned with the color image, I have put it on the list for review. There is, as for now, no timeline defined. It is tracked internally as "DSO-16639"

joelbudu, 2021-02-27:
@doronhi Thank you very much for working on this.

charlielito, 2021-03-05:
I think this feature is very important. My team moved from Dashing to Foxy because Intel said that this was the official repo with full support and features, so it is a little sad that this isn't implemented yet :/

MartyG-RealSense, 2021-03-08:
Thanks very much @charlielito for sharing your method with the RealSense ROS community!

doronhi, 2021-04-20:
Please check out #1806. It targets ROS2.

joelbudu, 2021-04-20:
@doronhi Many thanks for working on #1806. Looks really good to me. When is it likely to be merged in please?

doronhi, 2021-04-21:
Thanks.
The plan is to merge it soon, before the next release - 1-2 weeks tops.

MartyG-RealSense, 2021-06-20:
As @charlielito referenced, the requested feature for ROS2 was added in ROS2 wrapper **3.1.6**.  I will therefore close this case.

[image]

closed 2021-06-20 by MartyG-RealSense