realsenseai/realsense-ros #1681: No easy, systematic way to be notified if the firmware version is not desirable

130s, 2021-02-09:
Although it looks like firmware handling might be a scope of `librealsense`, I'm still posting on `realsense-ros` as that's the operational interface in my usecase.

## Environment

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415, L515 } | 
| Firmware Version                   | (5_12_8_200) | 
| Operating System & Version |   {Linux (Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0-96-generic                                        | 
| Platform                                 | Industrial PC |
| SDK Version                            |  { 2.39.0 }                          | 
| Language                            |  {C++/pcl }                          | 
| Segment			|  {Robot }                          |
| ROS wrapper version | Derived custom version based on 2.2.17 ([Link to our fork](https://github.com/plusone-robotics/realsense/releases/tag/100.2217.1)) |

## Issue Description

I've often seen when Realsense' performance is not what we expected to see that the firmware version was wrong. It is very difficult to notice (it's so easy to waste a few hours, with multiple people. I've seen that happened a few times..).

## Responding to triage

>    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
>    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

- I did look, but couldn't find something that is obviously talking about similar problems.
- PR https://github.com/IntelRealSense/realsense-ros/pull/458, which was closed without any discussion, seems to suggest an interesting feature.

## Possible approach
- If `librealsense` offers a functionality to check firmware versions, that's great.
- That said, printing a warning/error in the log is great, but a passive functionality like that may NOT be sufficient for various usecases. I'd say something a user can actively query will be convenient.
   - I am thinking to add a troubleshooting feature (using e.g. [roswtf plugin](http://wiki.ros.org/roswtf/Plugins) framework) that warns when the firmware doesn't meet the requirement.
      - Required version of firmware needs to be somehow set. Maybe add it as a ROS Parameter.
         - Required version can be minimum viable version.
      - If a firmware version checking mechanism already exists in the upstream like librealsense, this troubleshooting feature can simply call that upstream feature.
      - Being able to access such a feature via `ROS` can free the application development from Linux (or any OS) commands (can be crucial in production systems), hence I want that in `realsense-ros`.

## How much impact would the issue have?
I'd say this issue should not be underestimated, as I've seen it happened multiple times as I mentioned above.

CC @abhijitmajumdar

doronhi, 2021-02-10:
I'm not sure what feature you would like to have.
Are you suggesting a service that returns the firmware version?

How about a string parameter: force_fw_version = ["no" | "recommended" | "x.y.z.w"]?
"no" - don't check the firmware version. (current state)
"recommended" - check firmware version against the one recommended for that librealsense2 version.
"x.y.z.w" - check the firmware version against the one defined by the user.

I couldn't understand from [roswtf plugin](http://wiki.ros.org/roswtf/Plugins) what a plugin is. Could you please explain?

130s, 2021-02-10:
I just wanted to make sure the problem I raised is a valid one among the Realsense users, as it's possible that I might be just missing existing solution. But reading your comment I assume it's a valid :)

The I/O idea you posted looks good. I was envisioning something like that.

>  "recommended" - check firmware version against the one recommended for that librealsense2 version.

Is this already implemented on librealsense side? If so that's great.

I have no preference in how it's impelemented. I mentioned `roswtf` as one possible framework to build such a feature on. It's a troubleshooting command on `shell` that does dozens of checking and print the result on `shell` (no API available ATM, AFAIK. I tried but didn't get to finish https://github.com/ros/ros_comm/pull/1430). Using roswtf will only require `roswtf` to be added as a new dependency, which is minimal IMO and wouldn't be an issue as `realsense2_camera` already depends on a number of "ros package"s.
ROS Service might be actually better, more versatile. It can be called over network and distribute the result over (ROS master) network. And integrating ROS Service into other tools like `roswtf` is very easy and standardized.

I/someone in my team can think about contributing a PR although no guarantee atm.

RealSenseSupport, 2021-02-19:
@130s Yes, "recommended" for recommended firmware version has already been implemented on librealsense side. Also please feel free to submit your PR and thanks for your contribution!

RealSenseSupport, 2021-03-15:
@130s Did you get chance to submit your PR? And any other questions about this ticket? Looking forward to your update. Thanks!

RealSenseSupport, 2021-03-31:
@130s Any other questions from your side? Thanks!

RealSenseSupport, 2021-04-06:
@130s Sorry that we didn't get your update for weeks, will close this issue at this point. Please feel free to create another new one if you still have any other issue or questions. Thanks!

closed 2021-04-06 by RealSenseSupport