realsenseai/realsense-ros #1691: D435I publishes no imu data

kiwi-sherbet, 2021-02-12:
Hi there, I have a question about using `rs_camare.launch` D435I.
I can check that the package publishes depth data but no imu and gyro data. Does it require calibration? I thought it has in-factory calibration data.

If I need calibration, how can I calibrate it? I tried to use `librealsense/tools/rs-imu-calibration` but it fails to write calibration data to the EEPROM. It says,

```
RuntimeError: failed to set power state
```

Can I ask for advice to clear the issue?

MartyG-RealSense, 2021-02-12:
Hi @kiwi-sherbet  The IMU topics are disabled by default in the RealSense ROS wrapper.  You can enable them by adding the instructions below to the end of your roslaunch instruction:

**enable_gyro:=true enable_accel:=true**

kiwi-sherbet, 2021-02-12:
Thank you for your help! It works! :)

closed 2021-02-12 by kiwi-sherbet

MartyG-RealSense, 2021-02-13:
Great to hear @kiwi-sherbet  - thanks for the update!