realsenseai/realsense-ros #1692: How to change depth field of view of d435i to maximize horiztonal field of view?

alessandromorra, 2021-02-12:
Hi!

I have a question using the pointcloud launch file: `roslaunch realsense2_camera demo_pointcloud.launch`, see here: https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/demo_pointcloud.launch

I tried to change the field of view of the pointcloud by changing the following parameters:

```
      <arg name="depth_height"      value="360"/> 
      <arg name="color_height"      value="360"/>
```

In order to achieve a field of view of 86 x 57 degrees. My goal is to maximize the horizontal field of view. (datasheet: https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/Intel-RealSense-D400-Series-Datasheet.pdf, see p.57 for field of view)

When measuring the field of view of the pointcloud in rviz the angles are more like 74 degrees in horiztonal, and seem to not be affected by these parameters. Can someone help how to change this there?

When testing it with the `realsense-viewer` tool, changing the field of view works perfectly with my camera.

Many thanks for the support!

PS I also tried width x depth with the following parameter pair without success to maximize my the horizontal dimension: 848 x 480, 640 x 360, 1280 x 720

MartyG-RealSense, 2021-02-12:
Hi @alessandromorra   The sensor field of view size on RealSense cameras is **fixed**, with a +- 3 degree margin of difference to account for variances in the sensors that occur during the manufacturing process.

The proper way to widen the horizontal field of view would be to add another camera and position it so that its field of view overlaps with the neigboring camera's field of view.  For example, in January 2018 Intel demonstrated arranging four D435 cameras in a horizontal arc to create a 180 degree horizontal field of view.

https://www.intelrealsense.com/intel-realsense-volumetric-capture/

Intel have published a ROS tutorial for 'stitching' together point clouds from two cameras attached to the same computer into a single combined point cloud.

https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras

alessandromorra, 2021-02-14:
Hi @MartyG-RealSense 

Many thanks for your fast reply! However, this was not exactly what I was trying to ask, let me try again with some pictures, they both capture the same scene.

Picture 1: realsenseviewer with d435i with 640 x 360 resolution output.
[image]

Picture 2: rviz output of rosnode with d435i with 640 x 360 resolution output.
[image]

I have the feeling that some depth is cropped away in the ROS version. If we compare the right sides of the depth images, it seems that picture 1, has more information than picture 2, although the scene has not changed in between and they should have the same configuration.

As reference I tool a picture from the setup with my phone (my intel d435i is upside down, hence right in this picture is left and vice versa): 
[image]

Many thanks for your support.

MartyG-RealSense, 2021-02-14:
Image 1 (the RealSense Viewer) looks like a depth-only point cloud, whilst Image 2 (RViz) looks like a point cloud where depth is aligned to an RGB color image.

If Image 2 is a depth to color alignment: in RealSense, the stream that is being aligned is transformed to the field of view size of the source that is being aligned **to**.  So if depth is being aligned to color then the depth data is aligned to the color sensor's field of view.  

On the D435 and D435i, the color sensor's field of view size is smaller than the depth sensor.  So when aligning depth to color, the outer regions of the depth image can be excluded from the aligned image.

These principles are described in the opening paragraph of the documentation for the **rs-align** librealsense SDK example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

alessandromorra, 2021-02-14:
Thanks @MartyG-RealSense !

Is there a ROS parameter to deactivate this alignment directly within the launch file? Such that the point cloud is not cropped to the color sensor's field of view?

MartyG-RealSense, 2021-02-14:
The ROS wrapper documentation's information about pointcloud generation states that you can have a full depth to pointcloud, coloring the regions beyond the texture with zeros, by setting **allow_no_texture_points** to true.  You could try adding the instruction below to the end of the roslaunch instruction:

**allow_no_texture_points:=true**

alessandromorra, 2021-02-14:
@MartyG-RealSense 
Awesome thank you so much for the support, this is exactly what I have wanted.

MartyG-RealSense, 2021-02-14:
You're very welcome.  I'm pleased that the information was helpful to you.  :)

MartyG-RealSense, 2021-02-20:
Case closed due to solution provided and no further comments received.

closed 2021-02-20 by MartyG-RealSense