realsenseai/realsense-ros #1693: L515 pointcloud

senoa95, 2021-02-12:
I am working on the L515 in Ros. I am able to launch the node and am able to visualize the depth image using RQT. I am not able to get any messages when I enable the poincloud filters despite the topic being published. Can you provide assistance with this? FYI I have gone through the thread so far and tried all of the solutions including the camera reset and reduced FPS arguments as directed in https://github.com/IntelRealSense/realsense-ros/issues/1510.

senoa95, 2021-02-12:
After searching around for some time, I found this page that shows the ROS support for the L515. Does this mean that L515 cannot inherently produce pointclouds in ROS?
[image]

RealSenseSupport, 2021-02-12:
Hi @senoa95 

The information outlined on the intelrealsense.com site is correct and updated for all the cameras support for ROS.

senoa95, 2021-02-12:
Can you confirm that there is no inherent support for poincloud generated directly by the realsense package for the L515?

doronhi, 2021-02-14:
The option enable_pointcloud works just the same for the D400 series and for L515.
This description here: https://dev.intelrealsense.com/docs/ros-wrapper#section-l-500-lidar-depth-cameras should be fixed.
If you can visualize the depth topic and the color topic and the pointcloud topic (/camera/depth/color/points) is being published then we should search for the reason why you are not getting it.
Could you describe what version of ROS you are using and copy the log of realsense2_camera?
Also, what command do you run that does not receive the messages?

senoa95, 2021-02-14:
@doronhi 

Thank you for updating the documentation. I am using ROS kinetic on ubuntu 16.04. I installed the pre-built realsense2 package using apt-get. I run the package using both `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud` as well as modifying the launch file to enable pointclouds by default (I have attached the launch file). I have also attached the log file output. Both files are in the zip file.
[log_and_launch.zip](https://github.com/IntelRealSense/realsense-ros/files/5978286/log_and_launch.zip)

senoa95, 2021-02-14:
This shows the topics being published when I launched the node using `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`.
[image]

doronhi, 2021-02-15:
The issue here is that color images are not available to build the pointcloud. See the following line in the log file:
`[ WARN] [1613328673.323777027]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: None`
This is fixed in #1614. For now, set `color_width:=1280 color_height:=720`

senoa95, 2021-02-15:
Yes this fixed it. Thank you. I am now able to get both the RGB and the registered pointclouds.

closed 2021-02-15 by senoa95