realsenseai/realsense-ros #1694: D435I Orientation message

kiwi-sherbet, 2021-02-12:
I want to use D435I's 9-DOF  IMU data, but it does not seem to publish orientation values properly. 

I assumed that I need to calibrate the IMU to get orientation data, but when I tied to use librealsense/tools/rs-imu-calibration, it fails to write calibration data to the EEPROM. It says,
```
RuntimeError: failed to set power state
```
Do I need to calibrate the IMU to get data? If this is caused by issues other than calibration, how can I fix it? The following are ROS messages what I am getting. 

Thank you!

```
header: 
  seq: 4650
  stamp: 
    secs: 3226342211
    nsecs: 881966591
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: -0.003490658476948738
  y: 0.0
  z: -0.001745329238474369
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.6561354215697935
  y: -9.24352356840411
  z: 0.9595328294956944
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
---
```

MartyG-RealSense, 2021-02-13:
Hi @kiwi-sherbet   After you got the IMU topic publishing working in https://github.com/IntelRealSense/realsense-ros/issues/1691 do you still require assistance with this case about IMU calibration, please?

kiwi-sherbet, 2021-02-13:
Hi @MartyG-RealSense Yeah, I cannot get orientation information from ROS topics.

kiwi-sherbet, 2021-02-16:
Oh, I got it. I need to use additional packages, such as `imu_filter_madgwick` to get orientation data. Thank you!

closed 2021-02-17 by kiwi-sherbet