realsenseai/realsense-ros #1697: Subscribers dont receive messages

cf95, 2021-02-14:
Hey,

I'm currently trying to setup my D455 with foxy and I'm running into the problem, that my subscribers don't receive any messages.
Starting the ros node with "ros2 launch realsense2_camera rs_launch.py initial_reset:=true" leads to the following (for me correctly looking) output 
```bash
[INFO] [launch]: All log files can be found below /home/psaf/.ros/log/2021-02-14-18-44-59-535569-psaf-NUC8i7BEH-3652
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [3654]
[realsense2_camera_node-1] [INFO] [1613328299.734336383] [RealSenseCameraNode]: RealSense ROS v3.1.3
[realsense2_camera_node-1] [INFO] [1613328299.734382042] [RealSenseCameraNode]: Built with LibRealSense v2.34.0
[realsense2_camera_node-1] [INFO] [1613328299.734390168] [RealSenseCameraNode]: Running with LibRealSense v2.41.0
[realsense2_camera_node-1] [WARN] [1613328299.734395705] [RealSenseCameraNode]: ***************************************************
[realsense2_camera_node-1] [WARN] [1613328299.734405808] [RealSenseCameraNode]: ** running with a different librealsense version **
[realsense2_camera_node-1] [WARN] [1613328299.734413841] [RealSenseCameraNode]: ** than the one the wrapper was compiled with!   **
[realsense2_camera_node-1] [WARN] [1613328299.734425634] [RealSenseCameraNode]: ***************************************************
[realsense2_camera_node-1] [INFO] [1613328299.743011905] [RealSenseCameraNode]: Device with serial number 046122250373 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1613328299.743026126] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1613328299.743031194] [RealSenseCameraNode]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1613328299.743206132] [RealSenseCameraNode]: Device with port number 2-4 was found.
[realsense2_camera_node-1] [INFO] [1613328299.743213923] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1613328299.743218501] [RealSenseCameraNode]: Resetting device...
[realsense2_camera_node-1] [INFO] [1613328305.862385690] [RealSenseCameraNode]: Device with serial number 046122250373 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1613328305.862420515] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1613328305.862429697] [RealSenseCameraNode]: Device with name Intel RealSense D455 was found.
[realsense2_camera_node-1] [INFO] [1613328305.862690695] [RealSenseCameraNode]: Device with port number 2-4 was found.
[realsense2_camera_node-1] [INFO] [1613328305.862704076] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1613328305.864117315] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1613328305.864174389] [RealSenseCameraNode]: 707
[realsense2_camera_node-1] [INFO] [1613328305.866402975] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1613328305.866424106] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1613328305.866441626] [RealSenseCameraNode]: Device Name: Intel RealSense D455
[realsense2_camera_node-1] [INFO] [1613328305.866452253] [RealSenseCameraNode]: Device Serial No: 046122250373
[realsense2_camera_node-1] [INFO] [1613328305.866460869] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1613328305.866469916] [RealSenseCameraNode]: Device FW version: 05.12.06.00
[realsense2_camera_node-1] [INFO] [1613328305.866477915] [RealSenseCameraNode]: Device Product ID: 0x0B5C
[realsense2_camera_node-1] [INFO] [1613328305.866485551] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1613328305.866493055] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1613328305.866500627] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1613328305.866519413] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1613328305.870187885] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1]  14/02 18:45:05,870 WARNING [140342293726976] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1613328305.879889460] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1613328305.880184675] [RealSenseCameraNode]: Motion Module was found.
[realsense2_camera_node-1] [INFO] [1613328305.880210369] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1613328305.880221506] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1613328305.880229783] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1613328305.880240868] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1613328305.880251347] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1613328305.880261368] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1613328305.880273575] [RealSenseCameraNode]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1613328305.880283113] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1613328305.946718197] [RealSenseCameraNode]: Failed to set parameter:rgb_camera.power_line_frequency = 3[0, 2]
[realsense2_camera_node-1] parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1613328305.947281884] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1613328305.948082885] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1613328305.948446346] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[realsense2_camera_node-1] [INFO] [1613328305.948869173] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[realsense2_camera_node-1] [INFO] [1613328305.949866615] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1613328305.949892867] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1613328305.955311530] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1] [INFO] [1613328306.068335751] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1613328306.069838119] [RealSenseCameraNode]: RealSense Node Is Up!
```
Looking at "ros2 topic echo /camera/color/image_raw" also gives me some output of the images, but if I'm trying to look at them with i.e. rqt_image_view or subscribing to the topic with any other node, I'm not receiving any messages. Do you have any idea, what the issue is here?

Regrads
Chris

doronhi, 2021-02-15:
Hi Chris,
The first thing I noticed is the following 2 lines from your log:
```
[realsense2_camera_node-1] [INFO] [1613328299.734382042] [RealSenseCameraNode]: Built with LibRealSense v2.34.0
[realsense2_camera_node-1] [INFO] [1613328299.734390168] [RealSenseCameraNode]: Running with LibRealSense v2.41.0
```
This can't be healthy and you might want to look at it.
However, I believe the issue is the QoS: rqt_image_view uses a different QoS than the default for realsense2_camera. You can view the images using RViz.
The latest version in the foxy branch already includes an option for selecting the QoS you want, thanks to @gurbain.
You may change the QoS realsense2_camera using the parameters `color_qos`, `depth_qos` etc. Available values are the following strings: SYSTEM_DEFAULT, PARAMETER_EVENTS, SERVICES_DEFAULT, PARAMETERS, DEFAULT, SENSOR_DATA.
Try SYSTEM_DEFAULT.

MartyG-RealSense, 2021-02-15:
@doronhi  Thanks very much!

Hi @cf95   I agree with the advice of @doronhi 

Your librealsense log lines suggest that your RealSense ROS wrapper may have been built on librealsense SDK version **2.34.0** and then librealsense was upgraded to version **2.41.0**.  The RealSense ROS wrapper should be re-built after the librealsense version is changed.  Can you confirm whether you re-bult the ROS wrapper after installing librealsense 2.41.0 please?

cf95, 2021-02-15:
Hi @doronhi @MartyG-RealSense,
thanks for your answer, yes i can confirm rebuilding the wrapper (also removed the build, install and log directory to have a clean rebuild).
I just installed the new librealsense version 2.42.0 and rebuilt the wrapper, but it changed just the "running with" version and not the "built"-version, so i guess the 2.34 version is somewhere lying within my system, that the compiler prefers. I'll figure this out and will report back.
I tested displaying the images with RViz, but the image panel just says "no image"

MartyG-RealSense, 2021-02-15:
Hi @cf95  It is important to match the RealSense ROS wrapper version as closely as possible with the version of librealsense that the release notes of a particular ROS wrapper version recommend.  As 2.42.0 was only released yesterday at the time of writing this, **2.41.0** is the recommended librealsense version for ROS Foxy wrapper **3.1.3**

If you built the ROS wrapper using the catkin workspace, you could try deleting the workspace and creating a new one to rebuild the wrapper in.  This may help to deal with mis-detection of a previous librealsense version.

cf95, 2021-02-16:
So it took me some time to downgrade librealsense2 again (-dev got uninstalled and i didnt recognize it immediatly). The old version 2.34.0 resulted from the old ros2-foxy-librealsense package, that was still installed.
Long story short, i have the same versions for compiling and running, and starting with color_qos:=SYSTEM_DEFAULT results in rqt_image_view showing the video feed like i'm used to it. My own node still doesnt receive images, but i guess thats on my end now, thanks @MartyG-RealSense @doronhi for your fast support.

MartyG-RealSense, 2021-02-16:
You are very welcome @cf95  - thanks very much for the update about your progress   :)

MartyG-RealSense, 2021-02-22:
Hi @cf95  Do you require further assistance with this case, please?  Thanks!

cf95, 2021-02-22:
hey @MartyG-RealSense , thanks for asking, i'm good, thanks, the last part was to match the qos in my node.
am i supposed to close this issue? is this why you are asking?

MartyG-RealSense, 2021-02-22:
If you are satisfied that all of the problems related to RealSense ROS have been solved then you can close the issue with the **Close Issue** button under the comment box.

If you feel that there are remaining issues that you want to explore further though then I am happy to do so.

cf95, 2021-02-22:
All problems solved, thanks for your assistance again!

closed 2021-02-22 by cf95

MartyG-RealSense, 2021-02-22:
Thanks very much @cf95   :)