realsenseai/realsense-ros #1699: Heavier CPU usage compared to "old" ROS2 RealSense driver

AndreasAZiegler, 2021-02-15:
I'm on the way to move from the "old" ROS2 RealSense driver (https://github.com/intel/ros2_intel_realsense/tree/refactor) to this one. I have a ROS2 node that subscribes to the depth map and the color image of two RealSense cameras/nodes. I use a ```message_filters::TimeSynchronizer<sensor_msgs::msg::Image, sensor_msgs::msg::Image>``` to synchronize the color image and the depth map. After I saw that my ROS2 node complained that it didn't receive images (color and depth) for some time, I had a look at the CPU usage. I observer that this new ROS2 driver uses quite a bit more of CPU on the UP2 board I'm using.

```htop``` on this ("new") ROS2 RealSense driver
[image]

```htop``` on the ("old") ROS2 RealSense driver
[image]

I disabled some topics on the "new" ROS2 RealSense driver. This is ```ros2 topic list```
[image]

The same ```ros2 topic list``` with the "old" ROS2 RealSense driver
[image]

Do you know, where this performance difference is coming from? On a rather constrained UP2 board this makes a difference. One thought I had is to put all the tree nodes (2x RealSense and 1x my custom node) into a Component in a launch file (I didn't manage to do this yet) so that inter-process communication could be possible? But the last time I checked ```message_filters::TimeSynchronizer``` did not support int-er-process communication (https://github.com/ros2/message_filters/issues/31). Do you have any ideas/hints?

doronhi, 2021-02-16:
As the UP2 board is not available for me at the moment I used a much more powerful VM to compare. 
I also used RViz to show both color and depth images.
I also used `top` instead of `htop` as it was easier for me when the usage is summarized in one line.
I saw the opposite of what you describe:
Running with "old" wrapper:
```
  37681 doronhi   20   0 1290556 171128  65680 S 136.9   2.1   1:49.11 realsense_ros2_                                                                                    
  37716 doronhi   20   0 1492332 178340 119756 S  22.9   2.2   0:14.88 rviz2        
```

Running with "new" wrapper:
```
  37716 doronhi   20   0 1491616 175420 119804 S  21.9   2.2   0:28.72 rviz2                                                                                              
  37771 doronhi   20   0 1706164 104852  68412 S  14.6   1.3   0:14.69 realsense2_came  
```
Could it be that you are not using the same build of librealsense2 in your comparison?

AndreasAZiegler, 2021-02-16:
In the comparison I did, I did not change the librealsense2. I was using version ```2.41.0-0~realsense0.3962```. I now updated it (```sudo apt update; sudo apt upgrade```) and realized that the new version is ```(2.42.0-0~realsense0.4059``` while the recommended version for ROS2 eloquent is ```v2.40.0``` as stated [here](https://github.com/IntelRealSense/realsense-ros/tree/eloquent). Do I have to downgrade librealsense2 to the recommended version? If so, where do I get this "older" librealsense2 version?

doronhi, 2021-02-16:
Try this: `sudo apt-cache madison librealsense2`
Although I don't think it would matter much if you rebuilt both realsense2_camera and realsense_ros2_camera with librealsense2 version 2.42.0

AndreasAZiegler, 2021-02-16:
```
sudo apt-cache madison librealsense2
[sudo] password for mtr: 
librealsense2 | 2.42.0-0~realsense0.4059 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.41.0-0~realsense0.3962 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.40.0-0~realsense0.3814 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.39.0-0~realsense0.3571 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.39.0-0~realsense0.3569 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.38.1-0~realsense0.3378 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.38.1-0~realsense0.3376 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.36.0-0~realsense0.3169 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.2-0~realsense0.3083 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.2-0~realsense0.3077 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.2-0~realsense0.3075 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.2-0~realsense0.2994 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.2-0~realsense0.2932 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2750 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2740 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2738 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2734 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2729 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2723 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2711 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2709 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2707 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2701 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2689 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2676 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2672 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2670 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2666 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2664 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2662 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2660 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2650 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2646 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.35.0-0~realsense0.2644 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.34.0-0~realsense0.2251 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.34.0-0~realsense0.2249 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.34.0-0~realsense0.2207 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.34.0-0~realsense0.2205 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.34.0-0~realsense0.2187 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.34.0-0~realsense0.2185 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.34.0-0~realsense0.2183 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
[... cut ...]
librealsense2 | 2.16.2-0~realsense0.73 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.16.2-0~realsense0.71 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.16.1-0~realsense0.51 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
librealsense2 | 2.16.1-0~realsense0.42 | http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main amd64 Packages
```

doronhi, 2021-02-16:
`sudo apt-cache madison librealsense2` just gives you the list of available versions of the package.
to install use this:
`sudo apt-get install librealsense2=2.41.0-0~realsense0.3962`

AndreasAZiegler, 2021-02-16:
I now installed ```librealsense2=2.40.0-0~realsense0.3814``` (the official version for ROS2 Eloquent), built the whole ROS2 workspace again and get this results (this time also with ```top```)

Old wrapper:
```
20993 mtr       20   0 1819688  84180  53976 S  32.6  1.1   0:00.43 realsense_node           
20991 mtr       20   0 1262584  81896  53224 S  29.5  1.0   0:00.39 realsense_node
```
New wrapper:
```
27384 mtr       20   0 1418044 149928  53436 S  84.7  1.9   0:15.23 realsense2_came          
27370 mtr       20   0 1377060 147256  53516 S  73.8  1.9   0:20.24 realsense2_came
```

AndreasAZiegler, 2021-02-16:
After posting, I realized, that I choose a higher resolution in the configs for the new wrapper. This makes the difference.

closed 2021-02-16 by AndreasAZiegler

doronhi, 2021-02-16:
That's a relief :)