realsenseai/realsense-ros #1703: Low frame rate of Intel Realsense D435i on Nvidia Jetson Xavier

charlielito, 2021-02-17:
Development environment is as follows:
* Platform: Nvidia Jetson Xavier AGX
* OS: Ubuntu 20.04
* ROS : ROS2 foxy
* Set up: I am connected to the Jetson via Ethernet (and sharing Internet) LAN. The camera is connected via the USB C port from the Jetson.

I am configuring the node to publish all image topics to 15 FPS, nonetheless, when subscribed to any image topic (with `ros2 topic hz <topic>` the reported FPS are very low, less than 5 FPS. The QoS is the default from the launch file which is `SENSOR_DATA`.

What is even more strange, is that when I subscribe to any topic from __my machine__ (I have to set up the same `ROS_DOMAIN_ID` that is running on the Jetson), the FPS of the topic immediately goes to the supposed FPS (in this case 15).

Here are the logs of the ros2 launch:
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2021-02-16-23-46-51-543180-0a57d74-213
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [215]
[realsense2_camera_node-1] [INFO][1613519212.845032542]: RealSense ROS v3.1.3
[realsense2_camera_node-1] [INFO][1613519212.845334254]: Built with LibRealSense v2.41.0
[realsense2_camera_node-1] [INFO][1613519212.845404242]: Running with LibRealSense v2.41.0
[realsense2_camera_node-1] [INFO][1613519213.198150216]: Device with serial number 035422072739 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO][1613519213.198519324]: Device with physical ID 2-1-26 was found.
[realsense2_camera_node-1] [INFO][1613519213.198682692]: Device with name Intel RealSense D435I was found.
[realsense2_camera_node-1] [INFO][1613519213.201760904]: Device with port number 2-1 was found.
[realsense2_camera_node-1] [INFO][1613519213.201936433]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO][1613519213.211553584]: getParameters...
[realsense2_camera_node-1] [INFO][1613519213.212007016]: 707
[realsense2_camera_node-1] [INFO][1613519213.230694153]: setupDevice...
[realsense2_camera_node-1] [INFO][1613519213.230815567]: JSON file is not provided
[realsense2_camera_node-1] [INFO][1613519213.230923573]: Device Name: Intel RealSense D435I
[realsense2_camera_node-1] [INFO][1613519213.230973656]: Device Serial No: 035422072739
[realsense2_camera_node-1] [INFO][1613519213.231016666]: Device physical port: 2-1-26
[realsense2_camera_node-1] [INFO][1613519213.231109727]: Device FW version: 05.12.09.00
[realsense2_camera_node-1] [INFO][1613519213.231170402]: Device Product ID: 0x0B3A
[realsense2_camera_node-1] [INFO][1613519213.231211460]: Enable PointCloud: On
[realsense2_camera_node-1] [INFO][1613519213.231488563]: Align Depth: On
[realsense2_camera_node-1] [INFO][1613519213.231533718]: Sync Mode: On
[realsense2_camera_node-1] [INFO][1613519213.231689150]: Device Sensors: 
[realsense2_camera_node-1] [INFO][1613519213.282489577]: Stereo Module was found.
[realsense2_camera_node-1] [INFO][1613519213.341078002]: RGB Camera was found.
[realsense2_camera_node-1] [INFO][1613519213.342433946]: Motion Module was found.
[realsense2_camera_node-1] [INFO][1613519213.342581250]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO][1613519213.342641093]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO][1613519213.342677415]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO][1613519213.342712905]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO][1613519213.342751115]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO][1613519213.342787949]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO][1613519213.342843088]: Add Filter: pointcloud
[realsense2_camera_node-1] [INFO][1613519213.348837518]: num_filters: 1
[realsense2_camera_node-1] [INFO][1613519213.353477765]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO][1613519213.972039941]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO][1613519213.974676370]: depth stream is enabled - width: 640, height: 360, fps: 15, Format: Z16
[realsense2_camera_node-1] [INFO][1613519213.975818254]: infra1 stream is enabled - width: 640, height: 360, fps: 15, Format: Y8
[realsense2_camera_node-1] [INFO][1613519213.976994125]: infra2 stream is enabled - width: 640, height: 360, fps: 15, Format: Y8
[realsense2_camera_node-1] [INFO][1613519213.980674864]: color stream is enabled - width: 640, height: 360, fps: 15, Format: RGB8
[realsense2_camera_node-1] [INFO][1613519213.987368372]: setupPublishers...
[realsense2_camera_node-1] [INFO][1613519214.026142560]: setupStreams...
[realsense2_camera_node-1]  16/02 23:46:54,076 WARNING [547788020128] (ds5-motion.cpp:477) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[realsense2_camera_node-1]  16/02 23:46:54,246 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1] [INFO][1613519214.297631160]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO][1613519214.323409138]: RealSense Node Is Up!
[realsense2_camera_node-1]  16/02 23:46:54,447 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:54,504 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:54,556 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:54,609 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:54,661 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:55,481 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1] [ERROR][1613519215.496073427]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[realsense2_camera_node-1] [ERROR][1613519215.519229441]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[realsense2_camera_node-1]  16/02 23:46:55,537 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:55,699 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:55,750 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:55,902 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[realsense2_camera_node-1]  16/02 23:46:55,954 WARNING [547762842016] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```

And here the command used to get the FPS:
```
root@0a57d74:/usr/src/app# ros2 topic hz /camera/color/image_raw
average rate: 2.959
        min: 0.206s max: 0.471s std dev: 0.10182s window: 4
average rate: 1.985
        min: 0.134s max: 1.537s std dev: 0.47530s window: 6
average rate: 1.377
        min: 0.134s max: 2.911s std dev: 0.87200s window: 9
average rate: 1.625
        min: 0.068s max: 2.911s std dev: 0.76705s window: 13
average rate: 1.643
        min: 0.068s max: 2.911s std dev: 0.70126s window: 16
average rate: 2.036
        min: 0.066s max: 2.911s std dev: 0.63147s window: 22
average rate: 1.966
        min: 0.066s max: 2.911s std dev: 0.61649s window: 24
average rate: 1.939
        min: 0.066s max: 2.911s std dev: 0.59740s window: 26
average rate: 2.211
        min: 0.066s max: 2.911s std dev: 0.55606s window: 32
average rate: 2.262
        min: 0.066s max: 2.911s std dev: 0.53369s window: 35
average rate: 2.176
        min: 0.066s max: 2.911s std dev: 0.53613s window: 36
average rate: 2.280
        min: 0.066s max: 2.911s std dev: 0.51581s window: 40
```

charlielito, 2021-02-17:
After changing the DDS to [cyclone](https://github.com/ros2/rmw_cyclonedds/tree/foxy) and setting more memory for buffering, we were able to get the desired FPS in the jetson.  Nonetheless, this does not answer why it doesn't work using the default DDS or why when you connect from an external source it works like charm.

RealSenseSupport, 2021-02-18:
It's good to hear that the low frame rate issue is resolved by switching DDS to cyclone. 
Thanks for the update and please feel free to submit new tickets for any other question or problem. 

If we don’t hear from you in 7 days, this issue will be closed. Thank you.

closed 2021-02-19 by charlielito

ldg810, 2021-03-03:
> After changing the DDS to [cyclone](https://github.com/ros2/rmw_cyclonedds/tree/foxy) and setting more memory for buffering, we were able to get the desired FPS in the jetson. Nonetheless, this does not answer why it doesn't work using the default DDS or why when you connect to a topic from an external source/machine it works like charm.

I also tried to use D435i in Jetson Nano Dev kit, and I got very poor fps (1~5) for pointcloud data.
I increased buffer size for DDS and I got fps over 15!

Thanks a lot.