realsenseai/realsense-ros #1704: rs_from_file.launch failed to executed

gachiemchiep, 2021-02-17:
Hello realsense team. 

I used the "rs_from_file.launch" to read bag file. 

```xml
<launch>
    <!-- read directly from file-->
   <include file="$(find realsense2_camera)/launch/rs_from_file.launch" >
      <arg name="rosbag_filename"    value="/home/tr/Documents/d435i_sample_data/d435i_walk_around.bag" />
   </include>
</launch>

```

Unfortunately, the following error happened 

```bash
started roslaunch server http://jil-dev-truong:38899/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 0
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 0
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 0
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 0
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 0
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 0
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 0
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
 * /camera/realsense2_camera/gyro_fps: 0
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 0
 * /camera/realsense2_camera/infra_height: 0
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 0
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: /home/jil/Documen...
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: camera/odom_in
 * /camera/realsense2_camera/unite_imu_method: none
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [17221]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 46a92f2a-70f4-11eb-9174-00d8612f5384
process[rosout-1]: started with pid [17234]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [17241]
process[camera/realsense2_camera-3]: started with pid [17242]
[ INFO] [1613548003.054306494]: Initializing nodelet with 12 worker threads.
[ INFO] [1613548003.343302430]: RealSense ROS v2.2.21
[ INFO] [1613548003.343356270]: Built with LibRealSense v2.41.0
[ INFO] [1613548003.343384809]: Running with LibRealSense v2.41.0
[ INFO] [1613548003.345717968]: publish topics from rosbag file: /home/jil/Documents/d435i_sample_data/d435i_walk_around.bag
[FATAL] [1613548003.498935774]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 17241, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/jil/.ros/log/46a92f2a-70f4-11eb-9174-00d8612f5384/camera-realsense2_camera_manager-2.log].
log file: /home/jil/.ros/log/46a92f2a-70f4-11eb-9174-00d8612f5384/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 17242, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/jil/.ros/log/46a92f2a-70f4-11eb-9174-00d8612f5384/camera-realsense2_camera-3.log].
log file: /home/jil/.ros/log/46a92f2a-70f4-11eb-9174-00d8612f5384/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

```

The bag file was downloaded from [librealsense sample data](https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md), so I think the file itself is fine. 

Do you guys know how to solve this? 

Thank you

MartyG-RealSense, 2021-02-17:
Hi @gachiemchiep   There are some differences between ROS bags and bags created in the RealSense SDK like the sample bags.

https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

Having said that, as you are loading a bag that contains IMU information and IMU topics are disabled by default in the RealSense ROS wrapper, the action that I would recommend trying first is to enable the IMU topics in your roslaunch instruction by adding the instructions below to the end of it:

**enable_gyro:=true enable_accel:=true**

gachiemchiep, 2021-02-18:
@MartyG-RealSense 

I tried but the same thing happened. 

```xml
<launch>
   <include file="$(find realsense2_camera)/launch/rs_from_file.launch" >
      <arg name="rosbag_filename"    value="/home/jil/Documents/d435i_sample_data/d435i_walk_around.bag" />
      <arg name="enable_gyro"        value="true" />
      <arg name="enable_accel"        value="true" />
   </include>
</launch>
```

```bash
[ INFO] [1613614543.530308541]: Initializing nodelet with 12 worker threads.
[ INFO] [1613614543.843616671]: RealSense ROS v2.2.21
[ INFO] [1613614543.843673850]: Built with LibRealSense v2.41.0
[ INFO] [1613614543.843704140]: Running with LibRealSense v2.41.0
[ INFO] [1613614543.848096890]: publish topics from rosbag file: /home/jil/Documents/d435i_sample_data/d435i_walk_around.bag
[FATAL] [1613614556.321736400]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 19648, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/jil/.ros/log/337430da-718f-11eb-9174-00d8612f5384/camera-realsense2_camera_manager-2.log].
log file: /home/jil/.ros/log/337430da-718f-11eb-9174-00d8612f5384/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 19649, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/jil/.ros/log/337430da-718f-11eb-9174-00d8612f5384/camera-realsense2_camera-3.log].
log file: /home/jil/.ros/log/337430da-718f-11eb-9174-00d8612f5384/camera-realsense2_camera-3*.log

```

MartyG-RealSense, 2021-02-18:
An error with bag loading can occur if a full **absolute path** to the bag is not provided.  It looks as though you have done that though with **/home/jil/Documents/d435i_sample_data/d435i_walk_around.bag**

An error may also occur if the bag does not contain depth information according to the link below, but **d435i_walk_around.bag** does have a depth stream, which would seem to eliminate that possibility.

https://github.com/IntelRealSense/realsense-ros/issues/1180#issuecomment-633407652

Could you try playing the bag back in the RealSense Viewer to check that the bag is playable, please?  This can be done by drag and dropping the bag file into the center panel of the Viewer.

gachiemchiep, 2021-02-24:
@MartyG-RealSense 
Sorry for the late reply. 
The bag file can be played on realsense-viewer

[image]

MartyG-RealSense, 2021-02-24:
Thanks very much for the information @gachiemchiep 

Could you try accessing the bag file directly from the roslaunch instruction instead of an arg please?  For example:

**roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/home/jil/Documents/d435i_sample_data/d435i_walk_around.bag"**

MartyG-RealSense, 2021-03-02:
Hi @gachiemchiep  Do you require further assistance with this case, please?   Thanks!

gachiemchiep, 2021-03-02:
@MartyG-RealSense 

Sorry for the late response. The same error happened 

```bash
jil@jil-dev-truong:catkin_ws$ roslaunch realsense2_camera rs_from_file.launch rosbag_filename:="/home/jil/Documents/d435i_sample_data/d435i_walk_around.bag"
... logging to /home/jil/.ros/log/b236a412-7b2a-11eb-81bf-00d8612f5384/roslaunch-jil-dev-truong-12693.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jil-dev-truong:40471/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 0
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 0
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 0
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 0
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 0
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 0
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 0
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 0
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 0
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 0
 * /camera/realsense2_camera/gyro_fps: 0
[... 32 lines cut ...]
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [12728]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b236a412-7b2a-11eb-81bf-00d8612f5384
process[rosout-1]: started with pid [12739]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [12745]
process[camera/realsense2_camera-3]: started with pid [12747]
[ INFO] [1614670892.942232173]: Initializing nodelet with 12 worker threads.
[ INFO] [1614670893.497140409]: RealSense ROS v2.2.21
[ INFO] [1614670893.497201869]: Built with LibRealSense v2.41.0
[ INFO] [1614670893.497242159]: Running with LibRealSense v2.41.0
[ INFO] [1614670893.501724957]: publish topics from rosbag file: /home/jil/Documents/d435i_sample_data/d435i_walk_around.bag

[FATAL] [1614670918.465495615]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera_manager-2] process has died [pid 12745, exit code -11, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/jil/.ros/log/b236a412-7b2a-11eb-81bf-00d8612f5384/camera-realsense2_camera_manager-2.log].
log file: /home/jil/.ros/log/b236a412-7b2a-11eb-81bf-00d8612f5384/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has died [pid 12747, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/jil/.ros/log/b236a412-7b2a-11eb-81bf-00d8612f5384/camera-realsense2_camera-3.log].
log file: /home/jil/.ros/log/b236a412-7b2a-11eb-81bf-00d8612f5384/camera-realsense2_camera-3*.log

```

MartyG-RealSense, 2021-03-02:
I examined the case again from the beginning and re-researched it but made no progress unfortunately.  Given the past history of problems with loading files using **rs_from_file.launch**, it is likely that you are not doing anything wrong but it is probably not going to work.

doronhi, 2021-03-03:
I think the error is not related to the rosbag file. If there was an issue with it you would see the following error message: `Failed to create ros reader: Error opening file: /home/jil/Documents/d435i_sample_data/d435i_walk_around.bag`
I guess there is an issue with the installation or the environment. Does the node start with rs_camera.launch?

gachiemchiep, 2021-03-05:
@dorodnic Yes. The node can start with rs_camera.launch

By the way, I updated to the latest version 2.42.0 and the error is disappeared. 
So I'm fine now. Thank you for your guys's help

closed 2021-03-05 by gachiemchiep

MartyG-RealSense, 2021-03-05:
Excellent news that an SDK update solved your problem!