realsenseai/realsense-ros #1707: Concurrent dynamic reconfigure requests result in "set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy"

stwirth, 2021-02-17:
In our setup we use multiple ROS nodes (different processes) that interact with the RealSense camera ROS driver. Sometimes, two reconfigure requests are sent at the same time which results in
```
[ERROR] [1613582416.639314747]: Service call failed: service [/camera_down_orig/stereo_module/set_parameters] responded with an error: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
```
This is probably due to the fact that two calls to `sensor.set_option` are made from two parallel threads.

stwirth, 2021-02-17:
A solution could be to add a mutex for each option and lock it in the lambda ddynamic reconfigure callback functions.

MartyG-RealSense, 2021-02-17:
Hi @stwirth   An approach based on dynamic_reconfigure might be applicable to the ROS1 wrapper.  A solution may not be applicable to the ROS2 wrapper too though, as it does not have dynamic_reconfigure and handles dynamic parameters differently, apparently, and uses a **parameter reconfigure** plugin with rqt instead.

https://github.com/IntelRealSense/librealsense/issues/5825#issuecomment-699768692

Instead of using dynamic_reconfigure, would it instead be practical to apply the change before launching, as you suggested in a 2018 case?

https://github.com/IntelRealSense/realsense-ros/issues/345#issuecomment-374880904

stwirth, 2021-02-17:
Thanks @MartyG-RealSense for your quick response.
The use case is that we switch off the camera emitters on our robots when we don't need them and switch them on only during the execution of a specific action. As there are multiple actions that need the emitter on and multiple actions being started at the same time, the described issue occurs.

The environment is:
ROS1 on Ubuntu 16.04 w/ kernel 5.8.3-050803-generic

RealSense ROS v2.2.13
LibRealSense v2.33.1
Intel RealSense D410 with FW 05.12.07.100

I tried to reproduce the issue locally with a D435 (I don't have a D410 at hand) using the same software and firmware versions and unfortunately I could not observe the error.

MartyG-RealSense, 2021-02-17:
D415 (and presumably D410, which is in the same class of camera) handles the emitters differently to D435.  Intel's white-paper document about projectors states that D415's projector is always on, whilst D435's projector's default mode is to pulse its light to coincide with camera exposure.

https://dev.intelrealsense.com/docs/projectors?_ga=2.147584183.1877976161.1602709969-1041004708.1589641371#section-4-increasing-range

stwirth, 2021-02-17:
Interesting. That might be the reason on the low level. Anyhow, I think concurrent calls to `set_option` should be avoided.

I just tested again with upgraded software and firmware:
RealSense ROS v2.2.21
LibRealSense v2.41.0
Device FW version: 05.12.11.00
The issue still occurs on the robot (with D410).
To reproduce I'm just hammering the driver with dynamic reconfigure requests in a loop from two different processes.

MartyG-RealSense, 2021-02-17:
I recalled a past case in which ROS users were experiencing the busy error when using set_option.  They were using ROS nodes to access it.

https://github.com/IntelRealSense/librealsense/issues/6347

They tested the failure rate using multiple restarts.

https://github.com/IntelRealSense/librealsense/issues/6347#issuecomment-625252353

Their solution in the end that resolved the error for them was to create a function (to change exposure in their particular case) that was called from their main.cpp script.

https://github.com/IntelRealSense/librealsense/issues/6347#issuecomment-635290404

MartyG-RealSense, 2021-02-24:
Hi @stwirth   Do you require further assistance with this case, please?   Thanks!

stwirth, 2021-02-24:
@MartyG-RealSense I implemented a workaround on our side so that the dynamic reconfigure requests don't reach the camera driver in a concurrent way. Anyhow, I think the way how ddynamic_reconfigure is used in this driver is not thread safe and should be revised at some point.

MartyG-RealSense, 2021-02-24:
Excellent news @stwirth that you were able to implement a workaround!

Going forward with the new ROS2 wrapper, a plugin called **parameter reconfigure** is used in the RealSense ROS wrapper instead of the **dynamic_reconfigure** used in the ROS1 wrapper up until now.

https://github.com/IntelRealSense/librealsense/issues/5825#issuecomment-699768692

stwirth, 2021-03-02:
@MartyG-RealSense As I was able to use a workaround I don't need further assistance. Anyhow, I think this is a bug.

MartyG-RealSense, 2021-03-02:
Thanks very much @stwirth for the update of your success via a workaround.

closed 2021-03-02 by MartyG-RealSense