realsenseai/realsense-ros #1708: Change the parameters of stream aligned_depth_to_color

Schloern93, 2021-02-17:
Hello, 

I use the stream "/D435/aligned_depth_to_color/image_raw".
of my D435 camera.

This means that my depth image (which has a bigger fov than the color image) is adjusted to my color image.

Is there any way to change the parameters of this adjustment?
I use Ros2 Eloquent 

Thanks for the answer

MartyG-RealSense, 2021-02-17:
Hi @Schloern93   Are you creating a point cloud please?

If so, in a similar case recently involving depth information being excluded when depth was aligned to color, I suggested adding the instruction below to the end of the roslaunch instruction:

**allow_no_texture_points:=true**

The documentation for the Eloquent wrapper has the following description about allow_no_texture_points:

************

The depth FOV and the texture FOV are not similar.  By default, pointcloud is limited to the section of depth containing the texture.  You can have a full depth to pointcloud, coloring the regions beyond the texture with zeros, by setting allow_no_texture_points to true.

Schloern93, 2021-02-17:
Hello

Thanks for the quick reply.

I have already created a PointCloud. I did it using the package linked below. 
http://wiki.ros.org/depth_image_proc
The problem is, when I have a certain pixel from the camera color stream and I try to find the same pixel in the Point Cloud, it almost seems like each pixel is slightly off to the left of the color image.
Since I don't understand where this deviation comes from I wanted to look at the calculations of the align_depth_to_color stream

MartyG-RealSense, 2021-02-17:
Have you performed a Dynamic Calibration of your camera's sensors recently to help ensure that they are not mis-calibrated?

Schloern93, 2021-02-18:
No, not yet. I will test that and give feedback here again.

closed 2021-03-03 by MartyG-RealSense