realsenseai/realsense-ros #1710: Filters parameters for Foxy

rafaelRinconFo, 2021-02-18:
According to the Foxy documentation, the `filters` parameters can launch the following filters during the ros2 launch process:

- disparity
- spatial
- temporal
- hole_filling
- decimation

However, as far as can be seen in the documentation, there is no direct  way to modify the filters' parameters considered in the [librealsense documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md), such as the **Persistency index** in the **Temporal filter** among others.

Is there an easy way to modify the filter parameters? Is this considered to be included in future releases?

Thanks in advance.

MartyG-RealSense, 2021-02-18:
Hi @rafaelRinconFo   In the ROS1 wrapper at least, you could define a filter generally in the roslaunch instruction and then apply specific settings for the filter with a **rosrun dynamic_reconfigure dynparam set** instruction.

https://github.com/IntelRealSense/realsense-ros/issues/583

In the ROS2 wrapper though, instead of using dynamic_reconfigure you would use rqt with the **parameter reconfigure** plugin.

https://github.com/IntelRealSense/librealsense/issues/5825#issuecomment-699768692

I will tag @doronhi the RealSense ROS wrapper developer into this discussion to seek advice.

Hi @doronhi   Could you provide advice please about how to apply specific filter settings such as **magnitude** to a post-processing filter in the ROS2 Foxy wrapper if possible to do so, please?  Thanks!

rafaelRinconFo, 2021-02-19:
> Hi @rafaelRinconFo In the ROS1 wrapper at least, you could define a filter generally in the roslaunch instruction and then apply specific settings for the filter if that filter supported it using a **rosrun dynamic_reconfigure dynparam set** instruction.
> 
> #583
> 
> In the ROS2 wrapper though, instead of using dynamic_reconfigure you would use rqt with the **parameter reconfigure** plugin.
> 
> [IntelRealSense/librealsense#5825 (comment)](https://github.com/IntelRealSense/librealsense/issues/5825#issuecomment-699768692)
> 
> I will tag @doronhi the RealSense ROS wrapper developer into this discussion to seek advice.
> 
> Hi @doronhi Could you provide advice please about how to apply specific filter settings such as **magnitude** to a post-processing filter in the ROS2 Foxy wrapper if possible to do so, please? Thanks!

Thank you so much for you answer @MartyG-RealSense , I'm really looking forward to know what could be done.

doronhi, 2021-02-21:
while running the node with the filter on, you can list the parameters in a different console.
For this example, I'll use the temporal filter:
`ros2 param list | grep temporal`
yeilds:
```
  temporal.filter_smooth_alpha
  temporal.filter_smooth_delta
  temporal.frames_queue_size
  temporal.holes_fill
  temporal.stream_filter
  temporal.stream_format_filter
  temporal.stream_index_filter
```
The names of the arguments are sometimes different from those seen in the "realsense-viewer" app but they are the same controls.
For enum arguments, the values are documented in the description. For example:
`ros2 param describe /camera/camera temporal.holes_fill`
yeilds:
```
Parameter name: temporal.holes_fill
  Type: integer
  Description: Persistency mode
Disabled          - 0
Valid in 8/8      - 1
Valid in 2/last 3 - 2
Valid in 2/last 4 - 3
Valid in 2/8      - 4
Valid in 1/last 2 - 5
Valid in 1/last 5 - 6
Valid in 1/8      - 7
Always on         - 8
```

So now you can get the current value using:
`ros2 param get /camera/camera temporal.holes_fill`
of set a new one:
`ros2 param set /camera/camera temporal.holes_fill 0`

You can also set the value in the launch file. Add the following element to the "configurable_parameters" list in your rs_launch.py file:

```
 {'name': 'temporal.holes_fill',          'default': '0', 'description': 'Persistency mode'},
```
*Note:* If you build from source, don't forget to `colcon build` again, to copy the fixed launch file to the "install" library.
Now starting the node with the following command will set the value of "temporal.holes_fill" to 0:
`ros2 launch realsense2_camera rs_launch.py filters:=temporal,colorizer depth_qos:=SYSTEM_DEFAULT`
or you can set it as you see fit:
`ros2 launch realsense2_camera rs_launch.py filters:=temporal,colorizer depth_qos:=SYSTEM_DEFAULT temporal.holes_fill:=8`

MartyG-RealSense, 2021-02-21:
Thanks so much @doronhi  :)

rafaelRinconFo, 2021-02-25:
Thank you so much @doronhi  and @MartyG-RealSense . This will be really helpfull

closed 2021-02-25 by rafaelRinconFo