realsenseai/realsense-ros #1713: Problem updating from v2.2.21 to v2.2.22

mohammad-h-b, 2021-02-22:
Hi,
I wonder if you let me know whether my issues (#128) and (#129) are due to the my librealsense version (2.2.21)? I tried to upgradre it but don't know how to do so. I have SDK v2.42.

Please give also the following information:

Operating system and version (e.g. Ubuntu 16.04 or Windows 10): Ubuntu 18.04
Did you change the source code? (yes / no): no

MartyG-RealSense, 2021-02-22:
Hi @mohammad-h-b   If you are following the official Kimera-VIO-ROS instructions for D435i in the link below, they are currently out of date.  The RealSense ROS wrapper now requires that the IMU topics are enabled in its roslaunch instruction using **enable_gyro:=true enable_accel:=true** as they are disabled by default from wrapper version 3.2.15 onward (July 2020).

https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md

Adapting the Kimera-VIO-ROS launch instruction, it would now be:

**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation**

It is also worth mentioning that if you have updated the librealsense version since the last time that you built the RealSense ROS wrapper then the wrapper should be rebuilt.  It is also recommended to use wrapper version **2.2.22** with librealsense version **2.42.0**.

mohammad-h-b, 2021-02-22:
Hi @MartyG-RealSense Thanks for your answer. I'll definitely use the adapted launch file for Kimera-VIO-ROS. Regarding the wrapper, could you please let me know how I can get wrapper version 2.2.22 with librealsense version 2.42.0 working? When I run `realsense-viewer` it opens up v2.42.0 (as shown in the screenshot) but when I do `catkin build` in my `catkin_ws` and relaunch the camera node, it still reports:
```
[ INFO] [1613982438.367514029]: RealSense ROS v2.2.21
[ INFO] [1613982438.367535096]: Built with LibRealSense v2.41.0
[ INFO] [1613982438.367543429]: Running with LibRealSense v2.41.0
```

MartyG-RealSense, 2021-02-22:
There is not a problem with using librealsense 2.41.0 and wrapper 2.2.21 together.  

If you wish to try 2.42.0 though: as this SDK version is new at the time of writing this, I would recommend installing wrapper 2.2.22 by building it from source code with **Method 2** (i.e build librealsense first and then build 2.2.22 secondly with Method 2).

https://github.com/IntelRealSense/realsense-ros#method-2-the-realsense-distribution

If you installed librealsense and the ROS wrapper together with packages using **Method 1** then the installation packages may still be using 2.41.0 due to the current newness of SDK 2.42.0.  Building the wrapper from source with Method 2 means that you can use a new RealSense ROS wrapper version as soon as it is released.

mohammad-h-b, 2021-02-22:
Thank you so much @MartyG-RealSense !
So for now I stick to librealsense 2.41.0 and wrapper 2.2.21 together.
I've also tried the new launch file `**roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation**`. I am no longer getting erros in (https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/129) but still running `roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
` generates all zero values as in (https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/128) and there is no visulization in rviz.

Could you please let me know about that?

Output:
```
(base) mohammad@mohammad:~/catkin_ws$ roslaunch kimera_vio_ros kimera_vio_ros_realsense_IR.launch
... logging to /home/mohammad/.ros/log/16adb76e-74eb-11eb-8f79-c8e2658ec50b/roslaunch-mohammad-11572.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://mohammad:39771/

SUMMARY
========

PARAMETERS
 * /kimera_vio_ros/kimera_vio_ros_node/base_link_frame_id: base_link
 * /kimera_vio_ros/kimera_vio_ros_node/left_cam_frame_id: left_cam
 * /kimera_vio_ros/kimera_vio_ros_node/map_frame_id: map
 * /kimera_vio_ros/kimera_vio_ros_node/mono_ransac_threshold: 30
 * /kimera_vio_ros/kimera_vio_ros_node/online_run: True
 * /kimera_vio_ros/kimera_vio_ros_node/params_folder_path: /home/mohammad/ca...
 * /kimera_vio_ros/kimera_vio_ros_node/position_det_threshold: 0.3
 * /kimera_vio_ros/kimera_vio_ros_node/right_cam_frame_id: right_cam
 * /kimera_vio_ros/kimera_vio_ros_node/stereo_ransac_threshold: 20
 * /kimera_vio_ros/kimera_vio_ros_node/use_lcd: False
 * /kimera_vio_ros/kimera_vio_ros_node/use_online_cam_params: False
 * /kimera_vio_ros/kimera_vio_ros_node/velocity_det_threshold: 0.1
 * /kimera_vio_ros/kimera_vio_ros_node/world_frame_id: world
 * /kimera_vio_ros/posegraph_viewer/frame_id: world
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /use_sim_time: False

NODES
  /
    velo_link_broadcaster (tf/static_transform_publisher)
  /kimera_vio_ros/
    kimera_vio_ros_node (kimera_vio_ros/kimera_vio_ros_node)
    posegraph_viewer (pose_graph_tools/visualizer_node)

ROS_MASTER_URI=http://localhost:11311

process[kimera_vio_ros/kimera_vio_ros_node-1]: started with pid [11588]
process[kimera_vio_ros/posegraph_viewer-2]: started with pid [11589]
process[velo_link_broadcaster-3]: started with pid [11590]
[ INFO] [1613984132.290383413]: Initializing pose graph visualizer
I0222 03:55:32.432449 11588 RegularVioBackEnd.cpp:114] Using Regular VIO backend.
I0222 03:55:32.432530 11588 PipelineModule.h:449] MISO Pipeline Module: Display has no output queue registered.
I0222 03:55:32.432626 11588 Pipeline.cpp:479] Pipeline Modules launched (parallel_run set to 1).
I0222 03:55:32.432745 11588 KimeraVioRos.cpp:117] Statistics
-----------                                  #	Log Hz	{avg     +- std    }	[min,max]
Mesher [ms]                             	    0	
VioBackEnd [ms]                         	    0	
VioFrontEnd [ms]                        	    0	
Visualizer [ms]                         	    0	
backend_input_queue Size [#]            	    0	
data_provider_left_frame_queue Size [#] 	    0	
data_provider_right_frame_queue Size [#]	    0	
display_input_queue Size [#]            	    0	
mesher_backend Size [#]                 	    0	
mesher_frontend Size [#]                	    0	
stereo_frontend_input_queue Size [#]    	    0	
visualizer_backend_queue Size [#]       	    0	
visualizer_frontend_queue Size [#]      	    0	
visualizer_mesher_queue Size [#]        	    0	
I0222 03:55:32.432826 11614 Pipeline.h:199] Spinning Kimera-VIO.
I0222 03:55:33.432844 11588 KimeraVioRos.cpp:117] Statistics
```

MartyG-RealSense, 2021-02-22:
Given that Kimera makes use of infrared, the lack of data may be related to the same issue as the IMU topics.  The **infra** topics are also disabled by default from wrapper 2.2.15 onwards.  So to ensure that the infrared topics are enabled, please add the instructions below to the roslaunch instruction too:

**enable_infra1:=true enable_infra2:=true**

mohammad-h-b, 2021-02-22:
Thank @MartyG-RealSense I've added the intsctructions but still get zero values. Also when I run `roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:=linear_accel_cov` I get these warnings as well:
```
22/02 11:56:08,911 WARNING [140615801349888] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
```
Please note that for the `unite_imu_method` it only has the option `linear_accel_cov` not the `linear_interpolation`.

MartyG-RealSense, 2021-02-22:
This is a rare error that has only seemed to occur with librealsense 2.41.0.  It was apparently fixed in **2.42.0**.

https://github.com/IntelRealSense/librealsense/issues/8154#issuecomment-767175826

mohammad-h-b, 2021-02-22:
I see. So is there any fix for the zero values I'm getting?

MartyG-RealSense, 2021-02-22:
In the **Testing** section of the Kimera instructions, are you launching each instruction in its own separate ROS terminal as specified by the first line of this section, please?

https://github.com/MIT-SPARK/Kimera-VIO-ROS/blob/master/docs/hardware_setup.md#testing

mohammad-h-b, 2021-02-22:
Yes, I followed all the instructions each in separate terminals:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

MartyG-RealSense, 2021-02-22:
The above images indicate that librealsense 2.42.0 has been installed but the ROS wrapper has not been rebuilt afterwards.  After a change of librealsense version, the ROS wrapper has to be built again.

[image]

It is also worth bearing in mind that the more common approach to using SLAM with RealSense D435i is to do so with Intel's own D435i SLAM guide. 

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

mohammad-h-b, 2021-02-22:
That's right. I did a whole `caktkin build` in my `catkin_ws` but not sure it still generates that warning.

MartyG-RealSense, 2021-02-22:
If you get a version conflict when building in the catkin workspace, I would suggest deleting the catkin workspace and creating a new workspace to build the wrapper in to see whether that makes a difference.

mohammad-h-b, 2021-02-23:
Thanks @MartyG-RealSense I deleted the catkin workspace and built the wrapper in the new one. Now the warning is gone; however, I'm still getting zero values even with `roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true enable_infra1:=true enable_infra2:=true unite_imu_method:=linear_interpolation`

MartyG-RealSense, 2021-02-23:
I wonder whether the continued zero-values are related to calibration of the D435i IMU, and whether the IMU needs to first have a calibration written to the camera hardware using Intel's Python-based IMU calibration tool before using it with Kimera.

https://dev.intelrealsense.com/docs/imu-calibration-tool-for-intel-realsense-depth-camera

mohammad-h-b, 2021-02-23:
Thanks for the tip!

I'll look into using Intel's Python-based IMU calibration tool.

Regarding the sub-categories of calibration in point 3 of the Kimera instructions, it's mentioned that they are not needed to be done if you want to use the default calibration and launch files. I am using the exact same camera they've used (D435i), so I'm not sure if I need to carry them out.

MartyG-RealSense, 2021-03-01:
Hi @mohammad-h-b   Do you require further assistance with this case, please?  Thanks!

mohammad-h-b, 2021-03-01:
Hi @MartyG-RealSense Still I couldn't get it running on my system :(

MartyG-RealSense, 2021-03-02:
A member of the Kimera forums suggested testing with a bag file instead of live data to see whether there is a problem with Kimera instead of the camera:

https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/24#issuecomment-625397783

mohammad-h-b, 2021-03-02:
I just tested kimera with the given rosbag file and it works fine. So I guess the problem is with my D435i, although `realsense-viewer` brings up the camera and it works fine.

MartyG-RealSense, 2021-03-03:
On the Kimera-VIO-ROS, another D435i user used the details in the link below in their roslaunch file, a modification of kimera_vio_ros_realsense_IR.launch.  They set **online** mode to True in the launch file with **<arg name="online" default="True" />**

https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/92#issuecomment-698313394 

They also

mohammad-h-b, 2021-03-03:
Thanks @MartyG-RealSense I already set that parameter to true. Anyway, how can I run the Kimera-Semantics in Gazebo? There I could put a sample camera and see how it works.

MartyG-RealSense, 2021-03-03:
I researched your question.  I could not find an example of gazebo use with Kimera, though somebody else had asked about it.

https://github.com/MIT-SPARK/Kimera-VIO-ROS/issues/119

closed 2021-03-16 by MartyG-RealSense