realsenseai/realsense-ros #1715: Issues with topic matching on start-up

AndreasAZiegler, 2021-02-22:
I'm not sure if this issue is a general ROS2/DDS issue or specific to this ROS2 node. In my setup I have two RealSense nodes (each publishing the color images and the depth map) and an own node that receives the color images and the depth map. Because it is quite important that the nodes receives the data from both RealSense nodes, I implemented a functionality that detects if no data was received from a RealSense node for more than one second.

When I start all three nodes with the following launch file
```
import os
import launch
from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch_ros.actions import Node
from launch.substitutions import LaunchConfiguration
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import ThisLaunchFileDir
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

def generate_launch_description():

    # config the serial number and base frame id of each camera
    d435_base_frame_id = LaunchConfiguration('base_frame_id', default='d435_link')
    d435_serial_no = LaunchConfiguration('d435_serial_no', default='825312073597')
    
    d435i_base_frame_id = LaunchConfiguration('base_frame_id', default='d435i_link')
    d435i_serial_no = LaunchConfiguration('d435i_serial_no', default='841512070768')

    d435i_config = os.path.join(get_package_share_directory('safety-field-intrusion-detector-ros'), 'config', 'd435i.yaml')
    d435_config = os.path.join(get_package_share_directory('safety-field-intrusion-detector-ros'), 'config', 'd435.yaml')
    detector_config = os.path.join(get_package_share_directory('safety-field-intrusion-detector-ros'), 'config', 'params.yaml')

    d435_node = Node(
        package='realsense2_camera',
        node_namespace='/realsense2',
        node_executable='realsense2_camera_node',
        output='screen',
        parameters=[
            #{'serial_no': d435_serial_no,
            # 'base_frame_id': d435_base_frame_id}
            d435_config
        ])
    d435i_node = Node(
        package='realsense2_camera',
        node_namespace='/realsense1',
        node_executable='realsense2_camera_node',
        output='screen',
        parameters=[
            #{'serial_no': d435i_serial_no,
            # 'base_frame_id': d435i_base_frame_id}
            d435i_config
        ])
    detector_node = Node(
        package='safety-field-intrusion-detector-ros',
        node_executable='mt-detector-robot-node',
        node_name='safety_field_intrusion_detector',
        parameters=[
            detector_config
        ])

    return launch.LaunchDescription([d435_node, d435i_node, detector_node])
```
the three nodes start but when the two RealSense nodes are up, my custom node is still not able to receive data from both  cameras as can be seen [in this log](https://gist.github.com/AndreasAZiegler/8e2f4dbcebb9c63990271ea5b9536a30).

When I start the three nodes with the following bash script
```
#!/bin/bash
set -eo pipefail
source /opt/ros/eloquent/setup.bash
source /usr/share/safety-field-intrusion-detector-ros/local_setup.bash
ros2 run realsense2_camera realsense2_camera_node --ros-args --params-file `ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435i.yaml -r __ns:=/realsense1 & \
ros2 run realsense2_camera realsense2_camera_node --ros-args --params-file `ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435.yaml -r __ns:=/realsense2 & \
ros2 run safety-field-intrusion-detector-ros mt-detector-robot-node --ros-args --params-file `ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/params.yaml
```
everything works fine as can be seen [in this log](https://gist.github.com/AndreasAZiegler/7bcdb7b8b3d3cd01694361d5b15ded00).

The start via bash script works but I would prefer to start the nodes via launch file.

RealSenseSupport, 2021-03-16:
@AndreasAZiegler Do you always get the issue via launch file? Or just occasionally? I saw you set initial_reset as true in launch file. Could you please try to set it as false just as listed below to see if you still get the issue? 

https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_camera.launch#L60

RealSenseSupport, 2021-03-22:
@AndreasAZiegler Any update from your side? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-04-06:
@AndreasAZiegler Do you still need support on this? Could you please update? Thanks!

AndreasAZiegler, 2021-04-08:
@RealSenseSupport Sorry for my late response. I couldn't work on this project recently. I might get back to it next week.

RealSenseSupport, 2021-04-25:
@AndreasAZiegler Any update from your side? Thanks!

AndreasAZiegler, 2021-04-30:
@RealSenseSupport I'm no longer working at the company where I worked on this project and didn't manage to have another look at it before I left.

RealSenseSupport, 2021-04-30:
@AndreasAZiegler Thanks for your update! In this case, can we close this ticket accordingly? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-05-12:
@AndreasAZiegler As you don't work on the project anymore. Closed the ticket accordingly. Thanks!

closed 2021-05-12 by RealSenseSupport