realsenseai/realsense-ros #1716: Mis-alignment of depth to color for L515 when using realsense-ros

abhijitmajumdar, 2021-02-22:
Issue: Here at Plusone Robotics we are using the L515 for our applications and when using `rs_rgbd.launch` from `realsense-ros` to obtain pointcloud, either with `align_depth:=true` or with `align_depth:=false`, produces a mis-aligned pointcloud (~2-4px mis-alignment between the color and the depth information). However when visualizing the pointcloud using `realsense-viewer`, the point cloud looks aligned.

Please see attached video for a visual description of the issue.

Setup:
- Ubuntu 16.04
- librealsense 2.41.0
- realsense-ros 2.2.21
- Realsense L515 w/ firmware 01.05.03.00

Steps to reproduce issue:
- run `roslaunch realsense2_camera rs_rgbd.launch align_depth:=true enable_sync:=true`
- place a rectangular object at far distance (~3-4m) from the sensor
- make sure there is nothing around the rectangular object, and there is a contrasting (in color) background behind the rectangular object at farther distance (~5m)
- make sure ambient light is low so that both the rectangular object and the background are able to produce valid points in the point cloud
- observe the point cloud (say in Rviz). Because of the contrasting color of the rectangular object vs the background, about 2-4 pixel width of the edges of this rectangular object would take on color information from the background, thus indicating a mis-alignment during registration of the depth image to the color image
- observe the same scene with `realsense-viewer` and notice the misalignment minimized (`Shading` set to view the `Raw Point-Cloud`)

Notes:
- The same mis-alignment is observed irrespective of aligning the depth image using realsense-ros alignment (i.e. setting `align_depth:=true` in `rs_rgbd.launch`) as well as with ROS registration of the depth image to the color image (i.e. setting `align_depth:=false` in `rs_rgbd.launch`)
- Obtaining pointcloud using `realsense-ros` directly (i.e. setting `enable_pointcloud:=true`) is not a valid alternative because it generates a point cloud which is in the depth image frame with depth intrinsic and hence does not correlate to the color image directly
- please see attached video to observe the misalignment when using `rs_rgbd.launch` from `realsense-ros`

https://user-images.githubusercontent.com/24485799/108753139-f6c20d00-7509-11eb-9d02-c5cc1afacb3d.mp4

P.S. I logged the same issue on Zendesk (Issue #13895) but have not heard back for some time now, and hence logging the issue here since github is much more active

@130s Pinging you to be able to keep track on the status

doronhi, 2021-02-23:
Did you check the aligned-depth-to-color image? Is it well aligned to the color image? Could that be the cause of the shift you see with rgbd's pointcloud. If so, I suggest we start investigating it, starting with looking into previously asked issues.
rgbd_launch is a 3rd party package for which I can't answer unless we find fault in the output of realsense2_camera node.
As for your need for an order pointcloud aligned to the color image, it resembles the request at #1680.

abhijitmajumdar, 2021-02-23:
@doronhi Thanks for getting back on this.

> Did you check the aligned-depth-to-color image? Is it well aligned to the color image?

Yes I did and unfortunately no it does not. The aligned-depth-to-color image (when using the `align:=true`) produces the same misalignment as the alignment(registration) produced by [depth_image_proc/register](http://wiki.ros.org/depth_image_proc#depth_image_proc.2Fregister) (i.e. with `align:=false`), which is what makes me believe that the intrinsic may be wrong/not-configured properly.

>  Could that be the cause of the shift you see with rgbd's pointcloud

In my observation the aligned-depth-to-color image generated by `realsense-ros` is misaligned and hence causes the point cloud to be misaligned as well.

>  unless we find fault in the output of realsense2_camera node.

To eliminate launch files, I ran the following experiment
- run `realsense2_camera` nodelet and make sure `align` is set to `true` so that `aligned_depth_to_color` image is published
- created a test ROS node such that it subscribes to the `/camera/color/image_raw` and `/camera/aligned_depth_to_color/image_raw` and overlay them so the mis-alignment can be observed
- the video below shows the results (the slider defines the proportion of the color vs depth to overlay, similar to `rs-align`)
  
https://user-images.githubusercontent.com/24485799/108818883-ad140980-757f-11eb-9f05-b449c977a0bb.mp4
- I ran `rs-align` to see if this behavior is observed through `librealsense` as well. My observation was that when its set to "Align to color" it generates a similar behavior as observed above for misalignment, however when set to "Align to depth" the misalignment is minimized (but a lot of color points are missed). The following videos show the results of my test on the same scene
  
https://user-images.githubusercontent.com/24485799/108818933-bd2be900-757f-11eb-8a7b-61cc881fd427.mp4

https://user-images.githubusercontent.com/24485799/108818979-d16fe600-757f-11eb-881b-cf7199b67287.mp4

> starting with looking into previously asked issues.

I did find some older issues related to the same mis-alignment issue on the L515 (https://github.com/IntelRealSense/realsense-ros/issues/1499, https://github.com/IntelRealSense/realsense-ros/issues/1405) but they seem to be un-resolved

> As for your need for an order pointcloud aligned to the color image, it resembles the request at #1680

I agree, going through that issue, it seems like having the pointcloud generated by `realsense-ros` registered to the color camera intrinsic (while maintaining the organized structure) would generate the 1-to-1 correlation desired. You mention [here](https://github.com/IntelRealSense/realsense-ros/issues/1680#issuecomment-776602412) that this is possible but limited because of the current code structure. However if implemented would that provide the expected result?

Would you recommend some way to (re)calibrate the intrinsic/extrinsic for the depth and color cameras on the L515 (I found some resources for the D415 but none for the L515)?

doronhi, 2021-02-25:
The cup at the front of the image does not have depth calculated for so I'll ignore it for now.
Looking at the distant box, I don't see a difference in accuracy between "Align to Depth" and "Align to Color". What I do see seems to be an issue with the edges. Could you turn on: "Invalidation bypass" option in Realsense-viewer? (L500 Depth Sensor->Controls->Invalidation Bypass) The value will stay On when you close realsense-viewer and rerun rs-align.
Using ROS you can use: `rosrun dynamic_reconfigure dynparam set /camera/l500_depth_sensor invalidation_bypass true`

abhijitmajumdar, 2021-02-26:
I tried setting the `invalidation_bypass` param. The results look more mis-aligned than when the parameter is set to `false`

With `invalidation_bypass` set to `true`
[image]

With `invalidation_bypass` set to `false`
[image]

abhijitmajumdar, 2021-02-26:
This is in rs-align with `Invalidation bypass` enabled in `realsense-viewer` like you suggested

https://user-images.githubusercontent.com/24485799/109288443-7aba1480-77ea-11eb-8608-a7b13b84bdcf.mp4

abhijitmajumdar, 2021-02-26:
For reference this is what I meant when I said `realsnese-viewer` shows much less mis-alignment as compared to either using `rs-align` or `realsense-ros`. Hence the assumption that there might be some discrepancy between the processing in `librealsense` vs `realsense-ros`

https://user-images.githubusercontent.com/24485799/109289655-2152e500-77ec-11eb-8400-24ccc9cedba9.mp4

abhijitmajumdar, 2021-03-11:
any update on this?

doronhi, 2021-03-14:
Sorry for not coming back to you.
We see a misalignment between the depth and color images. That would definitely create a misalignment in the rgbd's pointclound. It is very difficult to access such delicate differences in published videos, although very detailed as the ones you published. In my own scenario where I can rotate and zoom at a whim, I can't be sure there is an accuracy difference between the pointclouds published by rgbd, realsense2_camera (using filters:=pointcloud), and the one created by realsense-viewer. I could be wrong or my camera's intrinsics may be wrong in a different way but all 3 seems to me to have the same misalignment.
If we could establish that there is a difference between the pointcloud in realsense-viewer and realsense2_camera then this is a valid software issue we should pursue here.
If the pointcloud published by realsense2_camera is satisfactory, besides the fact that it is aligned to the depth image and not to the color image, then this issue will be resolved while addressing #1680. Shortly, I hope.
If this issue can be summed up as a misalignment between color and depth then I fear I can only recommend posting an issue in the [librealsense2 page](https://github.com/IntelRealSense/librealsense/issues). If you do that I think you should start by posting only the rs-align images or maybe more data in that direction, but removing the pointcloud reference as it might be a distraction. This is just my opinion as I couldn't convince myself there is an accuracy difference between the aligned images and the pointcloud in my own created scenario.

abhijitmajumdar, 2021-03-22:
Apologies for not getting back earlier, I was running through my experiments again to validate them and I ended up with the same conclusion.

> We see a misalignment between the depth and color images. That would definitely create a misalignment in the rgbd's pointclound. It is very difficult to access such delicate differences in published [...]
If we could establish that there is a difference between the pointcloud in realsense-viewer and realsense2_camera then this is a valid software issue we should pursue here.
If the pointcloud published by realsense2_camera is satisfactory, besides the fact that it is aligned to the depth image and not to the color image, then this issue will be resolved while addressing #1680. Shortly, I hope.

You are right in that it is desirable to have the organized pointcloud published (by say `realsense2_camera` using `filters:=pointcloud`) in the color frame, or atleast have an option to do so, looking forward towards that development. However, I do think there is still a misalignment issue between the color and depth and so I'll go ahead and log the issue on librealsense like you suggested.

> If the pointcloud published by realsense2_camera is satisfactory, besides the fact that it is aligned to the depth image and not to the color image, then this issue will be resolved while addressing #1680. Shortly, I hope.

On the lines of #1680, I had the impression that when `align:=true`, the `aligned_depth_to_color` topic was depth image transformed into the color frame. Is that not the case? If it is the case I would image the `aligned_depth_to_color` image can be converted into a pointcloud and textured with the color image, and since its in the color frame, the resultant pointcloud would be in the color frame. This is what I was trying to do with `rgbd.launch`.

doronhi, 2021-03-22:
> On the lines of #1680, I had the impression that when align:=true, the aligned_depth_to_color topic was depth image transformed into the color frame. Is that not the case? If it is the case I would [...]

You are correct. This is what @charlielito suggested [here](https://github.com/IntelRealSense/realsense-ros/issues/1680#issuecomment-793027363). It takes some thinking and time to integrate it nicely as a feature and I just haven't got around to it yet.

abhijitmajumdar, 2021-03-22:
> You are correct. This is what @charlielito suggested here. It takes some thinking and time to integrate it nicely as a feature and I just haven't got around to it yet.

Got it, thanks for confirming that

doronhi, 2021-04-18:
I have submitted #1806 which allows the pointcloud to be aligned to the color image.

doronhi, 2021-04-20:
Yes, the plan is to migrate this feature back to ROS1 shortly.

abhijitmajumdar, 2021-04-29:
@doronhi Thanks for adding the feature to publish the pointcloud in the RGB frame, and for porting the changes to ROS1.

Using this feature I did similar experiments as before with the expectation of the alignment fixes. My observation was that the alignment seems to have resolved on the left side of the pointcloud but seems to get more mis-aligned towards the right side, which to me indicates that there might clearly be an error with extrinsic between the depth and the color used for doing this registration.

I experimented this with 2 different L515s, both producing the same behavior (see configuration and corresponding video below). The box shown in the video is placed at ~14' (4.2m) from the sensor and the wall behind it is ~16'(4.8m)
- L515 (1), Firmware: 1.5.1.3 

  https://user-images.githubusercontent.com/24485799/116524880-83d46a00-a89d-11eb-90ac-01d1d3d307a5.mp4

- L515 (2), Firmware: 1.5.5.0 

  https://user-images.githubusercontent.com/24485799/116525290-fc3b2b00-a89d-11eb-83b5-19bf0bdda569.mp4

Do you think my judgment of the extrinsic being off are correct and If so, do you think this can be fixed somehow?

doronhi, 2021-05-02:
It could be that the extrinsics are off but I can't be sure.
Could you compare to **realsense-viewer** and **rs-pointcloud** and give me your impression.
It may help very much to establish the source of the issue being in ROS or above.

abhijitmajumdar, 2021-06-11:
> It could be that the extrinsics are off but I can't be sure.

@doronhi Sorry I could not get back to you on this earlier. We have been running several tests to validate this and to the best of our conclusions, re-calibrating the extrinsics reduced the error significantly.

We wanted to use the custom calibrated extrinsics in a way that we could write the new calibration values into the sensor itself. I have been trying to do so a few ways but the closest I could get is similar to as suggested [here](https://github.com/IntelRealSense/librealsense/issues/1706#issuecomment-389557627). I can successfully write new calibration values (using `from_stream.register_extrinsics_to(to_stream, depth_to_rgb_cal);` instead of `rs2_extrinsics extrinsics = from_stream.get_extrinsics_to(to_stream);` where depth_to_rgb_cal is the calibrated values that worked for us, and is an instance of `rs2_extrinsics`) and verified it by reading back the configs from the same `rs2::device` instance. However if I re-enumerate the device (or disconnect and reconnect) it, then read back the extrinsics, it reverts back to the original values, which indicates to me that what I am doing is only locally modifying the extrinsic values in memory and not permanently on the device. Is there a way to write the calibrated values into the sensor (L515) itself?

closed 2021-09-17 by MartyG-RealSense