realsenseai/realsense-ros #1718: [Foxy] RealSense Viewer works but got no image from rqt_image_viewer or rviz2

RoboticsYY, 2021-02-23:
Ubuntu 20.04
ROS2: Foxy
Installed librealsense from debian package: 
[image]

RealSense Viewer woks OK with the firmware info below:
[image]

But got no image from the rqt_image_viewer or Rviz2.

The launch messages:
```bash
intel@intel-NUC7i7DNHE:~/ros2_realsense$ ros2 launch realsense2_camera rs_launch.py
[INFO] [launch]: All log files can be found below /home/intel/.ros/log/2021-02-23-20-38-49-949433-intel-NUC7i7DNHE-5361
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [5363]
[realsense2_camera_node-1] [INFO] [1614083930.179413329] [RealSenseCameraNode]: RealSense ROS v3.1.4
[realsense2_camera_node-1] [INFO] [1614083930.179458036] [RealSenseCameraNode]: Built with LibRealSense v2.42.0
[realsense2_camera_node-1] [INFO] [1614083930.179466228] [RealSenseCameraNode]: Running with LibRealSense v2.42.0
[realsense2_camera_node-1] [INFO] [1614083930.190173159] [RealSenseCameraNode]: Device with serial number 728612070279 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1614083930.190198165] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1614083930.190207054] [RealSenseCameraNode]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [1614083930.190438570] [RealSenseCameraNode]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1614083930.190450164] [RealSenseCameraNode]: Device USB type: 3.2
[realsense2_camera_node-1] [INFO] [1614083930.191720661] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1614083930.191819024] [RealSenseCameraNode]: 707
[realsense2_camera_node-1] [INFO] [1614083930.192701932] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1614083930.192715616] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1614083930.192726404] [RealSenseCameraNode]: Device Name: Intel RealSense D435
[realsense2_camera_node-1] [INFO] [1614083930.192734533] [RealSenseCameraNode]: Device Serial No: 728612070279
[realsense2_camera_node-1] [INFO] [1614083930.192742078] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1614083930.192750489] [RealSenseCameraNode]: Device FW version: 05.12.11.00
[realsense2_camera_node-1] [INFO] [1614083930.192757723] [RealSenseCameraNode]: Device Product ID: 0x0B07
[realsense2_camera_node-1] [INFO] [1614083930.192764536] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1614083930.192771307] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1614083930.192778276] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1614083930.192789981] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1614083930.195888232] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1]  23/02 20:38:50,196 WARNING [140247383926528] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[realsense2_camera_node-1] [INFO] [1614083930.204480357] [RealSenseCameraNode]: RGB Camera was found.
[realsense2_camera_node-1] [INFO] [1614083930.204523334] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614083930.204532758] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614083930.204540808] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614083930.204548478] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614083930.204557387] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614083930.204564961] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614083930.204573383] [RealSenseCameraNode]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1614083930.204580136] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1614083930.215008369] [RealSenseCameraNode]: Failed to set parameter:rgb_camera.power_line_frequency = 3[0, 2]
[realsense2_camera_node-1] parameter 'rgb_camera.power_line_frequency' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1614083930.215191105] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1614083930.216017042] [RealSenseCameraNode]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[realsense2_camera_node-1] [INFO] [1614083930.217217974] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[realsense2_camera_node-1] [INFO] [1614083930.217236854] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1614083930.219359708] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1]  23/02 20:38:50,239 WARNING [140247182599936] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
[realsense2_camera_node-1]  23/02 20:38:50,239 WARNING [140247182599936] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
[realsense2_camera_node-1]  23/02 20:38:50,239 WARNING [140247182599936] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
[realsense2_camera_node-1] [INFO] [1614083930.247926820] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1614083930.248951942] [RealSenseCameraNode]: RealSense Node Is Up!
[realsense2_camera_node-1]  23/02 20:38:50,239 WARNING [140247182599936] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
[realsense2_camera_node-1]  23/02 20:38:50,258 WARNING [140247174207232] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=a
[realsense2_camera_node-1]  23/02 20:38:50,258 WARNING [140247174207232] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=a
[realsense2_camera_node-1]  23/02 20:38:50,258 WARNING [140247174207232] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=a
[realsense2_camera_node-1]  23/02 20:38:50,258 WARNING [140247174207232] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=a
[realsense2_camera_node-1] [WARN] [1614083930.379627174] [RealSenseCameraNode]:
```

RoboticsYY, 2021-02-23:
Seems it is a QoS policy issue, according to [rqt_image_view/issues/36](https://github.com/ros-visualization/rqt_image_view/issues/36).

After change the policy to `system_default` in Rviz2, got the image.

closed 2021-02-23 by RoboticsYY

SimonOuyang7386, 2021-03-25:
How to change the policy to system_default in Rviz2?