realsenseai/realsense-ros #1719: realsense2_camera rs_camera.launch hangs at "Initalizing nodelet with 8 worker threads"

jrhapsodus, 2021-02-23:
Greetings,

I have a two 435 realsense setup that I am trying to automatically start when the device gets powered. Currently, I am able to launch and start the cameras just fine if I am plugged into it with a monitor, keyboard, and mouse. However, when I do a displayless boot up and ssh into the machine to start the real sense cameras it hangs at the point of "Initalizing nodelet with 8 worker threads".

The real sense viewer launches fine when I am connected via a monitor and launch through a normal terminal. The launch command I use is:

`roslaunch realsense2_camera rs_camera.launch camera:=front_camera serial_no:=XXXXXXXX color_width:=1280 color_height:=720`

The log I get is:

`localhost@localhost:~$ roslaunch realsense2_camera rs_camera.launch camera:=front_camera serial_no:=XXXXXXXXX color_width:=1280 color_height:=720 &
[2] 2612
localhost@localhost:~$ ... logging to /home/localhost/.ros/log/6d570814-75fd-11eb-bd85-4401bb9f7641/roslaunch-localhost-2612.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:44711/

SUMMARY
========

PARAMETERS
 * /front_camera/realsense2_camera/accel_fps: 250
 * /front_camera/realsense2_camera/accel_frame_id: front_camera_acce...
 * /front_camera/realsense2_camera/accel_optical_frame_id: front_camera_acce...
 * /front_camera/realsense2_camera/align_depth: False
 * /front_camera/realsense2_camera/aligned_depth_to_color_frame_id: front_camera_alig...
 * /front_camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: front_camera_alig...
 * /front_camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: front_camera_alig...
 * /front_camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: front_camera_alig...
 * /front_camera/realsense2_camera/aligned_depth_to_infra1_frame_id: front_camera_alig...
 * /front_camera/realsense2_camera/aligned_depth_to_infra2_frame_id: front_camera_alig...
 * /front_camera/realsense2_camera/allow_no_texture_points: False
 * /front_camera/realsense2_camera/base_frame_id: front_camera_link
 * /front_camera/realsense2_camera/calib_odom_file: 
 * /front_camera/realsense2_camera/clip_distance: -2.0
 * /front_camera/realsense2_camera/color_fps: 30
 * /front_camera/realsense2_camera/color_frame_id: front_camera_colo...
 * /front_camera/realsense2_camera/color_height: 720
 * /front_camera/realsense2_camera/color_optical_frame_id: front_camera_colo...
 * /front_camera/realsense2_camera/color_width: 1280
 * /front_camera/realsense2_camera/confidence_fps: 30
 * /front_camera/realsense2_camera/confidence_height: 480
 * /front_camera/realsense2_camera/confidence_width: 640
 * /front_camera/realsense2_camera/depth_fps: 30
 * /front_camera/realsense2_camera/depth_frame_id: front_camera_dept...
 * /front_camera/realsense2_camera/depth_height: 480
 * /front_camera/realsense2_camera/depth_optical_frame_id: front_camera_dept...
 * /front_camera/realsense2_camera/depth_width: 640
 * /front_camera/realsense2_camera/device_type: 
 * /front_camera/realsense2_camera/enable_accel: False
 * /front_camera/realsense2_camera/enable_color: True
 * /front_camera/realsense2_camera/enable_confidence: True
 * /front_camera/realsense2_camera/enable_depth: True
 * /front_camera/realsense2_camera/enable_fisheye1: False
 * /front_camera/realsense2_camera/enable_fisheye2: False
 * /front_camera/realsense2_camera/enable_fisheye: False
 * /front_camera/realsense2_camera/enable_gyro: False
 * /front_camera/realsense2_camera/enable_infra1: False
 * /front_camera/realsense2_camera/enable_infra2: False
 * /front_camera/realsense2_camera/enable_infra: False
 * /front_camera/realsense2_camera/enable_pointcloud: False
 * /front_camera/realsense2_camera/enable_pose: False
 * /front_camera/realsense2_camera/enable_sync: False
 * /front_camera/realsense2_camera/filters: 
 * /front_camera/realsense2_camera/fisheye1_frame_id: front_camera_fish...
 * /front_camera/realsense2_camera/fisheye1_optical_frame_id: front_camera_fish...
 * /front_camera/realsense2_camera/fisheye2_frame_id: front_camera_fish...
 * /front_camera/realsense2_camera/fisheye2_optical_frame_id: front_camera_fish...
 * /front_camera/realsense2_camera/fisheye_fps: 30
 * /front_camera/realsense2_camera/fisheye_frame_id: front_camera_fish...
 * /front_camera/realsense2_camera/fisheye_height: 480
 * /front_camera/realsense2_camera/fisheye_optical_frame_id: front_camera_fish...
 * /front_camera/realsense2_camera/fisheye_width: 640
 * /front_camera/realsense2_camera/gyro_fps: 400
 * /front_camera/realsense2_camera/gyro_frame_id: front_camera_gyro...
 * /front_camera/realsense2_camera/gyro_optical_frame_id: front_camera_gyro...
 * /front_camera/realsense2_camera/imu_optical_frame_id: front_camera_imu_...
 * /front_camera/realsense2_camera/infra1_frame_id: front_camera_infr...
 * /front_camera/realsense2_camera/infra1_optical_frame_id: front_camera_infr...
 * /front_camera/realsense2_camera/infra2_frame_id: front_camera_infr...
 * /front_camera/realsense2_camera/infra2_optical_frame_id: front_camera_infr...
 * /front_camera/realsense2_camera/infra_fps: 30
 * /front_camera/realsense2_camera/infra_height: 480
 * /front_camera/realsense2_camera/infra_rgb: False
 * /front_camera/realsense2_camera/infra_width: 640
 * /front_camera/realsense2_camera/initial_reset: False
 * /front_camera/realsense2_camera/json_file_path: 
 * /front_camera/realsense2_camera/linear_accel_cov: 0.01
 * /front_camera/realsense2_camera/odom_frame_id: front_camera_odom...
 * /front_camera/realsense2_camera/ordered_pc: False
 * /front_camera/realsense2_camera/pointcloud_texture_index: 0
 * /front_camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /front_camera/realsense2_camera/pose_frame_id: front_camera_pose...
 * /front_camera/realsense2_camera/pose_optical_frame_id: front_camera_pose...
 * /front_camera/realsense2_camera/publish_odom_tf: True
 * /front_camera/realsense2_camera/publish_tf: True
 * /front_camera/realsense2_camera/rosbag_filename: 
 * /front_camera/realsense2_camera/serial_no: 033322070064
 * /front_camera/realsense2_camera/tf_publish_rate: 0.0
 * /front_camera/realsense2_camera/topic_odom_in: odom_in
 * /front_camera/realsense2_camera/unite_imu_method: 
 * /front_camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /front_camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311/

process[front_camera/realsense2_camera_manager-1]: started with pid [2627]
process[front_camera/realsense2_camera-2]: started with pid [2628]
[ INFO] [1614101981.322606584]: Initializing nodelet with 8 worker threads.

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic

with Ros Melodic and the libRealSense head commit on Jan 20.
commit c5403a28af7a82a30e970f78d4e2c1a3649686b0 (HEAD -> development, origin/development, origin/HEAD)
`

 All my other ros nodes launch find via ssh or via a terminal plugged into a monitor. I am not the strongest with ubuntu so if anyone knows how to auto start the real sense cameras from a boot up, I would appreciate the tip!

Thanks in advance for your help

MartyG-RealSense, 2021-02-24:
Hi @jrhapsodus   I have not encountered the **No LSB modules are available** error with RealSense before.  It looks as though it is an Ubuntu related issue rather than a problem with librealsense or the RealSense ROS wrapper.  

My research indicates that it is related to a missing module called **lsb** that can be installed using the instruction below:

**apt-get install lsb-core**

In regard to an auto-start of a roslaunch after boot-up: whilst I do not have a RealSense-specific reference, some non-RealSense ROS users recommend a solution called **robot_upstart**.  There is a setup guide for it in the link below if you wish to explore that method.

https://roboticsbackend.com/make-ros-launch-start-on-boot-with-robot_upstart/

jrhapsodus, 2021-02-24:
I appreciate the response,

The "No LSB modules are available." was from an output of calling 'lsb_release -a' to get the Ubuntu version. I edited my original post to make the quote formatting more clear about what was from the actual output. I installed lsb-core but it did not change the results, I still get a hang trying to launch the realsense cameras through an ssh connection:

MartyG-RealSense, 2021-02-24:
Was the information about starting a roslaunch after boot with **robot_upstart** helpful to you, please?

Regarding the SSH issue, I am reminded of a ROS tutorial guide by Intel for using RealSense across two computers that makes use of ssh.

https://github.com/IntelRealSense/realsense-ros/wiki/showcase-of-using-3-cameras-in-2-machines

On the first computer, **roscore** runs in one terminal and any RealSense cameras attached to that computer each have their own separate terminal with their own individual roslaunch too.

The second computer is accessed with ssh login to that computer's ROS host IP address.  It can then use **export ROS_MASTER_URI** to link to the first computer with roscore running on it.

AustinDeric, 2021-02-25:
are you sure the topics are not being published?  You may want to add the `--screen` flag to the roslaunch command in order to get all the logs in the terminal, For example:
```
roslaunch realsense2_camera rs_camera.launch camera:=front_camera serial_no:=XXXXXXXX color_width:=1280 color_height:=720 --screen
```

jrhapsodus, 2021-03-02:
> You may want to add the `--screen` flag to the roslaunch command

Thanks for the suggestion! The robot_upstart package was a help, yes. I managed to get it running after tinkering with the system for a day. I am not sure what finally did it, but I can say that one of the big things was to scrutinize and cleanup ubuntu's environment setups in  .bash_profile .profile and .bashrc files.

closed 2021-03-02 by jrhapsodus

MartyG-RealSense, 2021-03-03:
Great news about your success @jrhapsodus - thanks very much for sharing details of your clean-up.  :)