realsenseai/realsense-ros #1721: Problem in creating a 2D map

mohammad-h-b, 2021-02-24:
Hi,
I've made the following point cloud but when running `rosrun map_server map_saver map:=/rtabmap/proj_map –f my_map_1` while I'm playing back my rosbag, I get the following message and it saves an incomple map. Could you please help me with that?

```
(base) mohammad@mohammad:~$ rosrun map_server map_saver -f map:=/rtabmap/proj_map my_map_1
[ INFO] [1614183445.487625035]: Waiting for the map
[ INFO] [1614183445.750561386]: Received a 120 X 159 map @ 0.050 m/pix
[ INFO] [1614183445.750589096]: Writing map occupancy data to my_map_1.pgm
[ INFO] [1614183445.751070031]: Writing map occupancy data to my_map_1.yaml
[ INFO] [1614183445.751157548]: Done
```

Saved point cloud:
[image]

Savede incomplde 2D map:
[image]

MartyG-RealSense, 2021-02-24:
Hi @mohammad-h-b   The SLAM D435i guide does not guarantee to provide a 2D map that is a close duplication of the point cloud, but instead "something like" it.

For example, the image below shows side by side the point cloud image and the 2D map version of it, which is noticably simpler in detail.

[image]

I note that your 2D map is zoomed in at 134%.  Do you see more of the image if you zoom out to 100%?

[image]

mohammad-h-b, 2021-02-24:
Thanks @MartyG-RealSense Zooming out doesn't give anymore detail. A simple 2D map is also good. I would like to know if I can extract the 2D map from the `rtabmap.db` file. Besides, why `rosrun map_server map_saver map:=/rtabmap/proj_map –f my_map_1` command doesn't save the 2D map while playing back the rosbag? Should I set any further settings in rviz?

[image]

MartyG-RealSense, 2021-02-24:
I found the same link about extracting a 2D map from RTABMAP that you did.

As that discussion is from 2015, I found an alternative reference that is more modern (2019) and seems to confirm the method used in the 2015 article.

https://answers.ros.org/question/316634/trouble-exporting-map-from-rtab-map/

I also researched your question about having to wait until the bag finishes before extracting the map.  It seems to be accepted practice to do so in order to ensure that the data finishes being processed before it is extracted.

MartyG-RealSense, 2021-03-02:
Hi @mohammad-h-b  Do you require further assistance with this case, please?   Thanks!

mohammad-h-b, 2021-03-02:
Hi @MartyG-RealSense No thanks, I'm good.

MartyG-RealSense, 2021-03-02:
Great, thanks very much @mohammad-h-b for the update!

closed 2021-03-02 by MartyG-RealSense