realsenseai/realsense-ros #1724: Can't install the IntelRealsense packages on Raspberry Pi 4

etienn8, 2021-02-26:
Hi,
 I followed the instructions in [Debian package instruction](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) and when I try to use `sudo apt update`, I get this:
```
Hit:1 http://packages.ros.org/ros/ubuntu focal InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Hit:3 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:6 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal InRelease' doesn't support architecture 'arm64'
```
It seems that there is no binary packages for arm64. So I can't download librealsense2 packages. Is this normal? 

Setup:
- Raspberry Pi 4
- Ubuntu 20.04.2 LTS (GNU/Linux 5.4.0-1028-raspi aarch64)
- ROS Noetic is installed

Thanks

MartyG-RealSense, 2021-02-26:
Hi @etienn8   My experience of cases involving using **Ubuntu** with Pi 4 and RealSense ROS is that the installation process tends to be awkward.  Would it be possible to instead use the **Raspbian** OS, as there is a guide called **eleccelerator** for setting up librealsense, ROS and the ROS wrapper on Pi 4 and Raspbian.

https://github.com/IntelRealSense/realsense-ros/issues/1375#issuecomment-691482673

If you prefer to use Ubuntu 20.04, I located a recently published guide for an Ubuntu **Server** 20.04 install of librealsense, ROS Noetic and the RealSense ROS wrapper on Pi 4.

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

MartyG-RealSense, 2021-03-05:
Hi @etienn8  Do you require further assistance with this case, please?  Thanks!

etienn8, 2021-03-07:
No thank you!

closed 2021-03-07 by MartyG-RealSense

MartyG-RealSense, 2021-05-27:
For the benefit of future readers of this case, I am adding a link to a YouTube video about installation of RealSense and ROS on a Pi 4 with Ubuntu Server 20.04.

https://www.youtube.com/watch?v=hOZinOGPY-s

magicmomo123, 2021-08-18:
Thank you for all your help Marty. I would like to add a solution that I used while installing realsense on a RP4 with Ubuntu 20.04.  

https://answers.ros.org/question/363889/intel-realsens-on-ubuntu-2004-ros-noetic-installation-desription/

MartyG-RealSense, 2021-08-19:
You are very welcome, @magicmomo123 - thanks very much for sharing with the RealSense community the solution that worked for you   :)