realsenseai/realsense-ros #1725: [ROS2 Foxy] - T265 -  Crash upon launch

paucarre, 2021-02-27:
Using:
- librealsense in tag `v2.42.0`
- realsense-ros in branch `foxy`
- foxy Ros2

Steps:
- Build librealsense using: `cmake ../ -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release  -DBUILD_PYTHON_BINDINGS=true -DBUILD_CV_EXAMPLES=true -DBUILD_PCL_EXAMPLES=true  -DBUILD_WITH_CUDA=true && make && sudo make install` 
- Opened `realsense-viewer`, seen **Tracking Module**, set to on, and visualized two _fisheye_ cameras as well as _accelerometer_, _gyro_ and _pose_. Closed the viewer.
- built using `colcon build`
- run `ros2 launch realsense2_camera rs_launch.py`
- ros launch crashed with following logs:
```
...
[realsense2_camera_node-1] [ERROR] [1614394661.981356271] [RealSenseCameraNode]: An exception has been thrown: /CONFIDENTIAL_FOLDER/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1901:Invalid profile configuration - setting a single FE stream is not supported
[realsense2_camera_node-1] Invalid profile configuration - setting a single FE stream is not supported
[realsense2_camera_node-1] stop_streaming() failed. T265 device is not streaming!
[realsense2_camera_node-1] terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
[realsense2_camera_node-1]   what():  close() failed. T265 device was not opened!
[ERROR] [realsense2_camera_node-1]: process has died [pid 1232161, exit code -6, cmd '/CONFIDENTIAL_FOLDER/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_u0t796wv'].

```
Where the parameters file contents are:
```
/camera/camera:
  ros__parameters:
    accel_fps: -1.0
    align_depth: false
    allow_no_texture_points: false
    calib_odom_file: ''
    camera_name: camera
    clip_distance: -2.0
    color_fps: -1.0
    color_height: -1
    color_qos: SENSOR_DATA
    color_width: -1
    confidence_fps: -1.0
    confidence_height: -1
    confidence_qos: SENSOR_DATA
    confidence_width: -1
    config_file: ''
    depth_fps: -1.0
    depth_height: -1
    depth_qos: SENSOR_DATA
    depth_width: -1
    device_type: ''
    enable_accel: false
    enable_color: true
    enable_confidence: true
    enable_depth: true
    enable_fisheye1: true
    enable_fisheye2: true
    enable_gyro: false
    enable_infra1: true
    enable_infra2: true
    enable_pointcloud: false
    enable_sync: false
    filters: ''
    fisheye_fps: -1.0
    fisheye_height: -1
    fisheye_qos: SENSOR_DATA
    fisheye_width: -1
    gyro_fps: -1.0
    infra_fps: -1.0
    infra_height: -1
    infra_qos: SENSOR_DATA
    infra_rgb: false
    infra_width: -1
    initial_reset: false
    json_file_path: ''
    linear_accel_cov: 0.01
    ordered_pc: false
    output: screen
    pointcloud_texture_index: 0
    pointcloud_texture_stream: RS2_STREAM_COLOR
    rosbag_filename: ''
    serial_no: ''
    tf_publish_rate: 0.0
    topic_odom_in: ''
    unite_imu_method: ''
    usb_port_id: ''
```
And the full output is:
```
[INFO] [launch]: All log files can be found below 
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [1232161]
[realsense2_camera_node-1] [INFO] [1614394661.956534960] [RealSenseCameraNode]: RealSense ROS v3.1.4
[realsense2_camera_node-1] [INFO] [1614394661.956631061] [RealSenseCameraNode]: Built with LibRealSense v2.42.0
[realsense2_camera_node-1] [INFO] [1614394661.956654668] [RealSenseCameraNode]: Running with LibRealSense v2.42.0
[realsense2_camera_node-1] [INFO] [1614394661.972837241] [RealSenseCameraNode]: Device with serial number 943222110295 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1614394661.972882987] [RealSenseCameraNode]: Device with physical ID 4-2-11 was found.
[realsense2_camera_node-1] [INFO] [1614394661.972898352] [RealSenseCameraNode]: Device with name Intel RealSense T265 was found.
[realsense2_camera_node-1] [INFO] [1614394661.973175203] [RealSenseCameraNode]: Device with port number 4-2 was found.
[realsense2_camera_node-1] [INFO] [1614394661.973209146] [RealSenseCameraNode]: Device USB type: 3.1
[realsense2_camera_node-1] [INFO] [1614394661.974674835] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1614394661.974777502] [RealSenseCameraNode]: 707
[realsense2_camera_node-1] [INFO] [1614394661.976940347] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1614394661.976970169] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1614394661.976981064] [RealSenseCameraNode]: Device Name: Intel RealSense T265
[realsense2_camera_node-1] [INFO] [1614394661.976987350] [RealSenseCameraNode]: Device Serial No: 943222110295
[realsense2_camera_node-1] [INFO] [1614394661.976993566] [RealSenseCameraNode]: Device physical port: 4-2-11
[realsense2_camera_node-1] [INFO] [1614394661.976998385] [RealSenseCameraNode]: Device FW version: 0.2.0.951
[realsense2_camera_node-1] [INFO] [1614394661.977006835] [RealSenseCameraNode]: Device Product ID: 0x0B37
[realsense2_camera_node-1] [INFO] [1614394661.977012981] [RealSenseCameraNode]: Enable PointCloud: Off
[realsense2_camera_node-1] [INFO] [1614394661.977018848] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1614394661.977023597] [RealSenseCameraNode]: Sync Mode: Off
[realsense2_camera_node-1] [INFO] [1614394661.977037775] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1614394661.977065572] [RealSenseCameraNode]: Tracking Module was found.
[realsense2_camera_node-1] [INFO] [1614394661.977081286] [RealSenseCameraNode]: (Depth, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614394661.977089877] [RealSenseCameraNode]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614394661.977096442] [RealSenseCameraNode]: (Infrared, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614394661.977103147] [RealSenseCameraNode]: (Infrared, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614394661.977110340] [RealSenseCameraNode]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614394661.977116626] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614394661.977125566] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1614394661.977133248] [RealSenseCameraNode]: num_filters: 0
[realsense2_camera_node-1] [INFO] [1614394661.977138905] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1614394661.977309108] [RealSenseCameraNode]: Failed to set parameter:tracking_module.frames_queue_size = 256[0, 32]
[realsense2_camera_node-1] parameter 'tracking_module.frames_queue_size' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1614394661.977513813] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1614394661.977718867] [RealSenseCameraNode]: fisheye1 stream is enabled - width: 848, height: 800, fps: 30, Format: Y8
[realsense2_camera_node-1] [INFO] [1614394661.977897731] [RealSenseCameraNode]: fisheye2 stream is enabled - width: 848, height: 800, fps: 30, Format: Y8
[realsense2_camera_node-1] [INFO] [1614394661.977914074] [RealSenseCameraNode]: pose stream is enabled - fps: 200
[realsense2_camera_node-1] [INFO] [1614394661.977918823] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1614394661.980886868] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1] [ERROR] [1614394661.981356271] [RealSenseCameraNode]: An exception has been thrown: CONFIDENTIAL_FOLDER/src/hw_interface/sensors/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1901:Invalid profile configuration - setting a single FE stream is not supported
[realsense2_camera_node-1] Invalid profile configuration - setting a single FE stream is not supported
[realsense2_camera_node-1] stop_streaming() failed. T265 device is not streaming!
[realsense2_camera_node-1] terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
[realsense2_camera_node-1]   what():  close() failed. T265 device was not opened!
[ERROR] [realsense2_camera_node-1]: process has died [pid 1232161, exit code -6, cmd 'CONFIDENTIAL_FOLDER/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_u0t796wv'].

```

Could anybody please help me or guide me to fix this error?

doronhi, 2021-02-28:
Thanks for reporting. Please check out the fix in #1728

paucarre, 2021-02-28:
> Thanks for reporting. Please check out the fix in #1728

Thanks, tested and working!

doronhi, 2021-03-01:
Great. Thanks for testing and sorry for the trouble. Closing the issue then.

closed 2021-03-01 by doronhi