realsenseai/realsense-ros #1726: Where can I set the resolution for color image and depth image?

DeepDuke, 2021-02-27:
For example, I want to change  640*480 resolution into 424*240 for both color image and depth image. Where can I set it?

MartyG-RealSense, 2021-02-27:
Hi @DeepDuke   You could edit the resolutions in the launch file.  You could also set it in the roslaunch instruction though by adding the commands below to the end of the roslaunch statement.

**depth_width:=424 depth_height:=240 color_width:=424 color_height:=240**

DeepDuke, 2021-02-27:
> Hi @DeepDuke You could edit the resolutions in the launch file. You could also set it in the roslaunch instruction by adding the commands below to the end of the roslaunch statement.
> 
> **depth_width:=424 depth_height:=240 color_width:=424 color_height:=240**

Hi, I tried to modify these parameters in `rs_camera.launch`, but when I execute `rostopic echo /camera/color/camera_info` , I still get the original resolution information. It seems changes in launch files don't work.

MartyG-RealSense, 2021-02-27:
After you perform roslaunch, the current settings applied for height and width for the depth and color should be listed in the launch log.  What do they say please?

[image]

DeepDuke, 2021-02-27:
> After you perform roslaunch, the current settings applied for height and width for the depth and color should be listed in the launch log. What do they say please?
> 
> [image]
> 
> Also, can you confirm please that you tried adding the provided configuration instructions to the end of the roslaunch instruction? If you use this method then you do not need to edit the details inside the launch file too.
> 
> **roslaunch realsense2_camera rs_camera.launch depth_width:=424 depth_height:=240 color_width:=424 color_height:=240**

Do you mean the information printed in the terminal where I launch `rs_camera.launch` or the `.log` file in `.ros` folder. In the terminal, I did see settings like `/camera/realsense2_camera/depth_width: 424`. But in the corresponding `*.log` file, I did't find similar information.  

I also tried to add configuration to the end of the launch instruction, but when I execute rostopic echo /camera/color/camera_info , I still get the original resolution information like before.

MartyG-RealSense, 2021-02-27:
I mean the information printed in the terminal when you launch **rs_camera.launch**.  It is my experience that the terminal launch log accurately reflects what the current resolution setting is.

DeepDuke, 2021-03-01:
Well, I finally write a node to  lower the resolution through subsampling images by hand

DeepDuke, 2021-03-01:
@MartyG-RealSense Thanks you all the same.

MartyG-RealSense, 2021-03-01:
Great news that you achieved a solution @DeepDuke - thanks very much for the update!  Do you require any further assistance, please?

DeepDuke, 2021-03-04:
No further assistance needed, thank you!

closed 2021-03-04 by DeepDuke

lorepieri8, 2021-06-30:
This worked for me:
`
roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30
`

MartyG-RealSense, 2021-06-30:
Thanks very much @lorepieri8 for sharing with the RealSense ROS community the solution that worked for you!

pythongo1, 2021-07-02:
@lorepieri8  
Thanks! This really works for me!
I think maybe because the color_fps in rs_camera.launch is originally -1, 
no matter how I modify the color_width and color_height, the resolution is not changed.

After modifying color_fps from -1 to 30, the world becomes beautiful.

tonydle, 2021-08-31:
Hi all, just passing by as I was also trying to do this. Specifying fps fixed the issue! Here's how to do it from a launch file for convenience:
```
<include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="align_depth" value="true" />
    <arg name="color_width" value="1280" />
    <arg name="color_height" value="720" />
    <arg name="color_fps" value="30" />
    <arg name="depth_width" value="1280" />
    <arg name="depth_height" value="720" />
    <arg name="depth_fps" value="30" />
</include>
```

MartyG-RealSense, 2021-08-31:
Thanks very much @tonydle for your advice to this discussion!  Yes, from ROS wrapper 2.2.22 onwards a custom configuration has to specify three factors (width, height, FPS), otherwise the launch deems the custom configuration invalid and applies the default configuration for the particular camera model being used.  Thanks again!

ok-kewei, 2023-05-30:
> Hi all, just passing by as I was also trying to do this. Specifying fps fixed the issue! Here's how to do it from a launch file for convenience:
>
> ```
> <include file="$(find [...]

Thanks for the launch file. But fps 30 doesn't work for 1280 * 720 resolution. I changed it to 15 and it worked.