realsenseai/realsense-ros #1730: (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

rafaelRinconFo, 2021-03-01:
Hello there!!

I'm currently getting the following log with a d435i camera:

(synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

I saw a similar issue #https://github.com/IntelRealSense/realsense-ros/issues/1604 and in that case, it seemed to be that it was a configuration parameters related Issue. However, our configuration is the following:

    
    <arg name="color_width"               default="640"/>
    <arg name="color_height"              default="360"/>
    <arg name="depth_width"               default="640"/>
    <arg name="depth_height"              default="360"/>
    <arg name="infra_fps"                 default="15"/>
Which is supposed to be supported. I would really appreciate your help, the camera works fine in theory, I'm getting color and depth data, but, this log is published every 2 or 3 times per second. Which can be really annoying and inconvenient for debugging processes related to other nodes.

MartyG-RealSense, 2021-03-01:
Hi @rafaelRinconFo   As a starting point in investigating your case, could you please try adding the command below to the end of your roslaunch instruction to perform a reset of the camera at launch:

**initial_reset:=true**

rafaelRinconFo, 2021-03-01:
Afted adding the parameter, it doesn't seem to remove the logs. @MartyG-RealSense

MartyG-RealSense, 2021-03-01:
Can you share your roslaunch instruction please?

Also, are you trying to make use of **infra** topics, as setting a custom infra FPS would suggest.  If you are using infra, the infra topics are disabled by default and so you should make sure they are enabled.  You can do so inside the launch file or by adding the commands below to the end of your roslaunch instruction:

**enable_infra1:=true enable_infra2:=true**

rafaelRinconFo, 2021-03-02:
These are my current configuration parameters in the rs_launch.py file (I'm currently running the node on ROS2 Foxy, I'm sorry I didn't mention it earlier). As you can see, both **enable_infra1** and **enable_infra2** are set as true. I didn't set the **initial_reset** param in true, I wanted you to see the original configuration file.

`configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': '', 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': '', 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': '', 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': '', 'description': 'yaml config file'},
                           {'name': 'enable_pointcloud',            'default': 'true', 'description': 'enable pointcloud'},
                           {'name': 'unite_imu_method',             'default': 'copy', 'description': '[copy|linear_interpolation]'},                           
                           {'name': 'json_file_path',               'default': '', 'description': 'allows advanced configuration'},                           
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},                           
                           {'name': 'depth_width',                  'default': '640', 'description': 'depth image width'},                           
                           {'name': 'depth_height',                 'default': '360', 'description': 'depth image height'},                           
                           {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'color_width',                  'default': '640', 'description': 'color image width'},                           
                           {'name': 'color_height',                 'default': '360', 'description': 'color image height'},                           
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'infra_width',                  'default': '640', 'description': 'infra width'},
                           {'name': 'infra_height',                 'default': '360', 'description': 'infra width'},
                           {'name': 'enable_infra1',                'default': 'true', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'true', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'fisheye_width',                'default': '848', 'description': 'fisheye width'},
                           {'name': 'fisheye_height',               'default': '800', 'description': 'fisheye width'},
                           {'name': 'enable_fisheye1',              'default': 'true', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye2',              'default': 'true', 'description': 'enable fisheye2 stream'},
                           {'name': 'confidence_width',             'default': '640', 'description': 'depth image width'},                           
                           {'name': 'confidence_height',            'default': '480', 'description': 'depth image height'},                           
                           {'name': 'enable_confidence',            'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'fisheye_fps',                  'default': '30.', 'description': ''},                           
                           {'name': 'depth_fps',                    'default': '15', 'description': ''},                           
                           {'name': 'confidence_fps',               'default': '30.', 'description': ''},                           
                           {'name': 'infra_fps',                    'default': '15', 'description': ''},                           
                           {'name': 'color_fps',                    'default': '15', 'description': ''},                           
                           {'name': 'gyro_fps',                     'default': '400.', 'description': ''},                           
                           {'name': 'accel_fps',                    'default': '250.', 'description': ''},    
                           {'name': 'color_qos',                    'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                           {'name': 'confidence_qos',               'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                           {'name': 'depth_qos',                    'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                           {'name': 'fisheye_qos',                  'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
[... cut ...]
                           {'name': 'calib_odom_file',              'default': '', 'description': ''},                           
                           {'name': 'topic_odom_in',                'default': '', 'description': 'topic for T265 wheel odometry'},
                           {'name': 'tf_publish_rate',              'default': '0.0', 'description': 'Rate of publishing static_tf'},
                           {'name': 'rosbag_filename',              'default': '', 'description': 'A realsense bagfile to run from as a device'},
                          ]`

MartyG-RealSense, 2021-03-02:
You could set **enable_fisheye1** and  **enable_fisheye2** to false instead of true, as the D435i does not have fisheye sensors - they are a RealSense T265 feature.

**enable_pointcloud** could be set to false if you are not using a pointcloud.

You could probably set **enable_infra2** to false to save some bandwidth and just use infra1.

**align_depth** could be set to false, as its purpose is to publish additional 'aligned' topics where all of the images are aligned to depth (e.g **/camera/aligned_depth_to_color/image_raw**).

rafaelRinconFo, 2021-03-02:
The PointCloud image and the  aligned depth image are currently being used. Hence, **enable_pointcloud** and **align_depth** must be kept as true. 

However, there is no Issue with **enable_fisheye1** and ** enable_fisheye2**. I would like to ask: which are the implications of using **enable_infra2** as false?

MartyG-RealSense, 2021-03-02:
Doronhi the RealSense ROS wrapper developer has said in the past about making infra2 false: 

"It will have no effect on the depth quality.  It only disables the infra2 images' transmission via the USB port.  They are still taken and the depth is calculated using them inside the device".

AndreV84, 2021-03-02:
we got similar issue
actually the complication is that /camera/color/info & /camera/color/image_red_raw are empty, while depth topics seems got pusblishing
in the terminal there is also
```
968 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,001 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,035 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,068 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,101 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,135 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,168 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,201 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,235 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,268 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,301 ERROR [547474106784] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_other_to_depthILi4EEvPhPKhPK4int2PK14rs2_intrinsicsS8_
[realsense2_camera_node-1]  02/03 11:51:17,335 ERROR [547474106784] (syn

rafaelRinconFo, 2021-03-04:
> Doronhi the RealSense ROS wrapper developer has said in the past about making infra2 false: 
> 
> "It will have no effect on the depth quality.  It only disables the infra2 images' transmission via the USB port.  They are still taken and the depth is calculated using them inside the device".

Hey there !! It seems that after all, the logs have been drastically reduced. I'm sorry for my late response but we were trying to be sure that everything was fine.

We turned off all the not available parameters in the D435i camera and the logs were reduced. 

Thanks for your support.

closed 2021-03-04 by rafaelRinconFo

MartyG-RealSense, 2021-03-04:
Great to hear @rafaelRinconFo - thanks very much for the update!