realsenseai/realsense-ros #1731: accel and gyro topics not published

sapoluri, 2021-03-01:
This is my setup:
Ubuntu 18.04.5
Kernel 5.4.0-1
realsense SDK v2.42.0
realsense FW v5.12.11
ros melodic
ros realsense v2.2.22

Neither Accel nor Gyro topics get published. The result is the same with Realsense v2.41.0, ROS Realsense v2.2.21 and FW 5.12.10. Please help with debugging this.

MartyG-RealSense, 2021-03-01:
Hi @sapoluri   The IMU topics are disabled by default in the RealSense ROS wrapper and need to be enabled.  This can be done by editing the launch file, or by adding the commands below to the end of your roslaunch instruction.

**enable_gyro:=true enable_accel:=true.**

The IMU topics also need a RealSense camera model equipped with an IMU, such as D435i or D455, in order to be published.

sapoluri, 2021-03-04:
That resolved the issue. Thank you!

closed 2021-03-04 by sapoluri

MartyG-RealSense, 2021-03-04:
Great to hear, @sapoluri - thanks very much for the update!