realsenseai/realsense-ros #1732: Roslaunch realsense2_camera rs_rs_camera.launch    FAILED!

wuyanze0226, 2021-03-02:
[ERROR] [1614674346.281907197]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1614674346.281930034]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1614674346.282040812]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 15803, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/wuyanze/.ros/log/bf5b7c46-7b32-11eb-b962-1c697a3b18d2/camera-realsense2_camera-3.log].
log file: /home/wuyanze/.ros/log/bf5b7c46-7b32-11eb-b962-1c697a3b18d2/camera-realsense2_camera-3*.log

wuyanze0226, 2021-03-02:
I have checked my USB and realsense-viewer and they are ok.I tried to reinstall,but there is still the error.Thanks for your help. 
I install it in kinetic and ubuntu16.04.

MartyG-RealSense, 2021-03-02:
Hi @wuyanze0226   Which librealsense SDK and RealSense ROS wrapper versions are you using, please?

wuyanze0226, 2021-03-02:
Hi.
Librealsense SDK is downloaded from https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md,
and RealSense ROS wrapper is downloaded from https://github.com/IntelRealSense/realsense-ros.
Is there anything wrong?
Thank you.

MartyG-RealSense, 2021-03-02:
I meant the version numbers (e.g librealsense SDK **2.42.0** and ROS wrapper **2.2.22**).  Still, if your roslaunch log does not mention a version mismatch then they are probably okay.

Did you install using the ROS wrapper's **Method 2** please (installing librealsense first and then buildng the ROS wrapper from source code separately).

https://github.com/IntelRealSense/realsense-ros#method-2-the-realsense-distribution

wuyanze0226, 2021-03-03:
Yes.I did.I want to know how to check my librealsense SDK and ROS wrapper? Is there any code or some command line?

MartyG-RealSense, 2021-03-03:
If you check the launch log that is generated after roslaunch then in the lower half of the log it lists the librealsense and ROS wrapper versions that are being used by the launch:

[image]

closed 2021-03-16 by MartyG-RealSense