realsenseai/realsense-ros #1733: Non-linearity in depth measurements

Omega-Phone, 2021-03-02:
Hello, 
I'm working with D435 camera for an AGV. In order to navigate in the environnement, i'm trying to perform a ground removal based on RANSAC algorithm. 
However, I am not able to fit a plane to the true ground since there is a non-linearity in the depth measurements.  
To illustrate the problem, here are some screenshots: 
[image]
[image]

The 2 pictures above are displaying the same thing: it is just a room with a clear ground and a desk at the complete right. The first one is taken from real sense viewer, while the second is a plot of a numpy array in octave. From here, everything seems ok.

But the 2 same pictures from left profile are different: 
[image]
[image]

Since the ground in realsense viewer appears to be completely plane, the ground plotted in octave is more following a quadratic function, which make hard to determine ground plane.
Maybe this is a normal behavior since the error depends on z², but I would like to know if there is a way to correct the measurements with calibration or something else. 

I hope I've been clear enough, thanks in advance.

MartyG-RealSense, 2021-03-03:
Hi @Omega-Phone  Although this case is posted on the RealSense ROS forum, it looks as though it is a **librealsense SDK** case.  Is that correct, please?

I could find very few references about using octave with RealSense, so it is difficult for me to advise about that approach to ground segmentation with RANSAC.  

If you are using librealsense, the SDK makes use of the plane-fit algorithm in the link below, which it has integrated into the SDK's **Depth Quality Tool** as a validated approach to implementing plane-fit.  

http://www.ilikebigbits.com/2015_03_04_plane_from_points.html

https://github.com/IntelRealSense/librealsense/issues/3866#issuecomment-487520955

More information about this can be found in the links below:

https://github.com/IntelRealSense/librealsense/issues/4440 

https://github.com/IntelRealSense/librealsense/issues/3866 

Though the algorithm is implemented as C++ in the Depth Quality Tool, a RealSense team member has stated that the algorithm can be re-written for Python.

A different, simpler approach that could be considered is to set up a **depth clamp** that defines a minimum and maximum depth distance, so that depth data outside of that range is excluded from the image.  If, for example, a minimum depth distance of 0.5 meters was set then the camera would exclude depth data under 0.5 meters, such as the ground directly in front of the AGV.  This could help to avoid mis-triggering of obstacle detection due to detection of the ground at short range.

Omega-Phone, 2021-03-03:
Hi @MartyG-RealSense, and first of all, thank for your time ! 

Indeed, it's more an issue for **librealrense SDK** than **ROS** forum, could I change it somehow ? 

Actually, Octave is not really the point, and RANSAC algorithm neither. I wrote a RANSAC algorithm in python and it is working as expected in itself.

My issue is that I cannot obtain from python librealsense a ground plane that is effectively a plane. Instead, I have a ground plane that follow a quadratic curve as the one plotted in Octave on the above pictures.

What is actually curious is that the point cloud from Real Sense Viewer give me a nice ground plane as I expect (The 2 other pictures above). However it is strictly the same room.

So I'm asking if you have an idea to explain the different results from librealsense and Real Sense Viewer, and if it exists a configuration or calibration to correct the ground curve in order to obtain a flat ground plane such in Real Sense Viewer or https://github.com/IntelRealSense/librealsense/issues/4440. 

Thanks in advance.

MartyG-RealSense, 2021-03-03:
It is no problem to continue the current librealsense-related case here.  For future librealsense cases, please post them at the forum linked to below.

https://github.com/IntelRealSense/librealsense/issues

The RealSense Viewer applies a range of configuration presets (such as depth colorization color-scheme) and post-processing filters by default when it is launched.  However, an application that you have created yourself lacks these pre-configured settings, meanng that they have to be deliberately programmed into your scripting in order to be recreated.  The output from a self-created program may therefore have differences to the output of the Viewer.

There are a number of different approaches to ground segmentation with Python, such as using PCL and its SACSegmentation class.

https://github.com/IntelRealSense/librealsense/issues/6338#issuecomment-623278392

A Python point cloud processing library called **pyntcloud** that is compatible with RealSense 400 Series cameras may meet your needs and be relatively straightforward to implement though.

https://github.com/IntelRealSense/librealsense/issues/6989#issuecomment-667570521

Omega-Phone, 2021-03-04:
In order to add those settings and post-processing filters (I have already add some), is there a way to access the whole configuration of RealSense Viewer ? This way I could be able to reproduce the same configuration in my program. 

Thank you for the link about ground segmentation, I will take a look !

MartyG-RealSense, 2021-03-04:
You can define settings in the options categories of the **Stereo Module** settings of the RealSense Viewer and then export them from the Viewer as a custom Visual Preset (json) file.  This json can then be loaded into your own program when it launches to apply the configuration settings in the json.  The json export option is near the top of the Viewer's side-panel.

[image]

Post-processing filters have to be manually programmed into an application.

The link below has an example of Python code for loading a custom json file.

https://github.com/IntelRealSense/librealsense/issues/6075

[image]

Omega-Phone, 2021-03-08:
Update: I found the problem, I was simply missing the fact that there are 2 coordinate systems. I didn't project from pixel to point with rs2_project_point_to_pixel function. 
Thanks again for your time.

MartyG-RealSense, 2021-03-08:
Excellent news that you found the solution @Omega-Phone  - thanks very much for the update!

closed 2021-03-14 by MartyG-RealSense