realsenseai/realsense-ros #1735: d435i camera imu sync issue

snakehaihai, 2021-03-03:
Hi 
I`m currently having issue running with 3/Mar/2020 realsense-ros build

The image and imu are completely out of sync. 

And is there any chance some 1 added the wrong digit at the wrong place e.g first 2 of the nsec should be at secs?

[image]

MartyG-RealSense, 2021-03-03:
Hi @snakehaihai   As a starting point in investigating your case, could you try setting **global time** to False please by adding the instruction below to the end of your roslaunch instruction to see whether that improves your results:

**global_time_enabled:=false**

snakehaihai, 2021-03-03:
> Hi @snakehaihai As a starting point in investigating your case, could you try setting **global time** to False please by adding the instruction below to the end of your roslaunch instruction to see whether that improves your results:
> 
> **global_time_enabled:=false**

Just comeback from lab. will try it tmr

snakehaihai, 2021-03-04:
Just a quick update. I tried on my friend's PC which loaded with i7+  41 version of librealsense and there is no such issue. so likely this is tie to tx2 + 42 version of librealsense issue.

snakehaihai, 2021-03-04:
> Hi @snakehaihai As a starting point in investigating your case, could you try setting **global time** to False please by adding the instruction below to the end of your roslaunch instruction to see whether that improves your results:
> 
> **global_time_enabled:=false**

I tried to launch it. still no effect 

[image]

snakehaihai, 2021-03-04:
hi
I`m using 41 now but with tx2. but the imu header timestamp is still count at the wrong position 
[image]

snakehaihai, 2021-03-04:
changing base_realsense_node.cpp code solved issue. There is a time stamp logic error in the code, i didn't manage to identify it. I just solve it by giving the time at publisher. 

[image]

snakehaihai, 2021-03-04:
sry.. still can not. just realize that the time are totally not sync and none-peroidc. with the modification on the code. this is the result
[image]

snakehaihai, 2021-03-04:
I tried it on DJI manifold 2C. both 41 and 42 version works without any time sync issue. 

So the issue only ties with DJI manifold 2G/TX2 based platform. likely due to some error during the patch for the kernel.

MartyG-RealSense, 2021-03-04:
Thanks very much for the information @snakehaihai - I look forward to an update after your kernel delay tests.   Good luck!

snakehaihai, 2021-03-05:
no luck. not the kernel delay issue as other IMUs.  The only possibility is to upgrade L4T version but DJI mf2G does not have other higher L4T options.

MartyG-RealSense, 2021-03-05:
There have been a couple of past cases where setting global time to false OR performing an initial_reset to reset the camera at launch have not improved timing, but performing BOTH global time = false and initial_reset have worked.  In other words, adding this to the end of the roslaunch instruction:

**global_time_enabled:=false initial_reset:=true**

closed 2021-03-18 by MartyG-RealSense