realsenseai/realsense-ros #1736: Aligned Ordered Pointcloud in ROS2 Foxy

rafaelRinconFo, 2021-03-04:
Hello there guys, It's me again.

Our team is having an issue:

The wrapper currently gives us the following topic for the pointcloud:

`/camera/depth/color/points`

The Pointcloud can be set in the configuration parameters as `ordered` and `unordered`  (as set by the `ordered_pc` parameter from the configuration file). However, we have found that there is a considerable difference between the ordered an unordered PointClouds: the former is not ailgned with the RGB image and the latter is indeed aligned.

This is not really convenient because our model uses the ordered PointCloud in order to get the XYZ coordinates of an RGB pixel in the real world, the unordered Pointcloud is not suitable for our algorithm.

Is there any form to obtain an aligned version of the Ordered Pointcloud with the RGB image?

rafaelRinconFo, 2021-03-04:
@charlielito

MartyG-RealSense, 2021-03-04:
Hi @rafaelRinconFo   The recent discussion in the link below may be relevant to your needs in regard to an ordered ROS2 Foxy point cloud.

https://github.com/IntelRealSense/realsense-ros/issues/1680

rafaelRinconFo, 2021-03-05:
After reading the issue, I can tell that up to this point, it is Work in progress. Is there any way to know if the team has made any progress on this subject?

MartyG-RealSense, 2021-03-05:
Internal discussions during the consideration by Intel of a feature request are not available publicly.  New features are usually mentioned in the **New features** section of the release notes of the wrapper version that they are introduced in.

https://github.com/IntelRealSense/realsense-ros/releases

MartyG-RealSense, 2021-03-12:
Hi @rafaelRinconFo  Do you require further assistance with this case, please?  Thanks!

rafaelRinconFo, 2021-03-12:
Hi @MartyG-RealSense, so far, there hasn't been a conclusive solution to this. @charlielito  gave a temporary solution in [`1680`](https://github.com/IntelRealSense/realsense-ros/issues/1680) but so far, we have no definitive answer to it.

I wouldn't say its closed yet.

MartyG-RealSense, 2021-03-12:
Given that there is no timeline currently available regarding the pull-request DSO-16639 that Doronhi created in relation to https://github.com/IntelRealSense/realsense-ros/issues/1680 it may be better to focus discussion at that case to make it easy to communicate any updates, and close this particular case here.

rafaelRinconFo, 2021-03-12:
If [1680](https://github.com/IntelRealSense/realsense-ros/issues/1680) will be kept as open, I have no issue in closing this one.

MartyG-RealSense, 2021-03-12:
Yes, https://github.com/IntelRealSense/realsense-ros/issues/1680 will be kept open.  So I will close this one with your agreement.  Thanks again!

closed 2021-03-12 by MartyG-RealSense