realsenseai/realsense-ros #1737: expected [string] got [integer] with ROS eloquent

nedo99, 2021-03-05:
Hi,

I am running realsense on Ubuntu 18.04 with ROS eloquent. I have multiple cameras connected and I need to specify which camera I want to use so I run:
`ros2 launch realsense2_camera rs_launch.py enable_pointcloud:=true serial_no:=1111111111111`
but I am getting:
`realsense_node_factory.cpp:314:expected [string] got [integer]`

I was running this with ROS melodic and ti was working fine.

What changed?

Regards,
Nedim

MartyG-RealSense, 2021-03-05:
Hi @nedo99   Could you put an underslash '_- in front of your serial numbers please?  

**ros2 launch realsense2_camera rs_launch.py enable_pointcloud:=true serial_no:=_1111111111111**

This is referenced in the Eloquent documentation:

"Serial number can also be defined with "_" prefix.  For instance, serial number 831612073525 can be set in command line as serial_no:=_831612073525.  That is a workaround until a better method will be found to ROS2's auto conversion of strings containing only digits into integers".

nedo99, 2021-03-05:
Hi @MartyG-RealSense 

Thank you for your quick reply. That fixed the issue with starting, but now I have a different one:
`(backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
[camera-1] Size 148480 out of 614655 bytes (24%)`

Regards,
Nedim

MartyG-RealSense, 2021-03-05:
A RealSens team member has said in the past about the **Incomplete video frame detected** error: "The partial frames issue usually indicates that the uvcvideo driver does not get enough USB buffers from the USB subsystem.  This can be either because the system is overloaded (possibly thermal-throttled), CPU utilisation is near 100%, or the USB controller is not handling the requests in time".

There was a case of a RealSense ROS user who reported that they received this error more often if they were using an aged or low-quality USB cable:

https://github.com/IntelRealSense/librealsense/issues/3189

MartyG-RealSense, 2021-03-12:
Hi @nedo99  Do you require further assistance with this case, please?  Thanks!

nedo99, 2021-03-12:
Hi,

forgot to update. It is fine. Thanks for support!

closed 2021-03-12 by nedo99

MartyG-RealSense, 2021-03-12:
Excellent news - thanks very much for the update!