realsenseai/realsense-ros #1739: [ERROR] Failed to load library /opt/ros/melodic/lib//librealsense2_camera.so

derektan95, 2021-03-05:
Hi, 

I installed the this package via binaries using `sudo apt-get install ros-melodic-realsense2-camera`, and launch attempted to launch start up my camera via: 

`roslaunch realsense2_camera rs_camera.launch`

Unfortunately, I encountered this error after i executed the abovementioned command:  

```
[ERROR] [1614943854.269977184]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library /opt/ros/melodic/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.42: cannot open shared object file: No such file or directory)
[ERROR] [1614943854.270165568]: The error before refreshing the cache was: Failed to load library /opt/ros/melodic/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.42: cannot open shared object file: No such file or directory)
[FATAL] [1614943854.270463072]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 13871, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/unitree/.ros/log/3a762a00-7da6-11eb-bb16-00044bde90f5/camera-realsense2_camera-2.log].
log file: /home/unitree/.ros/log/3a762a00-7da6-11eb-bb16-00044bde90f5/camera-realsense2_camera-2*.log
```

I attempted to reinstall the package via these commands but it didn't work. 
```
 sudo apt-get remove ros-melodic-realsense2-camera
 sudo apt-get install ros-melodic-realsense2-camera
```

Do you have any insights to this issue?

EDIT: Just wanted to note that I initially wanted to install this package by source, and cloned the repository and ran the shellscripts in the scripts folders as per [these instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). However, I stopped when building the package midway and did not perform any installation. I decided to install it via binaries instead and I encountered the abovementioned issue. In the past, installing via binaries (without attempting installation via source) works fine.

MartyG-RealSense, 2021-03-06:
Hi @derektan95   Is the problem resolved if you delete the catkin workspace and then create a new one and build fro source gaain, please?

derektan95, 2021-03-06:
Hi @MartyG-RealSense, 

I installed this via binaries rather than from source, because it is more convenient to do so. It seemed to work previously. 

Am I right to say that since this is installed via binaries, the solution should not be related to doing anything with my catkin workspace?

MartyG-RealSense, 2021-03-06:
When mixing package and source builds, there is the potential for conflicts to be introduced if you have files from both on the computer at the same time.  This is true for librealsense too.

derektan95, 2021-03-06:
Hi @MartyG-RealSense , 

Ah i see. That being said, based on the error messages, do you have any insights on how I can solve this? I just verified that the scriptable object file `/opt/ros/melodic/lib//librealsense2_camera.so` does exist in my directory.

MartyG-RealSense, 2021-03-06:
If you find a package build more convenient and do not plan to use a source-code build then you could try deleting the catkin workspace that the source-code build was performed in.

derektan95, 2021-03-06:
Hi @MartyG-RealSense, 

I have already attempted by deleting both the `build/` and `devel/` folders, which should contain all cache related to the realsense-ros package. However, this error is still appearing for some reason. It seems to imply that I am missing the `librealsense2_camera.so` file, but it is there when I checked.

MartyG-RealSense, 2021-03-06:
To me, deleting the catkin workspace means deleting all of it - **/catkin_ws/src**

doronhi, 2021-03-07:
The error message you posted suggests that what you are missing is **librealsense2.so.2.42**, not librealsense2_camera.so. It should have also been installed into `/opt/ros/melodic/lib`. I just tried it and it worked for me so right now I don't have any thoughts as to what the issue may be but maybe if you looked again at the installation log you'll find some clues.

derektan95, 2021-03-07:
Hi @doronhi , 

Thanks. Am I right to say that `librealsense2.so.2.42` should have been automatically installed in `/opt/ros/melodic/lib` when I use the command `sudo apt-get install ros-melodic-realsense2-camera`?

doronhi, 2021-03-07:
I am not an expert on debians but as far as I can tell and test, yes, the package should have been installed there automatically. I was hoping you may find something in the installation log.

derektan95, 2021-03-08:
Hi @doronhi, 

It turns out that there was something wrong with the cache. I did the following and the issue is fixed: 

```
sudo apt-get update
sudo apt-get autoremove
sudo apt-get install ros-melodic-realsense2-camera
``` 
 Many thanks for your help!

closed 2021-03-08 by derektan95

MartyG-RealSense, 2021-03-08:
You are very welcome @derektan95 - great news that you found the solution!

workingenius, 2021-03-11:
I met the same problem and @doronhi 's hint help me solve it.
I'm using ubuntu 20.04, ros noetic.

In my case, it was `librealsense2.so.2.41` that didn't found. I didn't get it by only reading the error message.
The lost .so file turned out to be in `/opt/ros/noetic/lib/x86_64-linux-gnu`. I modified `LD_LIBRARY_PATH` and it worked.

I have no idea why the .so was put there but not `/opt/ros/noetic/lib` as you said.
Maybe the installation guide could take that case into account.

Thanks.

HuaYuXiao, 2023-12-03:
> Hi @derektan95 Is the problem resolved if you delete the catkin workspace and then create a new one and build the RealSense ROS wrapper from source again, please?

This method works for me! Below shows my versions:

[image]
[image]
[image]
[image]

My ERROR message is as follows:

```
[FATAL] [1701582745.999741172]: Failed to load nodelet '/t265/realsense2_cameraof typerealsense2_camera/RealSenseNodeFactoryto managerrealsense2_camera_manager'
[ERROR] [1701582746.019403561]: Failed to load nodelet [/d400/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library /home/amov/realsense_ws/devel/lib//librealsense2_camera.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = librealsense2.so.2.48: cannot open shared object file: No such file or directory)
```

It tells that it tends to use `librealsense` of version2.48, which is different from the actual version2.36. I guess it may result from the previous built cache, so I cleaned up the workspace and `catkin_make` from scratch. Finally the launch runs perfectly!