realsenseai/realsense-ros #1740: D435i - how to get stereo mono image without ir markers?

srnetsr, 2021-03-05:
I would like to try VIO with two mono images from D435i.

But I can see IR markers on images.
Is there any _arg_ to disable it for stereo mono images?

MartyG-RealSense, 2021-03-06:
Hi @srnetsr   Could you try disabling the emitter with a rosparam, please?

```
<rosparam>
/camera/stereo_module/emitter_enabled: false
</rosparam>
```

srnetsr, 2021-03-08:
Thank you, @MartyG-RealSense 
`$ rosparam list`
I cannot find any parameter of `emitter_enabled`

I am just testing with `rs_camera.launch` file in the realsense-ros package.
Can you advise more specifically?

MartyG-RealSense, 2021-03-08:
A rosparam instruction would be manually added to the contents of the launch file, in-between <rosparam> and </rosparam> (the start and end points of the rosparam section).

I believe that if the file does not currently contain <rosparam> and </rosparam> definitions then you should paste the entire code block into the file.

```
<rosparam>
/camera/stereo_module/emitter_enabled: false
</rosparam>
```

Alternatively, you could test the instruction by adding **emitter_enabled:=false** to the end of your roslaunch instruction instead:

**roslaunch realsense2_camera rs_camera.launch emitter_enabled:=false**

srnetsr, 2021-03-10:
Hi @MartyG-RealSense ,
I think something wrong.

I did `roslaunch realsense2_camera rs_camera.launch emitter_enabled:=false` and run rviz. The camera is still showing the IR as shown in the attached screen capture
[image]

Also, when I run `$ rosparam list`
```
$ rosparam list
/camera/color/image_raw/compressed/format
/camera/color/image_raw/compressed/jpeg_quality
/camera/color/image_raw/compressed/png_level
/camera/color/image_raw/compressedDepth/depth_max
/camera/color/image_raw/compressedDepth/depth_quantization
/camera/color/image_raw/compressedDepth/png_level
/camera/color/image_raw/ffmpeg/bit_rate
/camera/color/image_raw/ffmpeg/encoder
/camera/color/image_raw/ffmpeg/fps_
/camera/color/image_raw/ffmpeg/gop_size
/camera/color/image_raw/ffmpeg/measure_performance
/camera/color/image_raw/ffmpeg/performance_interval
/camera/color/image_raw/ffmpeg/pix_fmt
/camera/color/image_raw/ffmpeg/preset
/camera/color/image_raw/ffmpeg/profile
/camera/color/image_raw/ffmpeg/qmax
/camera/color/image_raw/theora/keyframe_frequency
/camera/color/image_raw/theora/optimize_for
/camera/color/image_raw/theora/quality
/camera/color/image_raw/theora/target_bitrate
/camera/depth/image_rect_raw/compressed/format
/camera/depth/image_rect_raw/compressed/jpeg_quality
/camera/depth/image_rect_raw/compressed/png_level
/camera/depth/image_rect_raw/compressedDepth/depth_max
/camera/depth/image_rect_raw/compressedDepth/depth_quantization
/camera/depth/image_rect_raw/compressedDepth/png_level
/camera/depth/image_rect_raw/ffmpeg/bit_rate
/camera/depth/image_rect_raw/ffmpeg/encoder
/camera/depth/image_rect_raw/ffmpeg/fps_
/camera/depth/image_rect_raw/ffmpeg/gop_size
/camera/depth/image_rect_raw/ffmpeg/measure_performance
/camera/depth/image_rect_raw/ffmpeg/performance_interval
/camera/depth/image_rect_raw/ffmpeg/pix_fmt
/camera/depth/image_rect_raw/ffmpeg/preset
/camera/depth/image_rect_raw/ffmpeg/profile
/camera/depth/image_rect_raw/ffmpeg/qmax
/camera/depth/image_rect_raw/theora/keyframe_frequency
/camera/depth/image_rect_raw/theora/optimize_for
/camera/depth/image_rect_raw/theora/quality
/camera/depth/image_rect_raw/theora/target_bitrate
/camera/infra1/image_rect_raw/compressed/format
/camera/infra1/image_rect_raw/compressed/jpeg_quality
/camera/infra1/image_rect_raw/compressed/png_level
/camera/infra1/image_rect_raw/compressedDepth/depth_max
/camera/infra1/image_rect_raw/compressedDepth/depth_quantization
/camera/infra1/image_rect_raw/compressedDepth/png_level
/camera/infra1/image_rect_raw/ffmpeg/bit_rate
/camera/infra1/image_rect_raw/ffmpeg/encoder
/camera/infra1/image_rect_raw/ffmpeg/fps_
/camera/infra1/image_rect_raw/ffmpeg/gop_size
/camera/infra1/image_rect_raw/ffmpeg/measure_performance
/camera/infra1/image_rect_raw/ffmpeg/performance_interval
/camera/infra1/image_rect_raw/ffmpeg/pix_fmt
/camera/infra1/image_rect_raw/ffmpeg/preset
/camera/infra1/image_rect_raw/ffmpeg/profile
/camera/infra1/image_rect_raw/ffmpeg/qmax
/camera/infra1/image_rect_raw/theora/keyframe_frequency
/camera/infra1/image_rect_raw/theora/optimize_for
/camera/infra1/image_rect_raw/theora/quality
/camera/infra1/image_rect_raw/theora/target_bitrate
/camera/infra2/image_rect_raw/compressed/format
[... 70 lines cut ...]
/camera/realsense2_camera/infra2_frame_id
/camera/realsense2_camera/infra2_optical_frame_id
/camera/realsense2_camera/infra_fps
/camera/realsense2_camera/infra_height
/camera/realsense2_camera/infra_width
/camera/realsense2_camera/initial_reset
/camera/realsense2_camera/json_file_path
/camera/realsense2_camera/linear_accel_cov
/camera/realsense2_camera/odom_frame_id
/camera/realsense2_camera/pointcloud_texture_index
/camera/realsense2_camera/pointcloud_texture_stream
/camera/realsense2_camera/pose_frame_id
/camera/realsense2_camera/pose_optical_frame_id
/camera/realsense2_camera/publish_odom_tf
/camera/realsense2_camera/rosbag_filename
/camera/realsense2_camera/serial_no
/camera/realsense2_camera/topic_odom_in
/camera/realsense2_camera/unite_imu_method
/rosdistro
/roslaunch/uris/host_nuc_ranger__43227
/rosversion
/run_id
```

can you double check the way to disable IR emitter?

MartyG-RealSense, 2021-03-10:
There was a previous case where the emitter was turning on after launch even though it had been correctly set to false using the commands that I provided above.

https://github.com/IntelRealSense/realsense-ros/issues/1379

The solution by the RealSense user in that case was to "disable the emitter in the base node itself by default".

closed 2021-03-23 by MartyG-RealSense