realsenseai/realsense-ros #1741: L515 and '[ERROR] Device already streaming!' in 1024x768 depth resolution

bigbellmercy, 2021-03-08:
Hello,
The errror, '[ERROR] Device already streaming!', happens with L515 sensor:

### Error
> 08/03 17:57:23,583 ERROR [139747693733632] (l500-depth.cpp:733) Exception caught in l500_depth_sensor::open
> [ERROR] [1615193843.583584804]: An exception has been thrown: Device already streaming!
> [ERROR] [1615193843.583928698]: Exception: Device already streaming!

### Condition
But this error only happens when the parameters `depth_width` and `depth_height` are set to `1024` and `768` resolution maximally. But it does not happen when they are set to `640` and `480` resolution.

### Environment
Ubuntu 18.04 in x86 PC
L515 firmware v1.5.4.1
realsense2-ros (realsense2_camera) v2.2.22
librealsense2 v2.42.0

### Console log:
```
user@user-npc-2020:~/catkin_ws2/src/realsense-ros/realsense2_camera/launch$ roslaunch realsense2_camera l515.launch
... logging to /home/user/.ros/log/4b197414-7fec-11eb-b60a-0c7a158b174e/roslaunch-user-npc-2020-23527.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://user-npc-2020:41577/

SUMMARY
========

PARAMETERS
 * /camera1/realsense2_camera/accel_fps: 400
 * /camera1/realsense2_camera/accel_frame_id: camera1_accel_frame
 * /camera1/realsense2_camera/accel_optical_frame_id: camera1_accel_opt...
 * /camera1/realsense2_camera/align_depth: False
 * /camera1/realsense2_camera/aligned_depth_to_color_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra1_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/aligned_depth_to_infra2_frame_id: camera1_aligned_d...
 * /camera1/realsense2_camera/allow_no_texture_points: False
 * /camera1/realsense2_camera/base_frame_id: camera1_link
 * /camera1/realsense2_camera/calib_odom_file: 
 * /camera1/realsense2_camera/clip_distance: -2.0
 * /camera1/realsense2_camera/color_fps: 30
 * /camera1/realsense2_camera/color_frame_id: camera1_color_frame
 * /camera1/realsense2_camera/color_height: 1080
 * /camera1/realsense2_camera/color_optical_frame_id: camera1_color_opt...
 * /camera1/realsense2_camera/color_width: 1920
 * /camera1/realsense2_camera/confidence_fps: 30
 * /camera1/realsense2_camera/confidence_height: 480
 * /camera1/realsense2_camera/confidence_width: 640
 * /camera1/realsense2_camera/depth_fps: 30
 * /camera1/realsense2_camera/depth_frame_id: camera1_depth_frame
 * /camera1/realsense2_camera/depth_height: 768
 * /camera1/realsense2_camera/depth_optical_frame_id: camera1_depth_opt...
 * /camera1/realsense2_camera/depth_width: 1024
 * /camera1/realsense2_camera/device_type: 
 * /camera1/realsense2_camera/enable_accel: True
 * /camera1/realsense2_camera/enable_color: True
 * /camera1/realsense2_camera/enable_confidence: True
 * /camera1/realsense2_camera/enable_depth: True
 * /camera1/realsense2_camera/enable_fisheye1: False
 * /camera1/realsense2_camera/enable_fisheye2: False
 * /camera1/realsense2_camera/enable_fisheye: False
 * /camera1/realsense2_camera/enable_gyro: True
 * /camera1/realsense2_camera/enable_infra1: False
 * /camera1/realsense2_camera/enable_infra2: False
 * /camera1/realsense2_camera/enable_infra: False
[... 88 lines cut ...]
[ INFO] [1615193843.163341341]: RGB Camera was found.
[ INFO] [1615193843.163510444]: Motion Module was found.
[ INFO] [1615193843.163563477]: Add Filter: pointcloud
[ INFO] [1615193843.164047591]: num_filters: 1
[ INFO] [1615193843.164080155]: Setting Dynamic reconfig parameters.
[ INFO] [1615193843.189486543]: Done Setting Dynamic reconfig parameters.
[ INFO] [1615193843.190043580]: depth stream is enabled - width: 1024, height: 768, fps: 30, Format: Z16
[ INFO] [1615193843.191817254]: color stream is enabled - width: 1920, height: 1080, fps: 30, Format: RGB8
[ INFO] [1615193843.191947356]: confidence stream is enabled - width: 640, height: 480, fps: 30, Format: RAW8
[ INFO] [1615193843.191979243]: gyro stream is enabled - fps: 400
[ INFO] [1615193843.192000234]: accel stream is enabled - fps: 400
[ INFO] [1615193843.192023120]: setupPublishers...
[ INFO] [1615193843.198132475]: Expected frequency for depth = 30.00000
[ INFO] [1615193843.225567299]: Expected frequency for color = 30.00000
[ INFO] [1615193843.242185064]: Expected frequency for confidence = 30.00000
[ INFO] [1615193843.259373330]: setupStreams...
[ INFO] [1615193843.310746175]: insert Depth to L500 Depth Sensor
[ INFO] [1615193843.310908239]: insert Color to RGB Camera
[ INFO] [1615193843.310974598]: insert Gyro to Motion Module
[ INFO] [1615193843.311027805]: insert Accel to Motion Module
**[ INFO] [1615193843.311090334]: insert Confidence to L500 Depth Sensor
 08/03 17:57:23,583 ERROR [139747693733632] (l500-depth.cpp:733) Exception caught in l500_depth_sensor::open
[ERROR] [1615193843.583584804]: An exception has been thrown: Device already streaming!
[ERROR] [1615193843.583928698]: Exception: Device already streaming!**
```

RealSenseSupport, 2021-03-31:
@bigbellmercy I noticed that you configured different resolution for depth and confidence map as below. Please configure the same resolution to try again. Thanks!

[ INFO] [1615193843.190043580]: depth stream is enabled - width: 1024, height: 768, fps: 30, Format: Z16
[ INFO] [1615193843.191817254]: color stream is enabled - width: 1920, height: 1080, fps: 30, Format: RGB8
[ INFO] [1615193843.191947356]: confidence stream is enabled - width: 640, height: 480, fps: 30, Format: RAW8

bigbellmercy, 2021-04-01:
Thanks! @RealSenseSupport, I didn't know that both should be the same.
But I could not find how to set the resolution of the confidence stream. Because, I could not find any such parameter in README.md. Is there some `rs_l515.launch` in `launch` folder` as an example?

doronhi, 2021-04-06:
the parameters enable_confidence, confidence_width, confidence_height and confidence_fps appear in rs_camera.launch file.
Thanks for noticing they are missing from the README.md file. Fixed it now.

bigbellmercy, 2021-04-08:
Thanks @doronhi, I added the following in the launch file as following:
```
...
  <arg name="confidence_width"    default="1024"/>
  <arg name="confidence_height"   default="768"/>
  <arg name="enable_confidence"   default="true"/>
  <arg name="confidence_fps"      default="30"/>
...
      <arg name="confidence_width"         value="$(arg confidence_width)"/>
      <arg name="confidence_height"        value="$(arg confidence_height)"/>
      <arg name="enable_confidence"        value="$(arg enable_confidence)"/>
      <arg name="confidence_fps"           value="$(arg confidence_fps)"/>
...
```
Thanks @RealSenseSupport, though the resolutions for both depth and confidence became matched as below:
```
[ INFO] [1617871139.746544853]: depth stream is enabled - width: 1024, height: 768, fps: 30, Format: Z16
[ INFO] [1617871139.746985385]: infra stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1617871139.749791197]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1617871139.750669336]: confidence stream is enabled - width: 1024, height: 768, fps: 30, Format: RAW8
```
The same error happens as following:
```
[ERROR] [1617871140.092587114]: An exception has been thrown: Device already streaming!
[ERROR] [1617871140.092927797]: Exception: Device already streaming!
```
Furthermore, when I disabled the confidence by setting `enable_confidence` to `false` the same error happens.

So, it seems that there is other cause.

bigbellmercy, 2021-04-09:
Additionally inform that this error happens in recent versions:
```
[ INFO] [1617955751.376607034]: RealSense ROS v2.2.23
[ INFO] [1617955751.376851694]: Built with LibRealSense v2.43.0
[ INFO] [1617955751.376994836]: Running with LibRealSense v2.43.0
```

closed 2021-09-17 by MartyG-RealSense