realsenseai/realsense-ros #1747: ros foxy version is not support with D435i？

CaffeMrDe, 2021-03-10:
When I ros2 launch the rs_launch.py is showed the problem .

INFO] [launch]: All log files can be found below /home/de/.ros/log/2021-03-10-08-04-05-474818-fshs-28567
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [realsense2_camera_node-1]: process started with pid [28570]
[realsense2_camera_node-1] [INFO] [1615363446.347620235] [RealSenseCameraNode]: RealSense ROS v3.1.4
[realsense2_camera_node-1] [INFO] [1615363446.347784561] [RealSenseCameraNode]: Built with LibRealSense v2.42.0
[realsense2_camera_node-1] [INFO] [1615363446.347824556] [RealSenseCameraNode]: Running with LibRealSense v2.42.0
[realsense2_camera_node-1]  10/03 08:04:06,392 WARNING [139983948453632] (ds5-device.cpp:548) The IMU sensor is undefined for PID ad6 and imu_chip_id: �
[realsense2_camera_node-1] [INFO] [1615363446.412112793] [RealSenseCameraNode]: Device with serial number 833612070569 was found.
[realsense2_camera_node-1] 
[realsense2_camera_node-1] [INFO] [1615363446.412226912] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [1615363446.412278127] [RealSenseCameraNode]: Device with name Intel RealSense USB2 was found.
[realsense2_camera_node-1] [INFO] [1615363446.413289365] [RealSenseCameraNode]: Device with port number 2-2 was found.
[realsense2_camera_node-1] [INFO] [1615363446.413415579] [RealSenseCameraNode]: Device USB type: 2.1
[realsense2_camera_node-1] [WARN] [1615363446.413472531] [RealSenseCameraNode]: Device 833612070569 is connected using a 2.1 port. Reduced performance is expected.
[realsense2_camera_node-1]  10/03 08:04:06,411 WARNING [139983948453632] (ds5-active.cpp:78) Projector capacity is overrided and disabled by FW
[realsense2_camera_node-1] Device PID = 0xad6, Capabilities Vector = [Active Projector/RGB Sensor/IMU Sensor/Global Shutter/]
[realsense2_camera_node-1] [INFO] [1615363446.431064395] [RealSenseCameraNode]: getParameters...
[realsense2_camera_node-1] [INFO] [1615363446.431933819] [RealSenseCameraNode]: 707
[realsense2_camera_node-1] [INFO] [1615363446.438106846] [RealSenseCameraNode]: setupDevice...
[realsense2_camera_node-1] [INFO] [1615363446.438168885] [RealSenseCameraNode]: JSON file is not provided
[realsense2_camera_node-1] [INFO] [1615363446.438215036] [RealSenseCameraNode]: Device Name: Intel RealSense USB2
[realsense2_camera_node-1] [INFO] [1615363446.438248360] [RealSenseCameraNode]: Device Serial No: 833612070569
[realsense2_camera_node-1] [INFO] [1615363446.438291259] [RealSenseCameraNode]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0
[realsense2_camera_node-1] [INFO] [1615363446.438333763] [RealSenseCameraNode]: Device FW version: 05.10.13.00
[realsense2_camera_node-1] [INFO] [1615363446.438375402] [RealSenseCameraNode]: Device Product ID: 0x0AD6
[realsense2_camera_node-1] [INFO] [1615363446.438413466] [RealSenseCameraNode]: Enable PointCloud: On
[realsense2_camera_node-1] [INFO] [1615363446.438450809] [RealSenseCameraNode]: Align Depth: Off
[realsense2_camera_node-1] [INFO] [1615363446.438487361] [RealSenseCameraNode]: Sync Mode: On
[realsense2_camera_node-1] [INFO] [1615363446.438666830] [RealSenseCameraNode]: Device Sensors: 
[realsense2_camera_node-1] [INFO] [1615363446.463007295] [RealSenseCameraNode]: Stereo Module was found.
[realsense2_camera_node-1] [INFO] [1615363446.463108177] [RealSenseCameraNode]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463146049] [RealSenseCameraNode]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463189632] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463230023] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463269431] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463309930] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463350442] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463392357] [RealSenseCameraNode]: Add Filter: pointcloud
[realsense2_camera_node-1] [INFO] [1615363446.485005942] [RealSenseCameraNode]: num_filters: 1
[realsense2_camera_node-1] [INFO] [1615363446.485114361] [RealSenseCameraNode]: Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [WARN] [1615363446.501996112] [RealSenseCameraNode]: Failed to set parameter:pointcloud.stream_index_filter = 0[-1, 2.14748e+09]
[realsense2_camera_node-1] parameter 'pointcloud.stream_index_filter' could not be set: Parameter {} doesn't comply with integer range.
[realsense2_camera_node-1] [INFO] [1615363446.502139246] [RealSenseCameraNode]: Done Setting Dynamic reconfig parameters.
[realsense2_camera_node-1] [INFO] [1615363446.504395659] [RealSenseCameraNode]: setupPublishers...
[realsense2_camera_node-1] [INFO] [1615363446.504469410] [RealSenseCameraNode]: setupStreams...
[realsense2_camera_node-1] [INFO] [1615363446.504508338] [RealSenseCameraNode]: SELECTED BASE:Depth, 0
[realsense2_camera_node-1] [INFO] [1615363446.504757670] [RealSenseCameraNode]: RealSense Node Is Up!
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[realsense2_camera_node-1] [INFO] [1615363452.193582026] [rclcpp]: signal_handler(signal_value=2)
[realsense2_camera_node-1] [WARN] [1615363452.195432508] [RealSenseCameraNode]: BaseRealSenseNode::~BaseRealSenseNode()
[realsense2_camera_node-1] [WARN] [1615363452.195640009] [RealSenseCameraNode]: BaseRealSenseNode::~BaseRealSenseNode() - Done
[INFO] [realsense2_camera_node-1]: process has finished cleanly [pid 28570]

CaffeMrDe, 2021-03-10:
[realsense2_camera_node-1] [INFO] [1615363446.463108177] [RealSenseCameraNode]: (Color, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463146049] [RealSenseCameraNode]: (Infrared, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463189632] [RealSenseCameraNode]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463230023] [RealSenseCameraNode]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463269431] [RealSenseCameraNode]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463309930] [RealSenseCameraNode]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO] [1615363446.463350442] [RealSenseCameraNode]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[realsense2_camera_node-1] [INFO]

CaffeMrDe, 2021-03-10:
My device :

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 007: ID 1a2c:2124 China Resource Semico Co., Ltd 
Bus 002 Device 004: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 002 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 002 Device 010: ID 8086:0ad6 Intel Corp. Intel(R) RealSense(TM) Depth Camera 435i
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

MartyG-RealSense, 2021-03-10:
Hi @CaffeMrDe    Yes, the ROS2 wrapper does support D435i.

Could you try adding the command below to the end of your roslaunch instruction to reset the camera at launch, please:

**initial_reset:=true**

CaffeMrDe, 2021-03-10:
ok, I  do the actions to solve the problem. 
`
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger 
`
 And rebuild the source project .

Thank you for your replying.

closed 2021-03-10 by CaffeMrDe

MartyG-RealSense, 2021-03-10:
Thanks very much @CaffeMrDe for sharing your solution!