realsenseai/realsense-ros #1748: realsense-viewer Segmentation fault (core dumped)

XavierEmeralds, 2021-03-10:
Distribution: Ubuntu 20.04.2 LTS
Kernel: 4.4.0-19041-Microsoft
OS: Ubuntu terminal running in Windows 10

Installing Linux Distribution: https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

After running ```sudo apt-get install librealsense2-dkms``` and ``` sudo apt-get install librealsense2-utils```, attempting to run ```realsense-viewer``` gives the error:
```
Segmentation fault (core dumped)
```

Any idea what is happening? I've tried this multiple times with clean installs and everything. I'm relatively new to ROS and Ubuntu so any specific instructions or explanations would be very much appreciated.

MartyG-RealSense, 2021-03-10:
Hi @XavierEmeralds   RealSense users who have tried to get librealsense working with Windows Subsystem for Linux (WSL) have been known to have problems with doing so.  An Intel RealSense team member comments on this in the link below:

https://github.com/IntelRealSense/librealsense/issues/8217#issuecomment-764800820

May I confirm please whether it is vital to your project to use Linux in Windows?  The RealSense SDK has a Windows 10 version that has a simple automated installer program that installs the SDK and tools such as the RealSense Viewer on Windows.    

This installer file can be found by visiting the link below and then downloading and running the file **Intel.RealSense.SDK-WIN10-2.42.0.2845.exe** from the **Assets** file list at the bottom of the page

https://github.com/IntelRealSense/librealsense/releases/tag/v2.42.0

[image]

XavierEmeralds, 2021-03-10:
Okay thank you, I'll try to get it to run in Windows 10. I was under the impression that it needed to be inside of ROS in Linux to be able to get the data for whatever reason. I'll try this and come back with results.

MartyG-RealSense, 2021-03-10:
It's not just Linux, you can build the RealSense ROS wrapper for Windows too.   :)

https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

Good luck!

XavierEmeralds, 2021-03-11:
It worked a charm! Thank you for your help.

closed 2021-03-11 by XavierEmeralds

MartyG-RealSense, 2021-03-11:
Great news @XavierEmeralds - thanks very much for the update!