realsenseai/realsense-ros #1749: Point cloud with IR intensity values

FrankLin9981, 2021-03-11:
Dear all,

Recently, I want to generate point cloud with IR intensity values for later use.
The camera I use is RealSense D415.
Then I found an previous issue related to it. #1046 

However, I follow its instruction to use **pointcloud_texture_stream ="RS2_STREAM_INFRARED"** as a launch parameter.
And the point cloud **does have the intensity field**, but the intensity values of these points are **all zeros (10e-40)**.
But I check the values of each pixel of the infrared image. These values are non zeros and between 0-256 (Y8 format).

The following are the launch parameters I use, and the launch file I use is rs_camera.launch.
Do I misunderstand anything?

Hope you guys could do me a favor!
Thanks a lot!

```
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d415"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="false"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>
  
  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_INFRARED"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="true"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="false"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="2.0"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
```

MartyG-RealSense, 2021-03-11:
Hi @FrankLin9981  Given that point cloud intensity involves assigning **color** values, I would suggest trying **RS2_STREAM_COLOR** instead.

http://wiki.ros.org/rviz/DisplayTypes/PointCloud

FrankLin9981, 2021-03-11:
Hi @MartyG-RealSense  Thanks for your quick response!  

My requirement is **not to assign color values** to the point cloud.  
I want the point cloud with intensity field which stores the IR intensity values so that I can convert the PointCloud2 message into **pcl::PointCloud<pcl::PointCloudXYZI>** datatype.  

Does it can be achieved by setting ** pointcloud_texture_stream** parameter?

MartyG-RealSense, 2021-03-11:
There is very little information on generating a point cloud with intensity information using RealSense.  Of the information that is available, a research dissertation using RealSense and ROS that used PCL to convert the pointcloud2 message into an XYZI point cloud may fit your needs. The bottom of page 29 of the document provides details of this.

https://lib.ugent.be/fulltxt/RUG01/002/785/837/RUG01-002785837_2019_0001_AC.pdf

FrankLin9981, 2021-03-12:
Sorry, I find that I misunderstand the requirement.  
The thing I want to do is actually to obtain the point cloud **with each point has a value to indicate whether the point is empty, occupied, or unknown**.  
Do you have any idea about how to do it?  

Thanks a lot!

MartyG-RealSense, 2021-03-12:
The expectation that there may be invalid points present suggests to me that you would be using a **non-dense** point cloud, as dense clouds have no invalid points and non-dense clouds can have invalid points.

**Organized** clouds are non-dense, and **unorganized** clouds are dense.  If you are using the RealSense ROS1 wrapper and you want an organized cloud that has invalid points then you could generate it with the **rs_rgbd** roslaunch file, whilst **rs_camera** would provide an unorganized cloud.

Or if you wanted to be confident that every point was filled so that you don't have to check the point at all then you could generate an unorganized dense cloud.  

More information about this subject can be found in the link below.

https://answers.ros.org/question/234455/pointcloud2-and-pointfield/?answer=234472#post-id-234472

I hope that the information above is helpful, though I am not certain if I answered the question that you were asking.

MartyG-RealSense, 2021-03-18:
Hi @FrankLin9981  Do you require further assistance with this case, please?   Thanks!

FrankLin9981, 2021-03-18:
Thanks for your help!  
I find that the thing I want to do can be achieved by using Octomap server.  
Temporarily not require further assistance.

MartyG-RealSense, 2021-03-18:
Thanks very much @FrankLin9981 for the update!  I will close the case as you achieved a solution with Octomap. Feel free to re-open the case at a future date or create a new one if you require further assistance with this subject.  Good luck!

closed 2021-03-18 by MartyG-RealSense