realsenseai/realsense-ros #1750: Hardware Notification L515 LiDAR Camera

diogomes97, 2021-03-11:
Hello! I am using the L515 LiDAR Camera, and i am getting this warning:
`Hardware Notification:Receiver light saturation, stream stopped for 1 sec,1.61538e+12,Error,Hardware Error` . Does somebody know how to solve this? Another question, can i power my camera in a usb2.0 port?

RealSenseSupport, 2021-03-26:
Hi @diogomes97 

The camera can function under USB2 port but the supported resolutions due to bandwidth is reduced as well.  

Regarding the notification that you're getting, can you help explain your overall usage model and in what environment you're using the camera in?

diogomes97, 2021-03-30:
Hello,

Thank you for the reply. I am using the camera to perform Visual Inertial
Odometry with RTAB-Map and ORB-SLAM2 packages in ROS. The environment that
I am using the camera is a lab where I have multiple windows and has some
light. If you have any tips to help me perform my goals, please let me know.

Best regards,
Diogo Gomes

woscho, 2021-04-06:
Hi RealSenseSupport,
I've the same issue over here. The L515 Camera is running fine for a couple of minutes, then the Intel RealSense Viewer is reporting

_Timestamp: 1617727931952.655273
Severity: Error
Description: Receiver light saturation, stream stopped for 1 sec_

every 2-3 s. The stream is "pulsing" ... restarting all the time. Having a look at the controls, the temperatures are rising to about LDD=48, Mc=42, Ma=41,Apd=38, then the error occures. Stopping the stream for a couple of minutes to let the camera cool down again and toggle it on again, the error reapears with roughly the same temperature values. When the error occures, the Apd temperature reports "wild values" like -13. or -9999 then again a correct one and so on.

It's a temperature issue or the Apd is faulty, I would guess!?

I'm running the camera with newest firmware 1.5.5.0 and the viewer 2.43.0.3013 for device testing purposes in our lab (with no sun light) and an ambient room temperature around 24°C.
Does the camera need active cooling?

Thanks for your help!

RealSenseSupport, 2021-04-14:
Hi @diogomes97 and @woscho 

Keeping the camera cool is a good idea in a well circulated environment.  

I'm hoping you can provide me some bag files/recordings using the RealSense Viewer with the depth, IR and RGB streaming running before and after or up to you getting the error about the light saturation.

Please provide if you can.  We would really appreciate it to help in understanding what might be wrong.  It could be a HW component that is failing within the camera that you have.

RealSenseSupport, 2021-05-06:
Hi @woscho 

I noticed you did mention that you're able to use the camera when the temps are lower.  Similar to my response to the user on the other thread, your unit may also need to be RMA.

closed 2021-06-09 by RealSenseSupport