realsenseai/realsense-ros #1751: D455 can not change the resolution.

ctxqlxs, 2021-03-12:
I've set the image resolution, but the stream keep the original resolution. why?
`roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 color_width:=640 color_height:=480`
`SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: -1
 * /camera/realsense2_camera/infra_height: -1
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: -1
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/ordered_pc: False
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[camera/realsense2_camera_manager-1]: started with pid [21117]
process[camera/realsense2_camera-2]: started with pid [21118]
[ INFO] [1615514653.719402207]: Initializing nodelet with 12 worker threads.
[ INFO] [1615514653.792461803]: RealSense ROS v2.2.22
[ INFO] [1615514653.792486789]: Built with LibRealSense v2.42.0
[ INFO] [1615514653.792521133]: Running with LibRealSense v2.42.0
[ INFO] [1615514653.815102414]:  
[ INFO] [1615514653.820827135]: Device with serial number 044422250434 was found.

[ INFO] [1615514653.820873487]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video0 was found.
[ INFO] [1615514653.820916862]: Device with name Intel RealSense D455 was found.
[ INFO] [1615514653.821468464]: Device with port number 2-6 was found.
[ INFO] [1615514653.821508325]: Device USB type: 3.2
[ INFO] [1615514653.823820484]: getParameters...
[ INFO] [1615514653.892330781]: setupDevice...
[ INFO] [1615514653.892354607]: JSON file is not provided
[ INFO] [1615514653.892371103]: ROS Node Namespace: camera
[ INFO] [1615514653.892387835]: Device Name: Intel RealSense D455
[ INFO] [1615514653.892402317]: Device Serial No: 044422250434
[ INFO] [1615514653.892421863]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/video4linux/video0
[ INFO] [1615514653.892435801]: Device FW version: 05.12.11.00
[ INFO] [1615514653.892450352]: Device Product ID: 0x0B5C
[ INFO] [1615514653.892464055]: Enable PointCloud: Off
[ INFO] [1615514653.892477489]: Align Depth: On
[ INFO] [1615514653.892490733]: Sync Mode: On
[ INFO] [1615514653.892525419]: Device Sensors: 
[ INFO] [1615514653.899547470]: Stereo Module was found.
[ INFO] [1615514653.906501112]: RGB Camera was found.
[ INFO] [1615514653.906688805]: Motion Module was found.
[ INFO] [1615514653.906712127]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1615514653.906729735]: num_filters: 0
[ INFO] [1615514653.906738672]: Setting Dynamic reconfig parameters.
[ INFO] [1615514654.080735165]: Done Setting Dynamic reconfig parameters.
**[ INFO] [1615514654.081598027]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1615514654.083596165]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8**
[ INFO] [1615514654.085884673]: setupPublishers...
[ INFO] [1615514654.089059882]: Expected frequency for depth = 30.00000
[ INFO] [1615514654.126349242]: Expected frequency for color = 30.00000
[ INFO] [1615514654.146238795]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1615514654.159087659]: setupStreams...
[ INFO] [1615514654.210290077]: insert Depth to Stereo Module
[ INFO] [1615514654.210394804]: insert Color to RGB Camera
[ INFO] [1615514654.372010794]: SELECTED BASE:Depth, 0
[ INFO] [1615514654.394750089]: RealSense Node Is Up!`

MartyG-RealSense, 2021-03-12:
Hi @ctxqlxs  The custom width and height configuration that you defined for the depth and color streams in your roslaunch instruction is being applied in the upper half of your log.

[image]

In the lower half of the log though, the default stream configuration for your particular camera model is being applied, overwriting the earlier settings.  This was a feature introduced recently in ROS wrapper 2.2.22.  The release notes about it state: "Choose the default profile in case of an invalid request.  Notice that the rs_camera.launch file now selects the default profile for each device".

It may be worth trying with the **rs_aligned_depth*  roslaunch instruction instead:

**roslaunch realsense2_camera rs_aligned_depth.launch depth_width:=640 depth_height:=480 color_width:=640 color_height:=480**

I will seek input from the ROS wrapper developer regarding rs_camera.launch.

@doronhi   Does the statement **Notice that the rs_camera.launch file now selects the default profile for each device** mean that when rs_camera.launch is used, the default profile is applied even if a custom stream width / height is defined in the roslaunch instruction?  Thanks!

doronhi, 2021-03-14:
Hi, @MartyG-RealSense , @ctxqlxs , Sorry for the confusion.
A requested profile consists of 3 parameters: width, height, and fps. By supplying only the first 2 parameters the given set is "640, 480, -1" which is invalid, and therefore realsense2_camera resorts to defaults.
The following line will run realsense2_camera in the state you asked for:
`ros2 launch realsense2_camera rs_launch.py align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=30.0 color_width:=640 color_height:=480 color_fps:=30.0`
Also notice that the "fps" parameter must be of double type.
I think the current state is indeed confusing and I consider merging those 3 into a single string parameter, something like "depth_profile:=640,480,30". If you care, I would appreciate your opinions.

MartyG-RealSense, 2021-03-14:
Thanks very much for the clarification @doronhi - I understand now.

I like your depth_profile format idea.  It is neat and tidy, and makes for a shorter roslaunch instruction than having to define separate width, height and FPS commands for each stream type, which could add up to 6 commands in a 2-stream configuration (as demonstrated in your instruction above).  

In the meantime I will communicate the correct profile definition format to anyone else who has this issue.  Thanks again!

ctxqlxs, 2021-03-15:
`roslaunch realsense2_camera rs_camera.launch align_depth:=true depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30`

closed 2021-03-15 by ctxqlxs