realsenseai/realsense-ros #1753: D435 with RPi 4b using Raspbian Buster with ROS Noetic

Serfi7, 2021-03-14:
Hello.
I have a Realsense D435, Raspberry Pi 4 with installed Raspbian Buster and ROS Noetic. I would like to use [realsense-ros](https://github.com/IntelRealSense/realsense-ros) for mapping using RTAB-Map. I tried next methods:
1. Install using `apt-get`, but it doesn't work with error `Unable to locate package ros-noetic-realsense2-camera`. As I understand some libraries have some memory leak issues on ARM architecture, ok nevermind.
2. Next I tried to install realsense using `Eleccelerator Guide` from https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Before_You_Start Firstly I could not patch uvcvideo, `make` couldn't do some stuff again due to ARM unsupported issues. Next I skipped this step and tried to install `realsense-ros` from source. In `realsense-ros` [README](https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources) instruction says that deps can be found in [.travis.yml](https://github.com/IntelRealSense/realsense-ros/blob/development/.travis.yml), but there is no dist for noetic. I tried to install deps using `apt-get`, but after `sudo apt-get install ros-noetic-cv-bridge`, for example, I had `Unable to locate package ros-noetic-cv-bridge`. I added all deps written on `.travis.yml` to catkin workspace from github repositories of deps, when I had executed `catkin_make clean` I had next errors:
```
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 14 packages in topological order:
-- ~~  - geometry (metapackage)
-- ~~  - image_common (metapackage)
-- ~~  - eigen_conversions
-- ~~  - kdl_conversions
-- ~~  - realsense2_camera_msgs (plain cmake)
-- ~~  - realsense2_description (plain cmake)
-- ~~  - ddynamic_reconfigure
-- ~~  - camera_calibration_parsers
-- ~~  - image_transport
-- ~~  - camera_info_manager
-- ~~  - polled_camera
-- ~~  - realsense2_camera (plain cmake)
-- ~~  - tf
-- ~~  - tf_conversions
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
```
 Now I don't understand how I can add required deps. Thanks for your help!

MartyG-RealSense, 2021-03-15:
Hi @Serfi7   It has been my consistent experience that installing RealSense ROS in Raspbian on a Pi 4 can be extremely difficult.  In fact, I have never handled a case where it has been successful.  In all the Pi 4 RealSense ROS cases I have been involved in that had a successful outcome, the successful method was to use **Ubuntu Server** as the OS instead of Raspbian.

The link below has a recent example of a guide for doing so with Ubuntu Server 20.04.

https://ramith.fyi/setting-up-raspberry-pi-4-with-ubuntu-20-04-ros-intel-realsense/

Serfi7, 2021-03-15:
Thank you for your answer and help. I'll try to install ubuntu server with noetic ros.

closed 2021-03-15 by Serfi7