realsenseai/realsense-ros #1754: how to measure true FOV in different resolution

zhangzhouyuan, 2021-03-15:
In Intel-RealSense-D400-Series-Datasheet-June-2020.pdf page 62 tell us: Due to mechanical tolerances of +/-5%, Max and Min FOV values can vary from lens to lens and module to module by ~ +/- 3 degrees.

Have any way to measure true FOV ?

MartyG-RealSense, 2021-03-15:
Hi @zhangzhouyuan   In the link below, the Chief Technology Officer of the RealSense Group at Intel (agrunnet) provides advice on how to calculate the FOV using maths:

https://github.com/IntelRealSense/librealsense/issues/4179#issuecomment-500860874

MartyG-RealSense, 2021-03-21:
Hi @zhangzhouyuan  Do you require further assistance with this case, please?  Thanks!

zhangzhouyuan, 2021-03-24:
ye， we already resolved ，Thanks

closed 2021-03-24 by zhangzhouyuan