realsenseai/realsense-ros #1757: Where to find default Params

swiz23, 2021-03-16:
Now that the rs_camera.launch file sets many parameters to -1 (which I assume means 'use the default value on a device basis'), where can I find what the actual defaults are for each device?

MartyG-RealSense, 2021-03-16:
Hi @swiz23   You can find the default profile that is being applied by looking near the bottom of the launch log, under the heading **Setting Dynamic reconfig parameters.**. For example:

```
[ INFO] [1615514653.906738672]: Setting Dynamic reconfig parameters.
[ INFO] [1615514654.080735165]: Done Setting Dynamic reconfig parameters.
[ INFO] [1615514654.081598027]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1615514654.083596165]: color stream is enabled - width: 1280, height: 720, fps: 30, Format: RGB8
[ INFO] [1615514654.085884673]: setupPublishers...
[ INFO] [1615514654.089059882]: Expected frequency for depth = 30.00000
[ INFO] [1615514654.126349242]: Expected frequency for color = 30.00000
```

The change was introduced in RealSense ROS wrapper version **2.2.22** and is described in the release notes quote below:

"Choose the default profile in case of an invalid request.  Notice that the rs_camera.launch file now selects the default profile for each device".

Doronhi the RealSense ROS wrapper developer has said about it: "The feature is meant to deal with the fact that not all devices have the same profiles.  It also prevents the need to define default profiles in realsense2_camera, as they are already defined in the underlying librealsense2 library".

https://github.com/IntelRealSense/realsense-ros/issues/1359#issuecomment-789432493

Under the new system, if you define a custom stream height and width in the roslaunch instruction then you must also define the **FPS** speed for that particular stream and express it as a decimal point value such as **30.0**.

https://github.com/IntelRealSense/realsense-ros/issues/1751#issuecomment-798834284

doronhi, 2021-03-16:
Thanks, @MartyG-RealSense.
I would like to add further explanation. The default profiles are defined in the librealsense2 library. They vary for different devices and if they are connected using a usb2 or a usb3 port. They are internally defined and not part of the official API. Therefore, although they tend to stay the same, they might be altered one day without notice.
I think that an occasionally refreshed table is a good idea. However, since I don't think such a table exists and it will take me a while to attend and create one, the best I could offer is to look at the source. Look for "profile_tag::PROFILE_TAG_DEFAULT" in the librealsense2 code. For instance, here: 
[ds5-device.cpp](https://github.com/IntelRealSense/librealsense/blob/8fc9ad20da01e926ba610ce4ee1b058e61b86655/src/ds5/ds5-factory.cpp#L54)
With that said, for practical reasons, I think @MartyG-RealSense 's advice to look at realsense2_camera's log is the best.

swiz23, 2021-03-16:
Thanks for the tips!

I tried setting the fps using a decimal point value, but I get an error message about the type not being correct. I saw in the nodelet.launch.xml file that the 'type' for the fps parameters are set to 'int'.

MartyG-RealSense, 2021-03-16:
I think that the FPS is an int value when set inside the launch file as an arg, such as  `<arg name="depth_fps" default="30"/>`

Can you confirm that you put the FPS value on the end of the roslaunch instruction please (not inside the file).  For example:

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30.0**

swiz23, 2021-03-17:
Ahh, no. I put it as 30.0 inside the filename. Is that alright to do? It seems to work just fine.

MartyG-RealSense, 2021-03-17:
I researched the issue and FPS values are written as an int (30) in the launch file in every example that I saw.

doronhi, 2021-03-18:
For no good reason, for now, fps values are defined as integer in ROS1 and float in ROS2...

MartyG-RealSense, 2021-03-24:
Hi @swiz23  Do you require further assistance with this case, please?   Thanks!

swiz23, 2021-03-24:
Nope! Thanks.

closed 2021-03-24 by swiz23